arendst/Tasmota #20792: DS18x20 parasitic power

wrlaw, 2024-02-24:
### PROBLEM DESCRIPTION
DS18x20 does not work out of the box using parasitic power.    Looking at the source, a critical line of code seems to have been commented out.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): ESP32 Dev Kit_____
- [x] Tasmota binary firmware version number used: 13.4.0.1(tasmota32)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: ESPtool_____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
18:01:48.155 CMD: Backlog Template; Module; GPIO 255
18:01:48.180 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
18:01:48.403 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
18:01:48.660 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"1313":"DS18x202"},"GPIO13":{"0":"None"},"GPIO14":{"1312":"DS18x201"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
No rules used

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
18:04:56.051 CMD: Status 0
18:04:56.057 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_8663F4","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
18:04:56.065 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T01:37:50","StartupUTC":"2024-02-24T05:27:06","Sleep":50,"CfgHolder":4617,"BootCount":16,"BCResetTime":"2024-02-20T12:58:50","SaveCount":44}}
18:04:56.073 RSL: STATUS2 = {"StatusFWR":{"Version":"13.4.0.1(tasmota32)","BuildDateTime":"2024-02-19T13:22:38","Core":"2_0_14","SDK":"4.4.6.240105","CpuFrequency":160,"Hardware":"ESP32-D0WDQ6 v1.0","CR":"412/699"}}
18:04:56.080 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Bluebush",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","01000080","00006000","00004000","00000000"]}}
18:04:56.100 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1908,"Free":971,"Heap":145,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80B1","00000004"],"Drivers":"1,2,!3,!4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,82,!86,!87,!88,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
18:04:56.116 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-8663F4-1012","IPAddress":"192.168.0.37","Gateway":"192.168.0.1","Subnetmask":"255.255.255.192","DNSServer1":"192.168.0.1","DNSServer2":"0.0.0.0","Mac":"24:6F:28:86:63:F4","IP6Global":"","IP6Local":"fe80::266f:28ff:fe86:63f4%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.0.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
18:04:56.126 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8663F4","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:04:56.132 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-02-24T07:04:56Z","Local":"2024-02-24T18:04:56","StartDST":"2024-10-06T02:00:00","EndDST":"2024-04-07T02:00:00","Timezone":99,"Sunrise":"06:59","Sunset":"20:07"}}
18:04:56.166 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-02-24T18:04:56","DS18S20-1":{"Id":"00080126733F","Temperature":85.0},"DS18B20-2":{"Id":"0000053545F9","Temperature":85.0},"TempUnit":"C"}}
18:04:56.175 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-02-24T18:04:56","Uptime":"0T01:37:50","UptimeSec":5870,"Heap":148,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":0,"Berry":{"HeapUsed":4,"Objects":42},"Wifi":{"AP":1,"SSId":"Bluebush","BSSId":"00:23:69:29:08:69","Channel":7,"Mode":"11n","RSSI":100,"Signal":-34,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Connect DS18x20 in a parasitic power configuration

### EXPECTED BEHAVIOUR
The sensor should show the temperature but instead displays 85 degrees C.   That is, I expected the precompiled binary to support DS18x20 sensors using parasitic power

### SCREENSHOTS
void Ds18x20Convert(void) {
  for (uint32_t i = 0; i < DS18X20Data.gpios; i++) {
    ds = ds18x20_gpios[i];
    ds->reset();
#ifdef W1_PARASITE_POWER
    // With parasite power held wire high at the end for parasitically powered devices
    ds->write(W1_SKIP_ROM, 1);        // Address all Sensors on Bus
    ds->write(W1_CONVERT_TEMP, 1);    // start conversion, no parasite power on at the end
#else
    ds->write(W1_SKIP_ROM);        // Address all Sensors on Bus
    ds->write(W1_CONVERT_TEMP);    // start conversion, no parasite power on at the end
#endif
//    delay(750);                     // 750ms should be enough for 12bit conv
  }
}

### ADDITIONAL CONTEXT
Looking at the source, within the xsns_05_esp32_ds18x20.ino file I can see a key contributor to this problem.   That is, there is a line of code with delay(750) in it but this line has been commented out.   For a parasitically powered DS18x20 to be able to do a temperature conversion, it needs it's internal capacitor to be charged and the 750 ms delay provides time for this to occur.    Without that delay, the conversion fails.

Another related issue that is also immediately noticeable is the need to define W1_PARASITE_POWER.   With some additional code, this definition is not needed.   That is, it is possible to determine whether a DS18x20 is parasitically powered by asking it.    To determine if the sensor is parasitically powered, you can reset the 1 wire bus, select the specific device address, send the hex value 0xB4 to it and read back what it says.   The code then becomes much simpler because there is no need to define W1_PARASITE_POWER.   You can treat each device based on it's specific powering mode, parasitic or non-parasitic.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2024-02-24:
You are welcome to provide a PR with the solution you have in mind.

arendst, 2024-02-24:
As you're browsing the code have a better look regarding the not used delay. The function using it is called every odd second. The follow up code is then executed every even second so there is a delay of 1 second.

I suggest you dive deeper and find the real cause of your failing parasite powered device.

closed 2024-02-25 by wrlaw