arendst/Tasmota #20794: Fix support for STDS75 (based on LM75A support)

DeflateAwning, 2024-02-24:
### PROBLEM DESCRIPTION
When the ESP32 is connected to the STDS75 temperature sensor by I2C, it should detect it as an LM75A device, as they are nearly identical. Currently, it detects it as a ADS1115 ADC device. 

It should detect it on a high address (e.g., 0x91 when A0=A1=A2=GND); currently, it's being detected on 0x48.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32 Doit Dev Kit
- [x] Tasmota binary firmware version number used: v13.4.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: web flasher
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
09:46:35.447 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
09:46:35.647 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
09:46:35.887 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"640":"I2C SDA1"},"GPIO22":{"608":"I2C SCL1"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
09:47:00.601 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_82D178","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
09:47:00.636 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:06:28","StartupUTC":"2024-02-24T08:40:32","Sleep":50,"CfgHolder":4617,"BootCount":4,"BCResetTime":"2024-02-24T09:35:49","SaveCount":10}}
09:47:00.671 RSL: STATUS2 = {"StatusFWR":{"Version":"13.4.0(tasmota32)","BuildDateTime":"2024-02-14T16:17:40","Core":"2_0_14","SDK":"4.4.6.240105","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.1","CR":"423/699"}}
09:47:00.685 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
09:47:00.724 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1907,"Free":972,"Heap":142,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80B1","00000004"],"Drivers":"1,2,!3,!4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,82,!86,!87,!88,121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
09:47:00.792 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-82D178-4472","IPAddress":"192.168.11.164","Gateway":"192.168.11.10","Subnetmask":"255.255.255.0","DNSServer1":"1.1.1.1","DNSServer2":"8.8.8.8","Mac":"E4:65:B8:82:D1:78","IP6Global":"","IP6Local":"fe80::e665:b8ff:fe82:d178%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"1.1.1.1","DNSServer2":"8.8.8.8","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
09:47:00.843 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_82D178","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
09:47:00.861 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-02-24T08:47:00Z","Local":"2024-02-24T09:47:00","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"07:42","Sunset":"18:23"}}
09:47:00.956 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-02-24T09:47:00","ADS1115":{"A0":0,"A1":0,"A2":0,"A3":0}}}
09:47:00.966 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-02-24T09:47:00","Uptime":"0T00:06:28","UptimeSec":388,"Heap":145,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":54,"MqttCount":0,"Berry":{"HeapUsed":4,"Objects":45}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
1. Solder the STDS75 to a breakout board, with pull-up resistors. Set A0=A1=A2=GND.
2. Connect the STDS75 to the ESP32 pins GPIO 21 and GPIO22 (or any pins).
3. Set those pins as SDA and SCL, as is appropriate in the configuration.
4. Navitage to the home page after the reboot.

### EXPECTED BEHAVIOUR
I expect Tasmota to correctly detect the STDS75 device on address 0x91.

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
N/A

arendst, 2024-02-24:
Read this https://github.com/arendst/Tasmota/blob/development/I2CDEVICES.md and use command i2cdriver wisely.

DeflateAwning, 2024-02-24:
That solved my issue. It's too bad this isn't linked from each device page that covers I2C. Thanks!

closed 2024-02-24 by DeflateAwning