arendst/Tasmota #20822: DS18x20 not registering on Tasmota32 on Sonoff THR316

mangusbrother, 2024-02-27:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have flashed my Sonoff THR316 wiht tasmota32 13.4.0 and configured the gpio inputs, with 27 as output_hi and 25 as ds18x20 1. However the temperature is not showing on the UI at all, and i'm not seeing updates for it in the console.  This is the same (Rj22?) sensor that came with my order of the device, that worked on the original firmware through ewelink so i do not expect hardware issues here. 
Followed this for config https://templates.blakadder.com/sonoff_THR316.html

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Sonoff THR316
- [ ] Tasmota binary firmware version number used: 13.4.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: Tasmota-Esp-Flasher 2.2.3
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
21:01:02.175 CMD: Backlog Template
21:01:02.199 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
21:01:09.675 CMD: Module
21:01:09.677 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
21:01:13.326 CMD: GPIO 255
21:01:13.336 RSL: RESULT = {"GPIO0":{"64":"Button_n1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"321":"Led_i2"},"GPIO14":{"0":"None"},"GPIO15":{"576":"LedLink_i"},"GPIO16":{"320":"Led_i1"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"256":"Relay_i1"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"1312":"DS18x201"},"GPIO26":{"0":"None"},"GPIO27":{"3840":"Output Hi"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
21:01:56.254 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_E9ACA0","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
21:01:56.288 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:06:15","StartupUTC":"2024-02-27T19:55:41","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2024-02-27T20:27:28","SaveCount":25}}
21:01:56.322 RSL: STATUS2 = {"StatusFWR":{"Version":"13.4.0(tasmota32)","BuildDateTime":"2024-02-14T16:17:40","Core":"2_0_14","SDK":"4.4.6.240105","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.0","CR":"408/699"}}
21:01:56.335 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Tonberry",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
21:01:56.378 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1907,"Free":972,"Heap":147,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80B1","00000004"],"Drivers":"1,2,!3,!4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,82,!86,!87,!88,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
21:01:56.444 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-E9ACA0-3232","IPAddress":"192.168.3.100","Gateway":"192.168.3.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.4.9","DNSServer2":"192.168.4.10","Mac":"C8:F0:9E:E9:AC:A0","IP6Global":"","IP6Local":"fe80::caf0:9eff:fee9:aca0%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.4.9","DNSServer2":"192.168.4.10","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
21:01:56.495 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_E9ACA0","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:01:56.511 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-02-27T20:01:56Z","Local":"2024-02-27T21:01:56","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"07:36","Sunset":"18:28"}}
21:01:56.541 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-02-27T21:01:56"}}
21:01:56.547 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-02-27T21:01:56","Uptime":"0T00:06:15","UptimeSec":375,"Heap":149,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":4,"Objects":47},"POWER":"OFF","Wifi":{"AP":1,"SSId":"Tonberry","BSSId":"EA:63:DA:AD:D8:9B","Channel":1,"Mode":"11n","RSSI":46,"Signal":-77,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
nothing to experience since the issue is that there is nothing

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
- flash tasmota on thr316
- configure gpio inputs
- connect sensor 
- restart, just in case

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Temperature to be visible in the main page

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2024-03-04:
If the sensor is a WTS01 then it's not supported by Tasmota.

mangusbrother, 2024-03-09:
I can confirm.
cut off the bulge in the cable and re-crimped it in same order and it is now registering

closed 2024-03-09 by mangusbrother