arendst/Tasmota #20858: SetOption not working at Gosund EP2

DB1BMN-collection, 2024-03-03:
### PROBLEM DESCRIPTION
Installed Tasmota V13.4 on Gosund EP2 WLAN-Plug (it works again to update firmware from China App).
Tried to calibrate and configure all settings and found out thet `SetOption21 1` for displaying Voltage even when switched off seems not to work (any more).
This leads also to wrong calibration settings when you calibrate voltage in "Off"-Mode.

I've read #14634. Is it possible that also the EP2 has the voltage divider behind the relay?
The Plugs are bought Nov. 2022 and Jul. 2023 from Amazon.de.
The optical similar plug "Nous A1T" (bought Aug. 2022 also from Amazon.de) shows the voltage even when the relay is switched off.

And yes, of course I typed 'savedata 1' before and '0' after and restarted it.

Anyone can reproduce?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used: Gosund EP2 from Amazon.de
- [x] Tasmota binary firmware version number used: 13.4
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: flashed to V.9 via raspberry as described on homepage then updated via Web-UI to 13.4
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:

### SCREENSHOTS
Nous A1T shows Voltage even when Switched off (sorry this is a crop from an 2022-Screenshot):
[image]

The EP2 show Voltage only in "On"-Mode. Because Calibration was done in "Off"-Mode the Voltage is completyly way off:
[image]

barbudor, 2024-03-03:
If the power measurment circuit is after the relay, there is nothing that Tasmota can do against it

DB1BMN-collection, 2024-03-03:
This is in deed the question.
As I said: The Nous A1T can measure voltage in "Off-State" - the EP2 can't yet.
Both look quite the same.
Perhaps somebody has reverse engineerd the schematics?