arendst/Tasmota #20912: Sonoff Dual R3 v2 not responding to external switch

vdiogo, 2024-03-10:
### Discussed in https://github.com/arendst/Tasmota/discussions/20886

<div type='discussions-op-text'>

<sup>Originally posted by **vdiogo** March  6, 2024</sup>

### PROBLEM DESCRIPTION
Have flashed a Sonoff Dual R3 v2 and applied the template available here https://templates.blakadder.com/sonoff_DUALR3_v2 and performed the following:

- Ensure that the first relay opens the shutter
- Ensure that the second relay closes the shutter
- Set ShutterRelay<x>
- Set ShutterOpenDuration<x>
- Set ShutterCloseDuration<x>

I can control the shutter (venetian blinds with tilt) properly from the tasmota UI, but the external switch buttons does nothing

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Dual R3 v2
- [x] Tasmota binary firmware version number used: firstly tasmota32.factory.bin and later updated to tasmota32-bluetooth.factory.bin
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: initial flash with ESP-Flasher -dev-macOS
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`: 
```lua
  Configuration output here:
15:59:04.999 MQT: stat/tasmota_bridge/RESULT = {"NAME":"Sonoff Dual R3 v2","GPIO":[32,0,0,0,0,0,0,0,0,576,225,0,0,0,0,0,0,0,0,0,0,3200,8128,224,0,0,0,0,160,161,0,0,0,0,0,0],"FLAG":0,"BASE":1}
15:59:05.231 MQT: stat/tasmota_bridge/RESULT = {"Module":{"0":"Sonoff Dual R3 v2"}}
15:59:05.496 MQT: stat/tasmota_bridge/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"225":"Relay2"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"3200":"Serial Tx"},"GPIO26":{"8128":"BL0939 Rx"},"GPIO27":{"224":"Relay1"},"GPIO32":{"160":"Switch1"},"GPIO33":{"161":"Switch2"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
16:00:06.425 MQT: stat/tasmota_bridge/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota Dual R3","FriendlyName":["Tasmota Dual R3",""],"Topic":"tasmota_bridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[1,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
16:00:06.467 MQT: stat/tasmota_bridge/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:27:54","StartupUTC":"2024-03-06T14:32:12","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2024-03-05T17:14:29","SaveCount":44}}
16:00:06.504 MQT: stat/tasmota_bridge/STATUS2 = {"StatusFWR":{"Version":"13.4.0(bluetooth)","BuildDateTime":"2024-02-14T16:17:27","Core":"2_0_14","SDK":"4.4.6.240105","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.0","CR":"427/699"}}
16:00:06.523 MQT: stat/tasmota_bridge/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["google_diogo",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0020C009","2805C80001000600003C5A0A192800000000","40000080","00006000","00004002","00000000"]}}
16:00:06.562 MQT: stat/tasmota_bridge/STATUS4 = {"StatusMEM":{"ProgramSize":1635,"Free":1244,"Heap":69,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","8F9AC7CF","00148001","000000CF","010013C0","C0000981","40004080","00200000","5400082C","40000080","00000004"],"Drivers":"1,2,3,!4,!5,7,!8,9,10,12,!16,!20,!21,!24,26,27,29,!35,38,50,52,!59,62,!68,79,82,85,!121","Sensors":"1,2,3,5,6,52,62,127","I2CDriver":"7"}}
16:00:06.618 MQT: stat/tasmota_bridge/STATUS5 = {"StatusNET":{"Hostname":"tasmota-bridge-5936","IPAddress":"192.168.86.71","Gateway":"192.168.86.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.86.1","DNSServer2":"0.0.0.0","Mac":"A8:42:E3:54:17:30","IP6Global":"fd83:d468:8014:e01f:aa42:e3ff:fe54:1730","IP6Local":"fe80::aa42:e3ff:fe54:1730%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.86.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
16:00:06.682 MQT: stat/tasmota_bridge/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.86.249","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_541730","MqttUser":"MQTT","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
16:00:06.712 MQT: stat/tasmota_bridge/STATUS7 = {"StatusTIM":{"UTC":"2024-03-06T15:00:06Z","Local":"2024-03-06T16:00:06","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"07:20","Sunset":"18:41"}}
16:00:06.732 MQT: stat/tasmota_bridge/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0,0,0,0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
16:00:06.757 MQT: stat/tasmota_bridge/STATUS10 = {"StatusSNS":{"Time":"2024-03-06T16:00:06","Switch1":"OFF","Switch2":"OFF","ENERGY":{"TotalStartTime":"2024-03-05T17:16:42","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Voltage":236,"Current":[0.000,0.000],"BL09XX":{"Temperature":40.4}},"Shutter1":{"Position":90,"Direction":0,"Target":90,"Tilt":0},"TempUnit":"C"}}
16:00:06.803 MQT: stat/tasmota_bridge/STATUS11 = {"StatusSTS":{"Time":"2024-03-06T16:00:06","Uptime":"0T00:27:54","UptimeSec":1674,"Heap":74,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":39},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"google_diogo","BSSId":"CC:F4:11:55:F4:7F","Channel":1,"Mode":"11n","RSSI":48,"Signal":-76,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
16:00:46.942 MQT: stat/tasmota_bridge/RESULT = {"WebLog":4}
16:00:47.334 BL9: Rx 55F7020000000000000064113E490300000000000000000000000000020200FE030058
16:00:48.007 CFG: Saved, Count 45, Bytes 4096
16:00:48.364 BL9: Rx 558B0300000000000000CF083E940300000000000000000000000000020200FE030016
16:00:49.365 BL9: Rx 55110300000000000000B0073E790400000000000000000000000000020200FE0300CA
16:00:50.366 BL9: Rx 55E6020000000000000074EE3DA10300000000000000000000000000010200FE030026
16:00:51.366 BL9: Rx 55EF020000000000000029FE3D040300000000000000000000000000030200FE0300F3
16:00:52.367 BL9: Rx 5530040000000000000037033E310300000000000000000000000000020200FE030070
[... 27 lines cut ...]
16:01:17.375 BL9: Rx 55840300000000000000D3D23DB50300000000000000000000000000020200FE03002F
16:01:18.368 BL9: Rx 555803000E0000000000BBCF3D6F0300020000000000000000000000020200FE0300AC
16:01:19.377 BL9: Rx 55940300240000000000CBC83D3A0400070000000000000000000000020200FE030080
16:01:20.369 BL9: Rx 55F60300000000000000C2BC3D350400000000000000000000000000020200FE030063

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Connect the wiring and press the switch buttons (up or down)
[... cut ...]
External switch should be able to control the shutter

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
N/A

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
This topic started as an issue, and it was converted to a discussion. I am now converting it back to an issue, which is where I believe it belongs to, as the available configuration doesn't seem to be working

**(Please, remember to close the issue when the problem has been addressed)**
</div>

Jason2866, 2024-03-10:
Closing since there is #20886 and there is not verified to be an issue.
Please do not litter. Thx

closed 2024-03-10 by Jason2866

schrej-zz, 2024-04-03:
Just closing without having a solution is not a good behavior. The discussion did nit offer a solution.
I also have this issue with Sonoff DualR3 v2 and no real solution.