arendst/Tasmota #20945: I have a Problem with fade on CCT bulb.

KLLKP, 2024-03-13:
Hi to all here!

I use device with "SetOption92 1" (CCT - second PWM as Color Temp instead of Warm White) and it looks like fade command doesn't follow set SetOption92, effect is strange. 
Any ideas how to fix it?
Same about Fade when light is turned off and when turned on - strange effect. 
Strange effect means that during fade CT color is changing (not like using dimmer ruler only).
Tested at many values of speed. Don't know if fade follows DimmerRange too, can't judge it at now.

Two make my bulb works properly with tasmota web rulers (CT and dimmer)  i had to use SetOption92, change PWMs number (channels?) and invert one.

Template:
{"NAME":"Azzardo AZ3206 LightBulb","GPIO":[0,0,0,0,0,0,0,0,449,0,416,0,0,0],"FLAG":0,"BASE":18}
GPIO12: PWM_i - 2
GPIO14: PWM - 1

Commands:
SetOption92 1
Fade 1
Speed 2
DimmerRange 30,100

Device: Azzardo AZ3206 bulb with dimmer and color temp.
Board: Tuya TYWE2S 
ESP Chip Id: 12345371 (ESP8285N08)
Flash Size:	1024 KB
Program Size:	646 KB

Program Version	13.4.0(tasmota)
Build Date & Time	2024-02-14T16:13:56
Core/SDK Version	2_7_6/2.2.2-dev(38a443e)

Regards
Chris

KLLKP, 2024-03-14:
I investigated problem of fade a bit more... to better follow fade steps i used various speed (2-30).
1) when light is on: fade between CT Warm and Cool and back - works fine.
2) when temp is set to Cool {"POWER":"ON","Dimmer":100,"Color":"FF00","White":100,"CT":153,"Channel":[100,0]}
Fade in and out (power on/off) - works fine.
3) when temp is set to Warm {"POWER":"ON","Dimmer":100,"Color":"00FF","White":100,"CT":500,"Channel":[0,100]}
Fade in (power on) - works fine
Fade out (power off) - at the same time it race through dimm and colour - dimmer reduce brightnes (ok) but it also change temp from Warm to Cool at every fade steps (with pure cool at least).

So.. its all about fade out when CT is warm.

Ledtable on/off doesn't change that behavior.

KLLKP, 2024-03-23:
Hi,
I'v checked a bit more and... it works as should with tasmota-lite ver. 9.2.0 up to 12.5.0, since 13.0.0 it works bad.

Tested with official builds [https://ota.tasmota.com/]:
	tasmota-lite ver.  9.2.0 - OK :)
	tasmota-lite ver. 10.0.0 - OK :)
	tasmota-lite ver. 12.0.2 - OK :)
	tasmota-lite ver. 12.5.0 - OK :)
	tasmota-lite ver. 13.0.0 - BAD :( ..dimm with color race!
	tasmota-lite ver. 13.1.0 - BAD :( ..dimm with color race!
	tasmota-lite ver. 13.4.0 - BAD :( ..dimm with color race!

How to reproduce:
Needed: 
	Light bulb with Dimmer + CT (SO92: Cold/Warm white - second PWM as Color Temp instead of Warm White)
	{"NAME":"Azzardo AZ3206 LightBulb","GPIO":[0,0,0,0,0,0,0,0,449,0,416,0,0,0],"FLAG":0,"BASE":18}
Initial commands:
	SetOption92 1
	Fade 1
	Speed 8
	DimmerRange 30,100
Commands to prepare a test:
     Power1 1
     Dimmer 100
     CT 500
Run test with:
     Power1 0
Symptom:
     Instead of just dimm out it: dimm with changing color from warm to cold white.

Checked on 10 bulbs.

KLLKP, 2024-03-24:
Anyone read this? Something is missing here or unclear? Any details should be added?

s-hadinger, 2024-03-24:
Yes. Thanks for the details. Since it used to work fine, it will be easier to fix. I will work on it

KLLKP, 2024-03-25:
> Yes. Thanks for the details. Since it used to work fine, it will be easier to fix. I will work on it

Thanks.... its good to hear ;)
If You have any question.. just ask.
You can look into LightAnimate() [xdrv_04_light.ino], there probobly shlould be handled SO92 somewhere after call calcGamma.... There is solution for ChannelWhite_when_PWMCT, maybe its good place for fix.

s-hadinger, 2024-04-12:
I think the faulty commit is https://github.com/arendst/Tasmota/commit/9d57a19be3b2d48acb2885dd8435cd0bc204b029

closed 2024-04-12 by s-hadinger

KLLKP, 2024-04-18:
Hi 
I can confirm that problem is solved, good job.
Thank You.

PR #21159 (2024-04-12): Fade out on CCT bulb with `SO92 1`
## Description:

Fixes regression introduced in https://github.com/arendst/Tasmota/commit/9d57a19be3b2d48acb2885dd8435cd0bc204b029

The `ct_10` must not be adjusted according to brightness.

**Related issue (if applicable):** fixes #20945

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.6
  - [x] The code change is te [... cut]