arendst/Tasmota #20970: Disable LYWSD03MMC

darkman7hr, 2024-03-17:
Is there any way to disable LYWSD03MMC entirely?
In documentation there is "NRFUse", but it's not longer available.

barbudor, 2024-03-17:
What problem are you looking to solve ?

darkman7hr, 2024-03-17:
My ESP32-C3 Supermini is rebooting often, so I'm in process of diagnosing.

barbudor, 2024-03-17:
Look at https://tasmota.github.io/docs/Commands/#ble-esp32

There's a mi32block command
Never used it but may suis your need
You'll have to block them 1 by 1

darkman7hr, 2024-03-17:
I saw that, but scans will be made in background I guess. If I disable BLE, no reboots. I have one GAS sensor on pin analog 1 but his digital is connected to 5, maybe that's causing some issue.. 
Thank you anyhow.. maybe that NRFUse should be purged from code and documentation, it's confusing.