arendst/Tasmota #20976: BME680 shows 100% humidity all the time

peterdeak4, 2024-03-18:
### PROBLEM DESCRIPTION
I use a BME680 with Sonoff S55 and Tasmota outside since end of 2022. Yesterday, it started to show these and these only, no change:
```
BME680 Temperature: 34.6 °C
BME680 Humidity: 100.0 %
BME680 Humidity: 34.6 °C
BME680 Pressure: 733.1 hPa
BME680 Gas: 0.00 kΩ
```
(I'm in the middle of Europe so the real temperature was around 5C yesterday evening)
When I realised this and restarted the device, the data was normal for a few seconds and then back to the above. I tried to restart, upgrade to lates Tasmota (13.4.0 sensors), set gpio1,3 to `None`, and back to correct values, but the data is the same.

Is it a dead broke sensor after a year or is there anything I can do?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff S55
- [X] Tasmota binary firmware version number used: 13.4.0 sensors
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: OTA
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
10:49:07.777 MQT: stat/konnektor-udvar/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
10:49:07.991 MQT: stat/konnektor-udvar/RESULT = {"Module":{"8":"Sonoff S2X"}}
10:49:08.197 MQT: stat/konnektor-udvar/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"640":"I2C SDA"},"GPIO2":{"0":"None"},"GPIO3":{"608":"I2C SCL"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
10:49:47.995 MQT: stat/konnektor-udvar/STATUS = {"Status":{"Module":8,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"konnektor-udvar","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
10:49:48.000 MQT: stat/konnektor-udvar/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-sensors.bin.gz","RestartReason":"Software/System restart","Uptime":"0T04:07:52","StartupUTC":"2024-03-18T05:41:55","Sleep":50,"CfgHolder":4617,"BootCount":37,"BCResetTime":"2020-11-30T18:52:02","SaveCount":445,"SaveAddress":"FA000"}}
10:49:48.005 MQT: stat/konnektor-udvar/STATUS2 = {"StatusFWR":{"Version":"13.4.0(sensors)","BuildDateTime":"2024-02-14T16:14:20","Boot":31,"Core":"2_7_6","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285N08","CR":"392/699"}}
10:49:48.010 MQT: stat/konnektor-udvar/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Doug",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A002800000000","000000C0","00206000","00004000","00000000"]}}
10:49:48.029 MQT: stat/konnektor-udvar/STATUS4 = {"StatusMEM":{"ProgramSize":712,"Free":288,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14325E","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","8F9A8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","54000020","0008C081","00000000"],"Drivers":"1,2,!3,!4,!5,!6,7,!8,9,10,12,!14,!16,!17,!20,!21,!24,29,!34,62,!65,!66,!68","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,98,103,105","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,69,76"}}
10:49:48.037 MQT: stat/konnektor-udvar/STATUS5 = {"StatusNET":{"Hostname":"konnektor-udvar","IPAddress":"192.168.4.170","Gateway":"192.168.4.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.4.1","DNSServer2":"0.0.0.0","Mac":"9C:9C:1F:8C:97:C0","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
10:49:48.043 MQT: stat/konnektor-udvar/STATUS6 = {"StatusMQT":{"MqttHost":"*****","MqttPort":8883,"MqttClientMask":"konnektor-udvar","MqttClient":"konnektor-udvar","MqttUser":"****","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
10:49:48.049 MQT: stat/konnektor-udvar/STATUS7 = {"StatusTIM":{"UTC":"2024-03-18T09:49:48Z","Local":"2024-03-18T10:49:48","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"06:56","Sunset":"18:59"}}
10:49:48.057 MQT: stat/konnektor-udvar/STATUS10 = {"StatusSNS":{"Time":"2024-03-18T10:49:48","BME680":{"Temperature":34.6,"Humidity":100.0,"DewPoint":34.6,"Pressure":733.1,"Gas":0.00},"PressureUnit":"hPa","TempUnit":"C"}}
10:49:48.063 MQT: stat/konnektor-udvar/STATUS11 = {"StatusSTS":{"Time":"2024-03-18T10:49:48","Uptime":"0T04:07:53","UptimeSec":14873,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"****","BSSId":"****","Channel":6,"Mode":"11n","RSSI":36,"Signal":-82,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Use my BME680 sensor :)

### EXPECTED BEHAVIOUR
Correct data.

### SCREENSHOTS

### ADDITIONAL CONTEXT
Is it a dead broke sensor after a year or is there anything I can do?

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-03-18:
The stronger indication is the bad pressure (unless you're high in mountains).
`BME680 Pressure: 733.1 hPa`
That's no Tasmota issue, but the sensor misbehaving. Beware that "outside" incurs the risk of condensation physically corroding the sensor. That's my experience with having similar BME280 outside.

Another risk factor is if the sensor power situation is not good and stable. Risk factors include hardware being unsuitable (as you use a device not designed to extend with further stuff), degraded hardware, or maybe a temporary fluke. It might help power cycling it.

peterdeak4, 2024-03-18:
> The stronger indication is the bad pressure (unless you're high in mountains). `BME680 Pressure: 733.1 hPa` That's no Tasmota issue, but the sensor misbehaving. Beware that "outside" incurs the risk of condensation physically corroding the sensor or connections. That's my experience with having similar BME280 outside.
> 
> Another risk factor is if the sensor power situation is not good and stable. Risk factors include hardware being unsuitable (as you use a device not designed to extend with further stuff), degraded hardware, or maybe a temporary fluke. It might help power cycling it.

Thank you, I thought about the corroding, because before BME680 I used a cheap BMP280 and that was corroded but BME680 seems good.
Ok, I think I will remove the sensor and collect data from some weather forecast site, if it is not recommended to use sensor with sonoff s55.

sfromis, 2024-03-18:
It was not specifically about the Sonoff S55, but about risk factors with commercial devices intended for a specific purpose, with an often already penny-pinched power budget, and then seeing them as a platform to just add more stuff to. And BME680 is not among the sensors with the lowest power draw.