arendst/Tasmota #20993: Shelly Plus 1PM no reaction to KNX Telegrams but sending Status

Chri5chu, 2024-03-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic):Shelly Plus 1 PM
- [x] Tasmota binary firmware version number used: 13.4.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here: 
{"NAME":"Shelly Plus 1PM","GPIO":[0,0,0,0,192,2720,0,0,0,0,0,0,0,0,2656,0,0,0,0,2624,0,32,224,0,0,0,0,0,4736,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Updating to the latest binary after Flashing and using Safeload
Migrate with Partition Wizard
Connect to KNX Group Adresses

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Switch the Output after sending a telegram via ETS

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
As I said if i toggle via the Webinterface i get a Status and the cyclic power status is sended too.
But if I'm trying to switch the Output via KNX there is no reaction. What did i wrong? Btw I checked Filter Tables multiple times there is everything fine. Not even a read request is working.

**(Please, remember to close the issue when the problem has been addressed)**

Chri5chu, 2024-03-19:
Updatet to 13.4.0.3

barbudor, 2024-03-19:
[image]

Where are the logs ?

Chri5chu, 2024-03-19:
23:18:44.856 SRC: WebGui from 192.168.178.174
23:18:44.865 KNX: Output 1 = 1 sent to 6/2/14
23:18:44.868 RSL: RESULT = {"POWER":"ON"}
23:18:44.869 RSL: POWER = ON
23:18:45.123 HLW: EPC 16, CFlen 31852 usec
23:18:45.807 CFG: Saved, Count 50, Bytes 4096
23:18:46.123 HLW: EPC 6, CFlen 365665 usec
23:18:47.123 HLW: EPC 7, CFlen 137218 usec
23:18:48.123 HLW: EPC 11, CFlen 83618 usec
23:18:48.906 WIF: Checking connection...
23:18:49.123 HLW: EPC 12, CFlen 83347 usec
23:18:50.123 HLW: EPC 12, CFlen 90611 usec
23:18:51.123 HLW: EPC 11, CFlen 91732 usec
23:18:52.123 HLW: EPC 11, CFlen 97935 usec
23:18:53.123 HLW: EPC 10, CFlen 93767 usec
23:18:54.123 HLW: EPC 11, CFlen 98577 usec
23:18:55.123 HLW: EPC 10, CFlen 102928 usec
23:18:56.123 HLW: EPC 9, CFlen 102538 usec
23:18:57.123 HLW: EPC 11, CFlen 87824 usec
23:18:58.123 HLW: EPC 11, CFlen 87561 usec
23:18:59.123 HLW: EPC 10, CFlen 201605 usec
23:18:59.330 SRC: WebGui from 192.168.178.174
23:18:59.339 KNX: Output 1 = 0 sent to 6/2/14
23:18:59.345 RSL: RESULT = {"POWER":"OFF"}
23:18:59.346 RSL: POWER = OFF
23:18:59.759 CFG: Saved, Count 51, Bytes 4096
23:19:00.123 HLW: EPC 1, CFlen 204370 usec
23:19:08.883 WIF: Checking connection...
23:19:28.897 WIF: Checking connection...
23:19:48.943 WIF: Checking connection...

Chri5chu, 2024-03-19:
Thats what I get when I'm changing the switch via the web interface and then after that I sent the same telegram via KNX. My Network is just a regular FritzBox Router and a switch without managing.

barbudor, 2024-03-19:
If you were actually receiving something you should have logs such as `KNX: Received from X/Y/Z Command A: B to C`
With real values instead of X, Y, Z and A, B, C

Likely you have somehting wrong in your configuration. May be on ETS.
You can reduce the number of logs using level 3. No need for 4 here.

Chri5chu, 2024-03-20:
I thought so too at First and checked again. Even reprogrammed the ip Router. But still my Shelly1 is working Perfect. If I Switch via the Webinterface I get the Status on 6/2/14. If I sent a telegram on 6/2/14 which must be routet correctly there is nothing happening and on 6/0/14 neither. I really don't get what I did wrong here.

barbudor, 2024-03-20:
KNX protocol is using multicast UDP which is based on IGMP
Not all AP can forward multicast to Wifi
Do you have other KNX devices that are connected by WiFi ?
Is you ETS host connected by ethernet or wifi ?
Check in your router or AP is you have any setting related to IGMP that would need to be enabled

Chri5chu, 2024-03-20:
Thanks for the help. I'm checking if there is something blocked in the Router. ETS is connected via Wifi and the working shelly1 is also connected by wifi.

barbudor, 2024-03-20:
That's interresting info
So ETS over wifi can communicate both ways with a Shelly 1. I assume the Shelly 1 is connected to the same AP than the Shelly 1PM
Both are ESP8266 correct ? 

Sounds more like a configuration issue

Chri5chu, 2024-03-20:
It's mesh an After I looked at it, they are Not on the Same Hardware connected actually. Didn't think about it so far. The 1Pm is a Plus an so it's an esp32.

Noschvie, 2024-03-20:
> ETS is connected via Wifi and the working shelly1 is also connected by wifi.

your ETS application is connected via Wifi (and a tunnel) with your KNX IP router or interface device, but your Shelly communicates via Wifi and multicast with your KNX router oder KNXd, isn't it?
Do you have multicast enabled for your wifi?

If needed I can test with a Shelly plus 1 this weekend.

Chri5chu, 2024-03-20:
Got it. DHCP Server was not configured for the IP Adresses I've used. Sorry for disturbing you and many thanks for the help.

closed 2024-03-20 by Chri5chu

shivan, 2024-03-27:
@Chri5chu what exactly do you mean by "DHCP Server was not configured for the IP addresses"?

I'm currently having the problem that a message from KNX TP is sent, but not received by tasmota. On another tasmota device, it works. I tried watching multicast packages on MCAST_GRP = '224.0.23.12' and MCAST_PORT = 3671. But does not seem to get any, even on working devices. How exactly did you solve your problem?

Chri5chu, 2024-03-27:
The IP Adress of my device was 192.168.178.91 my dhcp Server was set to start at 192.168.178.100 so Communication was working but Not correctly. After dhcp to .75 everything was working Fine. Did you Check in Logs if the Telegramm is registrated by the Shelly?

Chri5chu, 2024-03-27:
@shivan is it the Same group adress or another? Your Problem Sounds a Bit Like a Filter table Problem.

shivan, 2024-03-28:
@Chri5chu it's another group address. I restarted the Tasmota device and then it worked. Before doing the restart, there was nothing in the logs of the tasmota device that it got any knx info. I thought it completely works over udp multicast? Tried to watch on my computer for multicast packages, but did not work however...

Tried this script:
```python
import socket
import struct

MCAST_GRP = '224.0.23.12'
MCAST_PORT = 3671

sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM, socket.IPPROTO_UDP)
sock.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)

sock.bind(('', MCAST_PORT))

mreq = struct.pack("4sl", socket.inet_aton(MCAST_GRP), socket.INADDR_ANY)
sock.setsockopt(socket.IPPROTO_IP, socket.IP_ADD_MEMBERSHIP, mreq)

print("Warte auf KNX IP Multicast-Nachrichten...")

try:
    while True:
        data, addr = sock.recvfrom(1024)
        print(f"Nachricht von {addr}: {data.hex()}")
except KeyboardInterrupt:
    print("Programm beendet")
finally:
    sock.close()
```

Although it would be better for traffic, when not using multicast, but so I currently do not understand how the messages for KNX IP work.

Using xknx and connecting to my knx ip router directly, I saw the commands sent to tasmota. (but not received by tasmota)

Device is using Tasmota 13.4.0

When it would be a filter problem, then I would not have it gotten by xknx and a restart would not have been helped I think. (Although I saw a filter option in my kxn ip router - but that was not yet configurable)

Chri5chu, 2024-03-28:
@shivan Strange but Great that it worked for you. Knx is definetly using Multicast. I couldn't visualise the Traffic either.