arendst/Tasmota #21000: SonoffZigbeeBridge in zigbee2mqtt addon Home Assistant

igori81, 2024-03-20:
### PROBLEM DESCRIPTION
Good afternoon I flashed SonOff ZigBee in Tasmota, flashed ZigBee. Everything in ZHA works. But I need to use the ZigBee2MQTT add-on and it doesn’t work there. The addon complains about the lack of a device. Please tell me what am I doing wrong?
[log.txt](https://github.com/arendst/Tasmota/files/14664804/log.txt)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
backlog rule1 on system#boot do TCPStart 8888 endon ; rule1 1 ; template {"NAME":"Sonoff ZHABridge","GPIO":[56,208,0,209,59,58,0,0,0,0,0,0,17],"FLAG":0,"BASE":18} ; module 0

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

igori81, 2024-03-20:
Forgot to specify the z2m config
`mqtt:
  server: mqtt://core-mosquitto:1883
  base_topic: zigbee2mqtt_kv_1
  version: 5
  user: addons
  password: Eet4uigh3Ith3AhqueiceL8quai4iepeiNai7Re4can8Ahquua9ve2iol3cheera
serial:
  port: tcp://192.168.1.82:8888
  baudrate: 115200
  disable_led: false
advanced:
  transmit_power: 20
  channel: 15
  pan_id: 18930
  network_key:
    - 59
    - 6
    - 183
    - 59
    - 31
    - 32
    - 244
    - 123
    - 165
    - 40
    - 150
    - 62
    - 123
    - 58
    - 181
    - 176
  availability_blocklist: []
  availability_passlist: []
  last_seen: ISO_8601
  log_level: info
homeassistant: true
permit_join: false
frontend:
  port: 8099
`

s-hadinger, 2024-03-20:
You are on the wrong forum. Tasmota has no active role with Zigbee2MQTT or ZHA, except passing opaque bytes. Please revert to Zigbee2MQTT community

closed 2024-03-20 by Jason2866

PhillyGilly, 2024-03-29:
> Forgot to specify the z2m config
>
> ```
> mqtt:
>   server: mqtt://core-mosquitto:1883
>   base_topic: zigbee2mqtt_kv_1
>   version: 5
>   user: addons
>   password: [...]

If you check my github I have documented how I got it working.

igori81, 2024-04-01:
> > Forgot to specify the z2m config
  I forgot to specify the adapter type (adapter: ezsp), everything worked with it. Thank you!