arendst/Tasmota #21012: SML / counter scripting: Bug when excess # is present - plus remedy

hagenbuch, 2024-03-22:
### PROBLEM DESCRIPTION

I am using Tasmota 13.2. on a device connected to an electricity counter via infrared communication. For documentation, I am using "Hichis" hardware and his precompiled binaries from

https://www.youtube.com/watch?v=RbyXqJiBC8Q

When I had version 11.1, this script ran flawlessly:

```
>D
lief=0
bez=0
zz=""
#
>B
=>sensor53 r
tper=60
>T
bez=WUA#bezug
lief=WUA#liefer
zz=WUA#zaehlerid
>S
if upsecs%tper==0
then
print zz: %zz% lief:: %lief% bez: %bez% atime: %tstamp% up: %uptime%

=> WebSend [example.com] /?id=myid&atime=%tstamp%&uptime=%uptime%&bezug=%bez%&lieferung=%lief%
endif
>M 1
+1,3,s,0,9600,WUA
1,77070100600100ff@#,Zählernummer,,zaehlerid,0
1,77070100010800ff@1,Bezug,Wh,bezug,0
1,77070100020800ff@1,Lieferung,Wh,liefer,0
#
```
Watch out for the "#" in the >D section! A time ago it took me several days to find out that at that moment in time, the # had been necessary, just as the last # is today!

However, when updating to 13.2. , the update went great but there was no more script functionality, the console showed weird input commands like "bRY" that naturally could not be understood by Tasmota.

Also, the command "sensor53" was not understood anymore!

Solution:

As soon as I deleted the first # in the >D section, everything works again.

It would be great if we could add either an error message (# not understood or the like) or make the software more resilient to those syntax problems.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [X] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
11:06:07.187 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"wua-e220-main","FriendlyName":["wua-e220-main"],"Topic":"tasmota_56FA33","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
11:06:07.192 RSL: STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T20:37:22","StartupUTC":"2024-03-21T13:28:45","Sleep":50,"CfgHolder":4617,"BootCount":32,"BCResetTime":"2022-09-21T20:41:40","SaveCount":446,"SaveAddress":"F7000"}}
11:06:07.196 RSL: STATUS2 = {"StatusFWR":{"Version":"13.1.0(tasmota)","BuildDateTime":"2024-02-10T17:23:12","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"383/699"}}
11:06:07.201 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["dellekom","wlan@dellekom.de"],"TelePeriod":60,"Resolution":"548980C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
11:06:07.210 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":621,"Free":380,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14325E","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","979A0787","04180001","0000008F","010013C0","00000008","00004000","00000000","04000020","00000000"],"Drivers":"1,2,3,4,5,7,8,9,10,11,12,20,21,62,68","Sensors":"1,2,3,6,53","I2CDriver":"7"}}
11:06:07.215 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-56FA33-6707","IPAddress":"192.168.22.37","Gateway":"192.168.22.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.22.1","DNSServer2":"0.0.0.0","Mac":"C8:C9:A3:56:FA:33","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
11:06:07.218 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_56FA33","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
11:06:07.224 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-03-22T10:06:07","Local":"2024-03-22T11:06:07","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"06:47","Sunset":"19:05"}}
11:06:07.229 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-03-22T11:06:07","WUA":{"zaehlerid":"0a014c475a00032fd4f2","bezug":85082090,"liefer":33934626}}}
11:06:07.234 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-03-22T11:06:07","Uptime":"0T20:37:22","UptimeSec":74242,"Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":2,"SSId":"wlan@dellekom.de","BSSId":"C0:25:E9:40:8A:76","Channel":11,"Mode":"11n","RSSI":100,"Signal":-26,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE

Include an additional # to the end of the >D section is a script.

### EXPECTED BEHAVIOUR

Either Tasmota should ignore the now obsolete # or it should complain in an error message

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2024-03-22:
this # in >D was never required ! 
it is not intended to add more checks for syntax errors since on such a small machine (especially esp8266)  it would add too much code.
you are free to add remarks to the docs what may cause such weird behavior

hagenbuch, 2024-03-22:
Thank you! I know that the # in >D had not been required but I could not get the script to run in 11.1 before I added the #!

I'll take a look into adding a remark in the docs as you suggested.

Sometimes removing code can improve code :)

closed 2024-04-03 by Jason2866