arendst/Tasmota #21019: Can not subscribe to domoticz/out/xxx MQTT topic

soif, 2024-03-24:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Subscribing to the domoticz/out/ MQTT topic don't work

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic):  **nodemcu** and **wemos-d1-mini**
- [x] Tasmota binary firmware version number used: **13.4.0.3, latest develop commit: 7088ee2be64dd475ecc4fd30af244158ad2e869a**
  - [ ] Pre-compiled
  - [x] Self-compiled **with SUPPORT_MQTT_EVENT + RULE_MAX_MQTT_EVENTSZ  512**
- [x] Flashing tools used: [espbuddy](https://github.com/soif/EspBuddy) (an esptool.py wrapper)
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

09:24:51.947 MQT: stat/tasmota_5A440E/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
09:24:52.176 MQT: stat/tasmota_5A440E/RESULT = {"Module":{"1":"Sonoff Basic"}}
09:24:52.430 MQT: stat/tasmota_5A440E/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

09:25:32.168 MQT: stat/tasmota_5A440E/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":193,"Free":318,"Rules":"ON mqtt#connected DO Backlog Subscribe test1,domoticz/out/1000,svalue1; Subscribe test2,domoti/out/1000,svalue1 ENDON ON event#test1 DO Power1 toggle ENDON ON event#test2 DO Power1 toggle ENDON"}}
09:25:32.399 MQT: stat/tasmota_5A440E/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
09:25:32.649 MQT: stat/tasmota_5A440E/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

09:26:48.978 MQT: stat/tasmota_5A440E/STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_5A440E","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
09:26:48.991 MQT: stat/tasmota_5A440E/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:15:42","StartupUTC":"2024-03-24T08:11:06","Sleep":50,"CfgHolder":4617,"BootCount":2,"BCResetTime":"2024-03-24T09:10:17","SaveCount":8,"SaveAddress":"F4000"}}
09:26:49.001 MQT: stat/tasmota_5A440E/STATUS2 = {"StatusFWR":{"Version":"13.4.0.3(tasmota)","BuildDateTime":"2024-03-24T09:07:44","Boot":31,"Core":"2_7_6","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"374/699"}}
09:26:49.009 MQT: stat/tasmota_5A440E/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["iot",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
09:26:49.028 MQT: stat/tasmota_5A440E/STATUS4 = {"StatusMEM":{"ProgramSize":646,"Free":356,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,!3,!4,!5,!6,7,!8,9,10,12,!16,!18,!19,!20,!21,!22,!24,26,!27,29,!30,!35,!37,!45,62,!68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
09:26:49.044 MQT: stat/tasmota_5A440E/STATUS5 = {"StatusNET":{"Hostname":"tasmota-5A440E-1038","IPAddress":"10.1.209.241","Gateway":"10.1.11.1","Subnetmask":"255.255.0.0","DNSServer1":"10.1.10.1","DNSServer2":"10.1.11.1","Mac":"5C:CF:7F:xx:xx:xx","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
09:26:49.053 MQT: stat/tasmota_5A440E/STATUS6 = {"StatusMQT":{"MqttHost":"10.1.12.114","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5A440E","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
09:26:49.067 MQT: stat/tasmota_5A440E/STATUS7 = {"StatusTIM":{"UTC":"2024-03-24T08:26:49Z","Local":"2024-03-24T09:26:49","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"06:43","Sunset":"19:08"}}
09:26:49.076 MQT: stat/tasmota_5A440E/STATUS10 = {"StatusSNS":{"Time":"2024-03-24T09:26:49"}}
09:26:49.085 MQT: stat/tasmota_5A440E/STATUS11 = {"StatusSTS":{"Time":"2024-03-24T09:26:49","Uptime":"0T00:15:43","UptimeSec":943,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"iot","BSSId":"8A:C3:97:xx:xx:xx","Channel":6,"Mode":"11n","RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

09:29:55.555 HTP: Console

      (sending test1) ----> No Output

     (sending test2) ---->
09:30:17.187 RUL: EVENT#TEST2 performs "Power1 toggle"
09:30:17.189 SRC: Rule
09:30:17.191 CMD: Grp 0, Cmd 'POWER', Idx 1, Len 6, Pld 2, Data 'toggle'
09:30:17.197 MQT: stat/tasmota_5A440E/RESULT = {"POWER":"OFF"}
09:30:17.201 MQT: stat/tasmota_5A440E/POWER = OFF
09:30:17.333 CFG: Saved to flash at FA, Count 10, Bytes 4096
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

(From the mosquitto server console)

Test 1 (tasmota ignore it) :
`mosquitto_pub  -t 'domoticz/out/1000' -m '{"svalue1": "40"}'`

Test 2 (tasmota recognize it properly)
`mosquitto_pub  -t 'domoti/out/1000' -m '{"svalue1": "40"}'`

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Tasmota should answer to domoticz/out/IDX

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

As advised by @arendst [here](https://github.com/arendst/Tasmota/issues/20299#issuecomment-1868300849), I've switched domoticz to output mqtt messages by **Index**, to be sure that mqtt message are not too big... Anyway, using the (previous) very short mosquitto_pub payloads, produce the same results.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2024-03-24:
As designed, ``domoticz/out`` (or any user defined DOMOTICZ_OUT_TOPIC) overrules the rule subscribe functionality. If you want to act on ``domotics/out/idx`` you will need to configure the index for the relay in the tasmota domoticz GUI config screen or use command ``DzIdx<relay number> <domoticz index>``.

soif, 2024-03-24:
Hi thanks for the fast answer 😎 

I've put (above) some very basic rules (to ease fast understanding of the issue) but in fact, **i want to grab some sensor values from Domoticz** (ie temperature, humidity, wind speed etc....) and  then display them on an Tasmota powered Oled display. 
So _DzIdx_ is not relevant in this case. 

When values come from other Tasmotas, it's work perfectly well by subscribing to tele/otherTasmota/sensor , but for others sensors (ie: RF, MySensors, EspEasy, other non MQTT sensors,etc...) I've no other choice than using the domoticz/out topic to grab them.

_(I know I could always make a custom domoticz script to do that, but its' a bit overkill for such simple tasks)_

Being supported directly in Tasmota, would be a lot easier. (if to complicated to change the code to receive domoticz/out both for relays AND for subscribing, it could also be implemented a compilation flag that switch esbetween one of the 2 behaviours)

Makes sense?

arendst, 2024-03-24:
In that case compile your own binary disabling Domoticz and you're free to use subscribe as you like.

soif, 2024-03-30:
will do. thanks for the tips. 👍 

> As designed, `domoticz/out` (or any user defined DOMOTICZ_OUT_TOPIC) overrules the rule subscribe functionality. 

This information would actually  deserve to be noted in the documentation at :
https://tasmota.github.io/docs/MQTT/#subscribe

Would you mind if i submit to this page something like: 

>!!! tip "Domoticz Users"
>    As designed, `domoticz/out` (or any user defined DOMOTICZ_OUT_TOPIC) overrules the rule subscribe functionality. This means that you can't subscribe to this topic, except if you had disabled the `USE_DOMOTICZ` flag when compiling your own firmware

Dylantje, 2024-03-30:
mmm is there perhaps someone that have a simple how to.. windows noob here.
Will use the 8 channels, with domoticz, only perhaps can i work with mqtt or node red..?

sfromis, 2024-03-30:
Last question related to https://github.com/arendst/Tasmota/discussions/21066

arendst, 2024-03-31:
Let's keep the conversation here where I will try to add functionality to allow users to use ``subscribe`` to DOMOTICZ_OUT_TOPIC extending the pre-compiled domoticz functionality.

arendst, 2024-03-31:
So try latest dev branch.

This allows to configure additional relays5 until 32 with domoticz command ``DzIdx<relay> <domoticz_idx>``.

To make the relation persistent add following rule where 403 and 1234 are known Domoticz idx:
```
rule1 on system#boot do backlog DzIdx5 403; DzIdx6 1234 endon
```

soif, 2024-03-31:
> This allows to configure additional relays5 until 32 with domoticz command DzIdx<relay> <domoticz_idx>.

Many Thanks for this smart addition. While it doesn't change anything, about the initial question to subscribe to Dz sensors, it will be a very helpful features for project (like a couple of mine, but certainly a lot of other users too) when we need to handle more than 4 "relays", easily. 😎 👍 

Regarding my proposal of amending the documentation regarding Dz/out subscribe limitations, I'm gonna pull a PR, for you to decide to accept or reject it.

Dylantje, 2024-03-31:
THANKS for the great work...
Likt to test, only is there ota link for upgrade to the relaease?

sfromis, 2024-03-31:
You do not need to wait for next release, fresh development builds are here:
https://ota.tasmota.com/tasmota/
And you can copy/paste from the "OTA URL" column for an upgrade.
(Do not copy the https links from the "OTA Firmware" column).

Dylantje, 2024-03-31:
this are teh realese?
Not the development ota, or am i wrong

sfromis, 2024-03-31:
No, the link I posted is development, and the timestamp shows how fresh they are. Also, the header clearly says "Development binaries".

If you wanted release instead: https://ota.tasmota.com/tasmota/release/

Dylantje, 2024-04-01:
> ```
> rule1 on system#boot do backlog DzIdx5 8077; DzIdx6 8078endon
> rule1 1
> ```
> 
> That's broken. First, you need a space before `endon`, otherwise you have no rule. Second, you cannot copy/paste independent commands in one operation, but have to issue two separate commands. As you only used one copy/paste, your output shows the bad result.
> 
> > 10:14:59.479 MQT: stat/homeassistant/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":63,"Free":448,"Rules":"**on system#boot do backlog DzIdx5 8077; DzIdx6 8078endon rule1 1**"}}

sorry, first time using rules....

So this should be good?
rule1 on system#boot do backlog DzIdx5 8077 endon rule1 1

Perhaps otherwise an good example rule, please

arendst, 2024-04-01:
I suggest you start to read here https://tasmota.github.io/docs/Rules/ and continue from there.

In short, this is the valid rule:
```
rule1 on system#boot do backlog DzIdx5 8077 endon
```
followed by a one-time command ``rule1 1`` enterd by you.

Dylantje, 2024-04-01:
> I suggest you start to read here https://tasmota.github.io/docs/Rules/ and continue from there.
> 
> In short, this is the valid rule:
> 
> ```
> rule1 on system#boot do backlog DzIdx5 8077 endon
> ```
> 
> followed by a one-time command `rule1 1` enterd by you.

WORKS!!!! only after a reboot.

Only now swiths 6:
rule2 on system#boot do backlog DzIdx5 8078 endon  [ enter]

rule2 2 [ enter ]?
Or rule2 1 ?

sfromis, 2024-04-01:
You need both DzIdx commands at the same time, just like what you originally had. No one was suggesting you to only have one of those, just to fix the errors in the rest of the line.

arendst, 2024-04-01:
No excuse not to to try to understand the docs. You managed Nodo in the past, Tasmota is no different.

Dylantje, 2024-04-01:
> No excuse not to to try to understand the docs. You managed Nodo in the past, Tasmota is no different.

nodo that was very hard to get for me.. A long time ago
And yes i did try that also, only then there was domoticz..:-) And then there was espeasy, and finaly tasmota :-)

Dylantje, 2024-04-01:
Did:
```
16:07:00.227 CMD: DzIdx0 0 
16:07:00.233 MQT: stat/Houthok_midden/RESULT = {"Command":"Error"}
```
Found something other stranges....
The relais are not working like they says in to the template explanation:
[image]
in this setting relais 6 and 8 are not working.

With this setting are all relais working:
[image]
When i use the webui....

After this cms:
`rule1 on system#boot do backlog DzIdx5 8078; DzIdx6 8105; DzIdx7 8080; DzIdx8 15822 endon `

Almost all the idx with domoticcz are woring only 6 not.
idx 8105 not working from domoticz, when i push the webui button working fine..

Try a lot of other settings, only  DzIdx6 8105;  is not working...

sfromis, 2024-04-01:
Using `DzIdx0 0` is not valid as you have no relay 0. While some commands will be for all indexes when using 0, this is per command, and not for the new `DzIdx` command.

Dylantje, 2024-04-02:
Is it not possible te set an switch to a other gpio?
I think there is something wrong in the template.

All working only relay 7 is not.. 
GPIO 4
[image]
All other gpio are not working, this is the only config that is working.
Strange is that with the rule code, the 7 is not working with a idx from domoticz, and in the webui from the wifiboard the swicht is working well.

Hope someone hava a idea.

closed 2024-04-03 by arendst