arendst/Tasmota #21047: Tasmota v13.4.0 - 2x PCF8574 Mode2 problem

jpier123, 2024-03-27:
Dear Tasmota Team,

I'm facing a problem with build v 13.4.0 and PCF8574 mode2.  Mode 2 under below template did work without any issues based on previous build: 13.2.0. After an upgrade, ESP reboots every 30 seconds. I suspect pcf8574.dat file syntax change since 13.2.0, which results in failed Mode2 validation and Mode 1 enforcement. See below Debug logs. 

Used pcf8574.dat file template: 
{"NAME":"PCF8574","BASE":1,"GPIO":[194,207,208,209,270,271,272,273,64,256,65,257,78,258,83,259]}

Backgroud:

I have two PCF8574 (adress: 0x21, 0x22) modules added and configured with 4 Inputs and 4 outputs each. 

my_user_config.h

#define PCF8574_INVERT_PORTS   true 
...
  #define USE_PCF8574                           
    #define USE_PCF8574_MODE2                    
//    #define USE_PCF8574_SENSOR                   
//    #define USE_PCF8574_DISPLAYINPUT           
//    #define USE_PCF8574_MQTTINPUT              
    #define PCF8574_ADDR1 0x20                   
    #define PCF8574_ADDR1_COUNT 4                
//    #define PCF8574_ADDR2 0x39                  
//    #define PCF8574_ADDR2_COUNT 6               

Symptopms: 

Invalid Button/Switch assigment makes ESP to force restart after couple seconds.

PCF8574 0x21

    Device 1 Port 0 = Input
    Device 1 Port 1 = Input
    Device 1 Port 2 = Input
    Device 1 Port 3 = Input
    Device 1 Port 4 = Output
    Device 1 Port 5 = Output
    Device 1 Port 6 = Output
    Device 1 Port 7 = Output

PCF8574 0x22

    Device 2 Port 0 = Input
    Device 2 Port 1 = Output
    Device 2 Port 2 = Input
    Device 2 Port 3 = Output
    Device 2 Port 4 = Input
    Device 2 Port 5 = Output
    Device 2 Port 6 = Input
    Device 2 Port 7 = Output

This issue template is meant to REPORT Tasmota software PROBLEMS ONLY

Please DO NOT OPEN AN ISSUE:
  - If your Tasmota version is not the latest from the development branch, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://ota.tasmota.com/tasmota/
  - If you have an issue when flashing was done via Tuya Convert, please address it to Tuya Convert Team
  - If your issue is a flashing issue, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
  - If your issue is compilation problem, please address it to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4)
  - If your issue has been addressed before (i.e., duplicated issue), please ask in the original issue
  - If your issue is a Wi-Fi problem or MQTT problem, please try the steps provided in the [FAQ](https://tasmota.github.io/docs/FAQ) and [Troubleshooting](https://tasmota.github.io/docs/Troubleshooting)

 Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

DO NOT DELETE ANY TEXT from this template! Otherwise the issue will be auto-closed.
-->

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: WebGUI
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:00.003 HDW: ESP32-D0WD-V3 v3.0 
00:00:00.164 UFS: FlashFS mounted with 288 kB free
00:00:00.194 CFG: Loaded from File, Count 12247
00:00:00.197 CFG: CR 587/699, Busy 0
00:00:00.210 I2C: Bus1 using GPIO22(SCL) and GPIO21(SDA)
00:00:00.212 I2C: DS3231 found at 0x68
00:00:00.217 RTC: UTC 2024-03-27T13:40:42Z, DST 2024-03-31T02:00:00, STD 2024-10-27T03:00:00
14:40:42.000 RTC: Synced by DS3231
14:40:42.006 I2C: PCF8574 found at 0x21
14:40:42.007 I2C: PCF8574 found at 0x22
14:40:42.067 PCF: Base 1, Template 'PCF8574'
14:40:42.084 TFS: File 'mcp23x.dat' not found
14:40:42.092 MCP: Invalid template
14:40:42.094 BTN: Add vButton1, State 1
14:40:42.095 BTN: Add vButton2, State 1
14:40:42.097 BTN: Add vButton15, State 1
14:40:42.098 BTN: Add vButton20, State 1
14:40:42.100 SWT: Add vSwitch3, State 1
14:40:42.102 SWT: Add vSwitch16, State 1
14:40:42.103 SWT: Add vSwitch17, State 1
14:40:42.104 SWT: Add vSwitch18, State 1
14:40:42.105 ROT: Mode 1
14:40:42.274 CFG: No '*.autoconf' file found
14:40:42.280 BRY: Berry initialized, RAM used 4255 bytes
14:40:42.306 BRY: No 'preinit.be'
14:40:42.309 I2C: LCD found at 0x27
14:40:42.410 DSP: LCD
14:40:42.426 SRC: Restart
14:40:43.286 Project ESP32 - ESPSMD Version 13.4.0(tasmota32)-2_0_14(2024-03-27T12:30:34)
14:40:43.288 SRC: Thermostat
14:40:43.300 RSL: RESULT = {"POWER2":"OFF"}
14:40:43.302 RSL: POWER2 = OFF
14:40:43.302 ETH: No ETH MDC and ETH MDIO GPIO defined
14:40:43.367 DSB: Sensors found 3
14:40:43.378 SHT: Sensor did not ACK command
14:40:43.398 iBeacon register for advert callbacks
14:40:44.433 SRC: Berry

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2024-03-27:
Please assign buttons, switches and Relais without holes.
So Relais 1 2 3, button 1 2 3 and switches 1 2 3. (Start offsets are allowed)
This is not valid and allowed.
```
14:40:42.095 BTN: Add vButton2, State 1
14:40:42.097 BTN: Add vButton15, State 1
14:40:42.098 BTN: Add vButton20, State 1
14:40:42.100 SWT: Add vSwitch3, State 1
14:40:42.102 SWT: Add vSwitch16, State 1
14:40:42.103 SWT: Add vSwitch17, State 1
14:40:42.104 SWT: Add vSwitch18, State 1
```

barbudor, 2024-03-27:
Do not use non consecutive index number

jpier123, 2024-03-28:
One extra note. I did get Mode 2 working like that with v12.4.0.3. There was event case #18232 opened that [arendst] managed help resolve. All what I'm saying is that the solution did work up to 13.2.0 build. I made straight 13.2.0 >> 13.4.0 firmware upgrade without any other file changes and the probblem did appear after first restart.

Jason2866, 2024-03-28:
There are many changes done in Tasmota when version number changes.
Forcing consecutive index number was one of them. Do not expect Tasmota behaves same when a major version number is increased.
So repeating, does it work when numbering is done as needed?

jpier123, 2024-03-28:
OK, Problem was resolved. 

Please note that your statement about consecutive index number is not quite true. 
Build 13.4.0 supports Mode 2 with non consecutive index number, as per my working pcf8574.dat template: 

                                                          A = Sn3, Sn16-18, Ri15-18,          B = Bn1Ri1, Bn2Ri2, Bn15Ri3, Bn20Ri20 
{"NAME":"PCF8574","BASE":1,"GPIO":[194,207,208,209,270,271,272,273,64,256,65,257,78,258,83,259]}

Debug log: 

00:00:00.003 HDW: ESP32-D0WD-V3 v3.0 
00:00:00.081 UFS: FlashFS mounted with 304 kB free
00:00:00.105 CFG: Loaded from File, Count 104
00:00:00.108 CFG: CR 590/699, Busy 0
00:00:00.121 I2C: Bus1 using GPIO22(SCL) and GPIO21(SDA)
00:00:00.123 I2C: DS3231 found at 0x68
00:00:00.128 RTC: UTC 2024-03-28T11:34:46Z, DST 2024-03-31T02:00:00, STD 2024-10-27T03:00:00
12:34:46.000 RTC: Synced by DS3231
12:34:46.017 I2C: PCF8574 found at 0x21
12:34:46.017 I2C: PCF8574 found at 0x22
12:34:46.055 PCF: Base 1, Template 'PCF8574'
12:34:46.065 TFS: File 'mcp23x.dat' not found
12:34:46.073 MCP: Invalid template
12:34:46.075 BTN: Add vButton1, State 1
12:34:46.076 BTN: Add vButton2, State 1
12:34:46.078 BTN: Add vButton15, State 1
12:34:46.079 BTN: Add vButton20, State 1
12:34:46.082 SWT: Add vSwitch3, State 1
12:34:46.094 SWT: Add vSwitch16, State 1
12:34:46.095 SWT: Add vSwitch17, State 1
12:34:46.096 SWT: Add vSwitch18, State 1
12:34:46.097 ROT: Mode 1
12:34:46.189 CFG: No '*.autoconf' file found
12:34:46.196 BRY: Berry initialized, RAM used 4255 bytes
12:34:46.209 BRY: No 'preinit.be'
12:34:46.212 I2C: LCD found at 0x27
12:34:46.313 DSP: LCD
12:34:46.323 SRC: Restart
12:34:47.221 Project ESP32 - ESPSMD Version 13.4.0(tasmota32)-2_0_14(2024-03-27T12:30:34)
12:34:47.223 SRC: Thermostat
12:34:47.235 RSL: RESULT = {"POWER2":"OFF"}
12:34:47.236 RSL: POWER2 = OFF
12:34:47.237 ETH: No ETH MDC and ETH MDIO GPIO defined
12:34:47.316 DSB: Sensors found 4
12:34:47.328 SHT: Sensor did not ACK command
12:34:47.348 iBeacon register for advert callbacks
12:34:47.690 SRC: Berry
12:34:47.692 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 80, Pld -99, Data 'WebButton17 17.FET3; WebButton18 18.FET4; WebButton19 19.PWM; WebButton20 20.LCD'
12:34:47.704 SRC: Berry
12:34:47.705 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 121, Pld -99, Data 'DS18RetryRead 1;CounterDebounceLow 5, SetOption79 1; Power20 1; Timezone 99; TimeStd 0,0,10,1,3,60; TimeDst 0,0,3,1,2,120'
12:34:47.891 SRC: Berry
12:34:47.892 CMD: Grp 0, Cmd 'DS18ALIAS', Idx 1, Len 19, Pld 490822, Data '490822B0E4304A28,T1'
12:34:47.901 RSL: RESULT = {"T1":{"Id":"490822B0E4304A28"},"DS18B20-2":{"Id":"93955C4B09646128"},"DS18B20-3":{"Id":"0A1F7A4B09646128"},"DS18B20-4":{"Id":"100822B108284F28"}}
12:34:47.954 SRC: Berry
12:34:47.956 CMD: Grp 0, Cmd 'DS18ALIAS', Idx 1, Len 19, Pld 93955, Data '93955C4B09646128,T2'
12:34:47.963 RSL: RESULT = {"T1":{"Id":"490822B0E4304A28"},"T2":{"Id":"93955C4B09646128"},"DS18B20-3":{"Id":"0A1F7A4B09646128"},"DS18B20-4":{"Id":"100822B108284F28"}}
12:34:48.016 SRC: Berry
12:34:48.018 CMD: Grp 0, Cmd 'DS18ALIAS', Idx 1, Len 19, Pld 0, Data '0A1F7A4B09646128,T3'
12:34:48.025 RSL: RESULT = {"T1":{"Id":"490822B0E4304A28"},"T2":{"Id":"93955C4B09646128"},"T3":{"Id":"0A1F7A4B09646128"},"DS18B20-4":{"Id":"100822B108284F28"}}
12:34:48.078 SRC: Berry
12:34:48.080 CMD: Grp 0, Cmd 'DS18ALIAS', Idx 1, Len 19, Pld 100822, Data '100822B108284F28,T4'
12:34:48.087 RSL: RESULT = {"T1":{"Id":"490822B0E4304A28"},"T2":{"Id":"93955C4B09646128"},"T3":{"Id":"0A1F7A4B09646128"},"T4":{"Id":"100822B108284F28"}}
12:34:48.139 BRY: Successfully loaded 'autoexec.be'
12:34:48.627 SRC: Backlog
12:34:48.628 CMD: Grp 0, Cmd 'WEBBUTTON', Idx 17, Len 7, Pld 17, Data '17.FET3'
12:34:48.631 RSL: RESULT = {"WebButton17":"17.FET3"}
12:34:48.713 WIF: Attempting connection...
12:34:49.000 WIF: Connecting to AP2 SS02 Channel 3 BSSId 3C:52:A1:0E:D9:E2 in mode 11n as ESPSMD-6C3594-5524...
12:34:50.603 WIF: IPv4 192.168.11.64, mask 255.255.255.0, gateway 192.168.11.1
12:34:51.283 SRC: Backlog
12:34:51.285 CMD: Grp 0, Cmd 'WEBBUTTON', Idx 18, Len 7, Pld 18, Data '18.FET4'
12:34:51.288 RSL: RESULT = {"WebButton18":"18.FET4"}
12:34:51.491 SRC: Backlog
12:34:51.493 CMD: Grp 0, Cmd 'WEBBUTTON', Idx 19, Len 6, Pld 19, Data '19.PWM'
12:34:51.496 RSL: RESULT = {"WebButton19":"19.PWM"}
12:34:51.727 SRC: Backlog
12:34:51.729 CMD: Grp 0, Cmd 'WEBBUTTON', Idx 20, Len 6, Pld 20, Data '20.LCD'
12:34:51.732 RSL: RESULT = {"WebButton20":"20.LCD"}
12:34:51.916 WIF: IPv6 Local fe80::e25a:1bff:fe6c:3594%st1
12:34:51.963 SRC: Backlog
12:34:51.965 CMD: Grp 0, Cmd 'DS18RETRYREAD', Idx 1, Len 1, Pld 1, Data '1'
12:34:51.972 RSL: RESULT = {"DS18RetryRead": 1}
12:34:52.110 WIF: Connected
12:34:52.604 RTC: UTC 2024-03-28T11:34:55Z, DST 2024-03-31T02:00:00, STD 2024-10-27T03:00:00
12:34:55.001 RTC: Synced by NTP
12:34:55.065 SRC: Backlog
12:34:55.067 CMD: Grp 0, Cmd 'COUNTERDEBOUNCELOW', Idx 1, Len 16, Pld 5, Data '5, SetOption79 1'
12:34:55.074 RSL: RESULT = {"CounterDebounceLow":5}
12:34:55.180 HTP: Web server active on ESPSMD-6C3594-5524 with IP address 192.168.11.64
12:34:55.300 SRC: Backlog
12:34:55.302 CMD: Grp 0, Cmd 'POWER', Idx 20, Len 1, Pld 1, Data '1'
12:34:55.316 RSL: RESULT = {"POWER20":"ON"}
12:34:55.318 RSL: POWER20 = ON

Thank you for your support.

Best regards,

JP

Jason2866, 2024-03-28:
May working. Not supported to use non consecutive numbers.

Jason2866, 2024-03-28:
Closing since no issues

closed 2024-03-28 by Jason2866