arendst/Tasmota #21070: Documentation for VL53Lxx mentiones incorrect GPIO numbers

linesquarecube, 2024-03-31:
### PROBLEM DESCRIPTION

Hi everyone,

just want to mention that the documentation for using [VL53L0X and VL53L1X laser ranging modules](https://tasmota.github.io/docs/VL53Lxx/) has a switch in the GPIO numbers mentioned in the example for [Step 1 - Configure I2C](https://tasmota.github.io/docs/VL53Lxx/#step-1-configure-i2c). 

> On ESP8266 almost any GPIO can be used for I2C except for GPIO15. However the standard is to use GPIO4 (D1) for I2C SCL and GPIO5 (D2) for I2C SDA.
Corret is:
- I2C SCL (D1) has to be GPIO5
- I2C SDA (D2) has to be GPIO4

Thanks for correcting this to avoid confusion of others - took me some time to figure out why the given example was not working.
Thank you by the way for providing Tasmota :heart:

barbudor, 2024-03-31:
Please provide a PR on https://github.com/tasmota/docs

Jason2866, 2024-03-31:
There is no standard at all. The example will work using the mentioned GPIOs.

sfromis, 2024-03-31:
While it is correct that there is no formal "standard", the D1 Mini form factor shields for I2C devices has SDA on gpio4/D2 and SCL on gpio5/D1. This is enough of a convention to be thought of as a "standard" starting point.

linesquarecube, 2024-03-31:
> Each docs page has an edit link (pencil) top right, making it easy to go into Github edit for the page.

This is true, but I may not have the necessary rights to edit the documentation. I only have the possibility to fork the documentation repository to propose changes.

closed 2024-03-31 by linesquarecube