arendst/Tasmota #21094: Devices with energy monitor: Tasmota boot loops on powerup UNLESS web client is attempting to connect at boot, the webserver is disabled, or the energy monitoring connections are removed from template

jgow, 2024-04-04:
### PROBLEM DESCRIPTION
This issue affects Knightsbridge CU9KW UK wall sockets with embedded energy monitoring and the template specified in this report.

It is important to note that if the energy monitoring is disabled (by disconnecting the three GPIO lines from 'BL0937 CF', 'HLWBL SEL_i and HLWBL_CF1 and specifying them as 'none'), the issue does NOT occur and the device functions normally, albeit with no energy monitoring. The issue ONLY arises when the energy monitoring is enabled by connecting it in the template.

It took quite some time and effort to drill down in to the exact circumstances that triggered this issue, so I have tried to be brief here but there is considerably more information below in 'Additional Context' that may or may not be helpful in tracking this down:

If energy monitoring is connected AND the webserver is enabled on the device (WebServer 1 or 2), the device will not complete its boot, and restarts before connecting to the MQTT host UNLESS a web client is attempting to connect to the web UI at the time of boot - in which case the boot completes and the device becomes operational.

If EITHER the lines associated with energy monitoring are disconnected OR the web server is disabled (WebServer 0), boot continues normally, MQTT connects and the device functions normally.

Once the device has been encouraged to complete its boot sequence successfully, it functions normally and all functions work (including the energy monitoring if enabled)

There seems to be some interaction between the energy monitoring and the web server that causes a boot issue.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Knightsbridge CU9KW
- [X] Tasmota binary firmware version number used: 13.4.0.4(tasmota)-2_7_6(2024-04-03T17:41:14)
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: serial flash/esptool
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
11:20:43.067 MQT: stat/PS1-LAB/RESULT = {"NAME":"Knightsbridge Dual Socket","GPIO":[321,544,320,544,225,2720,1,1,2624,193,2656,224,192,1],"FLAG":0,"BASE":18}
11:20:43.314 MQT: stat/PS1-LAB/RESULT = {"Module":{"0":"Knightsbridge Dual Socket"}}
11:20:43.570 MQT: stat/PS1-LAB/RESULT = {"GPIO0":{"321":"Led_i2"},"GPIO1":{"544":"LedLink"},"GPIO2":{"320":"Led_i1"},"GPIO3":{"544":"LedLink"},"GPIO4":{"225":"Relay2"},"GPIO5":{"2720":"BL0937 CF"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"193":"Switch_n2"},"GPIO14":{"2656":"HLWBL CF1"},"GPIO15":{"224":"Relay1"},"GPIO16":{"192":"Switch_n1"},"GPIO17":{"0":"None"}}

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  No rules are used.

```
- [X] Provide the output of this command: `Status 0`:
```lua
2024-04-04T11:13:43.849682+01:00 PS1-LAB-4443 ESP-MQT: stat/PS1-LAB/STATUS = {"Status":{"Module":0,"DeviceName":"PS1-LAB","FriendlyName":["PS1-LAB",""],"Topic":"PS1-LAB","ButtonTopic":"0","Power":"00","PowerLock":"00","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[3,3,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
2024-04-04T11:13:43.851094+01:00 PS1-LAB-4443 ESP-MQT: stat/PS1-LAB/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:47:14","StartupUTC":"2024-04-04T09:26:29","Sleep":50,"CfgHolder":4617,"BootCount":47,"BCResetTime":"2024-03-26T19:56:11","SaveCount":243,"SaveAddress":"F5000"}}
2024-04-04T11:13:43.853220+01:00 PS1-LAB-4443 ESP-MQT: stat/PS1-LAB/STATUS2 = {"StatusFWR":{"Version":"13.4.0.4(tasmota)","BuildDateTime":"2024-04-03T17:41:14","Boot":7,"Core":"2_7_6","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"370/699"}}
2024-04-04T11:13:43.854668+01:00 PS1-LAB-4443 ESP-MQT: stat/PS1-LAB/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":4,"LogHost":"<ip>","LogPort":514,"SSId":["WF2-PRINTERNET",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000800B","2805C80000000600003C5A0A192800010000","00000040","00206000","00004000","00000000"]}}
2024-04-04T11:13:43.856759+01:00 PS1-LAB-4443 ESP-MQT: stat/PS1-LAB/STATUS4 = {"StatusMEM":{"ProgramSize":700,"Free":300,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14605E","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","8F8AC7C7","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,3,!4,!5,!6,!8,9,10,12,!16,!18,!19,!20,!21,!22,!24,26,!27,29,!30,!35,!37,!45,62,!68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
2024-04-04T11:13:43.858042+01:00 PS1-LAB-4443 ESP-MQT: stat/PS1-LAB/STATUS5 = {"StatusNET":{"Hostname":"PS1-LAB-4443","IPAddress":"<ip-address>","Gateway":"<gw>","Subnetmask":"255.255.255.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"<mac>","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
2024-04-04T11:13:43.860261+01:00 PS1-LAB-4443 ESP-MQT: stat/PS1-LAB/STATUS6 = {"StatusMQT":{"MqttHost":"<mqtt-ip>","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_25B15B","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
2024-04-04T11:13:43.861751+01:00 PS1-LAB-4443 ESP-MQT: stat/PS1-LAB/STATUS7 = {"StatusTIM":{"UTC":"2024-04-04T10:13:43Z","Local":"2024-04-04T11:13:43","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"06:20","Sunset":"19:25"}}
2024-04-04T11:13:43.863672+01:00 PS1-LAB-4443 ESP-MQT: stat/PS1-LAB/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
2024-04-04T11:13:43.865184+01:00 PS1-LAB-4443 ESP-MQT: stat/PS1-LAB/STATUS10 = {"StatusSNS":{"Time":"2024-04-04T11:13:43","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2024-03-26T19:57:13","Total":0.163,"Yesterday":0.002,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
2024-04-04T11:13:43.867281+01:00 PS1-LAB-4443 ESP-MQT: stat/PS1-LAB/STATUS11 = {"StatusSTS":{"Time":"2024-04-04T11:13:43","Uptime":"0T00:47:14","UptimeSec":2834,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"<ssid>","BSSId":"<id>","Channel":3,"Mode":"11n","RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
2024-04-04T10:06:38.520898+01:00 PS1-LAB-4443 ESP-HDW: ESP8266EX
2024-04-04T10:06:38.522729+01:00 PS1-LAB-4443 ESP-CFG: Loaded from flash at F5, Count 235
2024-04-04T10:06:38.524157+01:00 PS1-LAB-4443 ESP-QPC: Count 1
2024-04-04T10:06:38.525534+01:00 PS1-LAB-4443 ESP-CFG: CR 370/699, Busy 0
2024-04-04T10:06:38.526949+01:00 PS1-LAB-4443 ESP-TYA: Active=0
2024-04-04T10:06:38.528456+01:00 PS1-LAB-4443 ESP-ROT: Mode 1
2024-04-04T10:06:38.531185+01:00 PS1-LAB-4443 ESP-NRG: Init driver 1
2024-04-04T10:06:38.532668+01:00 PS1-LAB-4443 ESP-SRC: Restart
2024-04-04T10:06:38.534050+01:00 PS1-LAB-4443 ESP-Project PS - PS1-LAB Version 13.4.0.4(tasmota)-2_7_6(2024-04-03T17:41:14)
2024-04-04T10:06:48.501041+01:00 PS1-LAB-4443 ESP-HDW: ESP8266EX
2024-04-04T10:06:48.505318+01:00 PS1-LAB-4443 ESP-CFG: Loaded from flash at F5, Count 235
2024-04-04T10:06:48.505916+01:00 PS1-LAB-4443 ESP-CFG: CR 370/699, Busy 0
2024-04-04T10:06:48.507254+01:00 PS1-LAB-4443 ESP-TYA: Active=0
[... 228 lines cut ...]
#undef  MQTT_PORT
#define MQTT_PORT         8883

#undef  NTP_SERVER1
#undef  NTP_SERVER2
#undef  NTP_SERVER3

#define NTP_SERVER1      "<ntp-ip>"   
#define NTP_SERVER2      "<ntp-ip>"
#define NTP_SERVER3      "<ntp-ip>"

#undef USE_DOMOTICZ
#undef USE_EMULATION_HUE
#undef USE_EMULATION_WEMO
#undef USE_SONOFF_RF
#undef USE_SHELLY_DIMMER
#undef USE_I2C
#undef USE_TELEGRAM

#undef BOOT_LOOP_OFFSET
#define BOOT_LOOP_OFFSET 12

```

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2024-04-04:
Doing TLS on ESP8266 is a challenge due to lack of RAM. I suggest you enable heap logging (``SO130 1``) and watch it decline during boot. 

I noticed you're having Tuya in the mix. If not needed I suggest to remove it to get some more free RAM.

arendst, 2024-04-04:
You're right regarding the non heap logging in syslog. 

The exception 2 is worrying though as it shouldn't be there. May have many causes but this is a known one: ```Occurs when out of stack (chk with debug and FreeMem 1)```.

Stack space is just max 4k and depending on function deep calls it is used up quitte soon. 

Pls try to find out if you always get the same exception 2 with the same call stack (it differs on every build you make so keep the same build and the accompanied map file for debugging) en see what happens at the call chain addresses.

jgow, 2024-04-04:
Interestingly I could not repeat the exception 2. However, on the next successful boot after the failure an exception 28 is repeatable. The call trace is very similar but not exactly the same:
```(lua)
2024-04-04T15:35:17.238523+01:00 PS1-LAB-4443 ESP-MQT: tele/PS1-LAB/INFO3 = {"Info3":{"RestartReason":{"Exception":28,"Reason":"Exception","EPC":["4024d8fc","00000000","00000000"],"EXCVADDR":"0000036e","DEPC":"00000000","CallChain":["4024d9ec","4024e02c","4024da1c","4024e034","4024e120","40275ab8","40275ac5","40275b0a","40243dd0","40000f49","40000f49","40000f49","40000e19","40105871","40105877","4010000d","4026e530","4026e4e1","40100230","401011c5","401011b4","401010ec","4000050c","40257ebd","40257ea8","4028b7d0","401011c5","40258487","4025837b","4028b6b4","402584b4"]},"BootCount":65}}

tele/PS1-LAB/INFO3 = {"Info3":{"RestartReason":{"Exception":28,"Reason":"Exception","EPC":["4024d8fc","00000000","00000000"],"EXCVADDR":"0d194b4d","DEPC":"00000000","CallChain":["4024d9ec","4024e02c","4024da1c","4028dfbc","4024e034","4024e120","40275ab8","40275ac5","40275b0a","40243dd0","40000f49","40000f49","40000f49","40000e19","40105871","40105877","4010000d","4026e530","4026e4e1","40100230","401011c5","401011b4","401010ec","4000050c","402580ed","40257ea8","4028b7d0","401011c5","40258487","4025837b","4028b6b4"]},"BootCount":66}}
15:28:51.160-021 MQT: tele/PS1-LAB/INFO3 = {"Info3":{"RestartReason":{"Exception":28,"Reason":"Exception","EPC":["4024d8fc","00000000","00000000"],"EXCVADDR":"00002923","DEPC":"00000000","CallChain":["4024d9ec","4024e02c","4024da1c","4024e034","4024e120","40275ab8","40275ac5","40275b0a","40243dd0","40000f49","40000f49","40000f49","40000e19","40105871","40105877","4010000d","4026e530","4026e4e1","40100230","401011c5","401011b4","401010ec","4000050c","40258105","40257ea8","4028b770","4015b4a0","4028b7d0","40258487","4025837b","4028b6b4"]},"BootCount":61}}

tele/PS1-LAB/INFO3 = {"Info3":{"RestartReason":{"Exception":28,"Reason":"Exception","EPC":["4024d8fc","00000000","00000000"],"EXCVADDR":"000016fb","DEPC":"00000000","CallChain":["4024d9ec","4024e02c","4024da1c","4024e034","4024e120","40275ab8","40275ac5","40275b0a","40243dd0","40000f49","40000f49","40000f49","40000e19","40105871","40105877","4010000d","4026e530","4026e4e1","40100230","401011c5","401011b4","401010ec","4000050c","40258074","40257ea8","4028b7d0","40258487","4025837b","4028b6b4","402584b4","4028b6b4"]},"BootCount":63}}

tele/PS1-LAB/INFO3 = {"Info3":{"RestartReason":{"Exception":28,"Reason":"Exception","EPC":["4024d8fc","00000000","00000000"],"EXCVADDR":"01dd6f3d","DEPC":"00000000","CallChain":["4024d9ec","4024e02c","4024da1c","4024e034","4024e120","40275ab8","40275ac5","40275b0a","40243dd0","40000f49","40000f49","40000f49","40000e19","40105871","40105877","4010000d","4026e530","4026e4e1","40100230","401011c5","401011b4","401010ec","4000050c","40257eca","40257ea8","4028b7d0","40258487","4025837b","4028b6b4","402584b4","4028b6b4"]},"BootCount":64}}
```
Looking at the call trace and map file, 0x4024d9ec corresponds to tcp_process_refused_data in liblwip2-1460.a.
At the other end 0x4028b6b4 corresponds to br_sha256_vtable, within libbearssl.
I will keep looking but I am unfamiliar with bearssl internals.

jgow, 2024-04-05:
It seems fairly consistent across builds. The exception occurs when an interrupt (possibly a timer in the TCP stack as the call sequence is lwip_cyclic_timers -> tcp_tmr -> tcp_fasttmr -> tcp_process_refused_data) occurs during the execution of one or more functions in sha2small.c, usually br_sha224_update or br_sha256_vtable.

I can't trace this back further in brssl thanks to the vtables used therein, and there is no way I can use any form of serial debugging (this is a nonisolated mains powered device - a wall socket - and I currently do not have an isolated serial interface available to me). However I think the exception may be in the initial MQTT connection code, as if I look in the broker logs in the fault condition, there is an attempt to initiate the connection but the handshake is never completed - this also seems to be consistent with the point in the boot sequence that the device restarts.

Have tried a ham-fisted approach by disabling interrupts around the functions in sha2small.c, but the exception just occurs earlier.

I suspect, but can't confirm, that if stack space is tight then this interrupt in the middle of some TLS processing may just be enough.

This is about as far as I can get. Disablling the web server certainly prevents the issue from occurring, so I suspect that the interrupt is somehow related to the web server.

arendst, 2024-04-05:
You might want to try increase TLS stackspace in file ``\lib\lib_ssl\tls_mini\src\StackThunk_light.cpp`` line 43 and up.

```
//#define _stackSize (5600/4)
#if defined(USE_MQTT_AWS_IOT) || defined(USE_MQTT_AWS_IOT_LIGHT) || defined(USE_MQTT_AZURE_IOT)
  #define _stackSize (5300/4)   // using a light version of bearssl we can save 300 bytes
#else
  #define _stackSize (4800/4)   // no private key, we can reduce a little, max observed 4300
#endif
```

As you enabled USE_MQTT_AWS_IOT try to change 5300/4 into 5600/4 or 5800/4

jgow, 2024-04-05:
That fixed it - setting the stack size to 5600/4. I will update the other sockets to this build and allow the devices to run for a while to check for any unintended consequences of the increased stack size, then close issue closed if all is well.

I will just have to remember to change this in the build when I upgrade the firmware on these devices in future (unless this value could be made user-configurable?).

Can I check that I really need to enable USE_MQTT_AWS_IOT in order just to include the private CA? I only need to include my private CA certificate; I do not care about AWS or LetsEncrypt. The documentation seems to suggest that I do, and if I disable the setting the device will not connect so I assume this is correct?

jgow, 2024-04-06:
I spoke too soon - using 5600/4 improved the situation but the device still failed on some boots. I had to increase this to 5800/4 to get stable operation and the fault no longer appears on reboots.

I will close this issue now as I can build a version of Tasmota that is stable and appears to work reliably with these devices.

closed 2024-04-06 by jgow