arendst/Tasmota #21134: Command unknown: shutterStopRelay1 1

sedlacekdavid, 2024-04-10:
### PROBLEM DESCRIPTION
**shutterStopRelay** command unknown - it is advised by the shutter config guide to enable support for a third "stop" relay to stop moving shutters.
Otherwise my shutter config behaves as expected.

`07:17:36.884 CMD: shutterStopRelay1 1
07:17:36.891 RSL: RESULT = {"Command":"Unknown"}`

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff 4ch pro
- [x] Tasmota binary firmware version number used: 13.4.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: web browser
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
07:21:30.988 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
07:21:31.195 RSL: RESULT = {"Module":{"23":"Sonoff 4CH Pro"}}
07:21:31.448 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"226":"Relay3"},"GPIO5":{"225":"Relay2"},"GPIO9":{"33":"Button2"},"GPIO10":{"34":"Button3"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"35":"Button4"},"GPIO15":{"227":"Relay4"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
07:22:06.291 RSL: STATUS = {"Status":{"Module":23,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4"],"Topic":"tasmota_C7B735","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
07:22:06.325 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Power On","Uptime":"0T00:06:38","StartupUTC":"2024-04-10T06:15:28","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2024-04-10T06:28:21","SaveCount":38,"SaveAddress":"F6000"}}
07:22:06.359 RSL: STATUS2 = {"StatusFWR":{"Version":"13.4.0(tasmota)","BuildDateTime":"2024-02-14T16:13:56","Boot":31,"Core":"2_7_6","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285N08","CR":"366/699"}}
07:22:06.377 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["7WZYRGRX",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C80001000600003C5A0A192800000000","40000080","00006000","00004000","00000000"]}}
07:22:06.409 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":646,"Free":356,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14325E","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,!3,!4,!5,!6,7,!8,9,10,12,!16,!18,!19,!20,!21,!22,!24,26,27,29,!30,!35,!37,!45,62,!68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
07:22:06.444 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-C7B735-5941","IPAddress":"192.168.1.131","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"34:98:7A:C7:B7:35","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
07:22:06.468 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C7B735","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
07:22:06.490 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-04-10T06:22:06Z","Local":"2024-04-10T07:22:06","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"06:08","Sunset":"19:34"}}
07:22:06.508 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-04-10T07:22:06","Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":0}}}
07:22:06.520 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-04-10T07:22:06","Uptime":"0T00:06:38","UptimeSec":398,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"7WZYRGRX","BSSId":"1E:E8:29:97:0D:18","Channel":1,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:04"}}}
07:22:06.553 RSL: STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"1"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-04-10:
The `ShutterStopRelay` command is part of a [new feature](https://github.com/arendst/Tasmota/pull/20758) only added in latest development builds.

sedlacekdavid, 2024-04-10:
Thanks, so the guide has been updated but it's not yet in the official release. And according to the feature you mentioned it seems it will be supported only by ESP32...

closed 2024-04-13 by Jason2866