arendst/Tasmota #21136: ZBInfo does not list all devices

Agnes56fr, 2024-04-10:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

I'm facing a strange problem with a sonoff Zibgbee bridge flashed with Tasmota 13.4.0.4 
12 devices appear in the main menu
if I make a command zbinfo 1, Zbinfo 2, ... up to 12 everything works
If I make zbinfo then only 11 devices are listed 

any advice ?

Agnes56fr, 2024-04-10:
This is the result of Zbinfo command

10:39:48.387 RSL: SENSOR = {"ZbInfo":{"0x4067":{"Device":"0x4067","Name":"leak_spa","IEEEAddr":"0xA4C138A01BB7384C","ModelId":"TS0207","Manufacturer":"_TZ3000_ocjlo4ea","Endpoints":[1],"Config":["A01.4"],"ZoneType":42,"ZoneStatus":0,"Reachable":false,"BatteryPercentage":27,"LastSeen":6245,"LastSeenEpoch":1712735743,"LinkQuality":0}}}
10:39:48.398 RSL: SENSOR = {"ZbInfo":{"0x4DEA":{"Device":"0x4DEA","Name":"temp_etage","IEEEAddr":"0xA4C13853E0677A42","ModelId":"TS0201","Manufacturer":"_TZ3000_fllyghyj","Endpoints":[1],"Config":["T01"],"Temperature":21.63,"Humidity":47.1,"Reachable":true,"BatteryPercentage":100,"BatteryLastSeenEpoch":1712740344,"LastSeen":1644,"LastSeenEpoch":1712740344,"LinkQuality":58}}}
10:39:48.409 RSL: SENSOR = {"ZbInfo":{"0xD1A7":{"Device":"0xD1A7","Name":"Temp_Exterieure","IEEEAddr":"0xA4C13870440D3D7E","ModelId":"TS0201","Manufacturer":"_TZ3000_fllyghyj","Endpoints":[1],"Config":["T01"],"Temperature":13.92,"Humidity":80.4,"Reachable":true,"BatteryPercentage":57,"BatteryLastSeenEpoch":1712740254,"LastSeen":534,"LastSeenEpoch":1712741454,"LinkQuality":52}}}
10:39:48.419 RSL: SENSOR = {"ZbInfo":{"0x43B0":{"Device":"0x43B0","Name":"temp_ssol","IEEEAddr":"0xA4C138EB643973FB","ModelId":"TS0201","Manufacturer":"_TZ3000_fllyghyj","Endpoints":[1],"Config":["T01"],"Temperature":15.01,"Humidity":71.64,"Reachable":true,"BatteryPercentage":17,"BatteryLastSeenEpoch":1712740734,"LastSeen":18,"LastSeenEpoch":1712741970,"LinkQuality":97}}}
10:39:48.430 RSL: SENSOR = {"ZbInfo":{"0xD76C":{"Device":"0xD76C","Name":"Temp_SalleSport","IEEEAddr":"0xA4C138CE7343D647","ModelId":"TS0201","Manufacturer":"_TZ3000_xr3htd96","Endpoints":[1],"Config":["T01"],"Temperature":19.94,"Humidity":43.14,"Reachable":true,"BatteryPercentage":100,"BatteryLastSeenEpoch":1712740649,"LastSeen":1338,"LastSeenEpoch":1712740650,"LinkQuality":50}}}
10:39:48.443 RSL: SENSOR = {"ZbInfo":{"0x7493":{"Device":"0x7493","Name":"Leak_arriere_cuisine","IEEEAddr":"0xA4C138A2E5BE0A8F","ModelId":"TS0207","Manufacturer":"_TZ3000_kstbkt6a","Endpoints":[1],"Config":["A01.4"],"ZoneType":42,"ZoneStatus":0,"Reachable":false,"BatteryPercentage":9,"LastSeen":10672,"LastSeenEpoch":1712731316,"LinkQuality":71}}}
10:39:48.455 RSL: SENSOR = {"ZbInfo":{"0xF49F":{"Device":"0xF49F","Name":"Temp_Salon","IEEEAddr":"0xA4C138263243EF05","ModelId":"TS0201","Manufacturer":"_TZ3210_huzkzqyk","Endpoints":[1,2],"Config":["~01.1","I01","T02"],"Illuminance":0,"Temperature":20.3,"Humidity":50.1,"Reachable":true,"BatteryPercentage":100,"BatteryLastSeenEpoch":1712739920,"LastSeen":17,"LastSeenEpoch":1712741971,"LinkQuality":55}}}
10:39:48.466 RSL: SENSOR = {"ZbInfo":{"0x56D6":{"Device":"0x56D6","Name":"_Prise_1","IEEEAddr":"0x144405602BD5B370","ModelId":"TS011F","Manufacturer":"_TZ3000_okaz9tjs","Endpoints":[1],"Config":["O01","P01"],"Power":0,"RMSVoltage":229,"ActivePower":0,"Reachable":true,"LastSeen":12,"LastSeenEpoch":1712741976,"LinkQuality":50}}}
10:39:48.477 RSL: SENSOR = {"ZbInfo":{"0x2EAF":{"Device":"0x2EAF","Name":"_Prise_2","IEEEAddr":"0x164705602BD5B370","ModelId":"TS011F","Manufacturer":"_TZ3000_okaz9tjs","Endpoints":[1],"Config":["O01","P01"],"Power":0,"RMSVoltage":230,"ActivePower":0,"Reachable":true,"LastSeen":12,"LastSeenEpoch":1712741976,"LinkQuality":71}}}
10:39:48.488 RSL: SENSOR = {"ZbInfo":{"0x5A70":{"Device":"0x5A70","Name":"_Prise_3","IEEEAddr":"0xD04805602BD5B370","ModelId":"TS011F","Manufacturer":"_TZ3000_okaz9tjs","Endpoints":[1],"Config":["O01","P01"],"Power":0,"RMSVoltage":237,"ActivePower":0,"Reachable":true,"LastSeen":12,"LastSeenEpoch":1712741976,"LinkQuality":97}}}
10:39:48.499 RSL: SENSOR = {"ZbInfo":{"0x9332":{"Device":"0x9332","Name":"_Prise_4","IEEEAddr":"0xAE4B05602BD5B370","ModelId":"TS011F","Manufacturer":"_TZ3000_okaz9tjs","Endpoints":[1],"Config":["P01","O01"],"RMSVoltage":230,"ActivePower":0,"Power":0,"Reachable":true,"LastSeen":13,"LastSeenEpoch":1712741975,"LinkQuality":31}}}
10:39:48.506 RSL: RESULT = {"ZbInfo":"Done"}

the first device is not listed but appear with "zbinfo 1"
10:41:08.085 CMD: zbinfo 1
10:41:08.094 RSL: SENSOR = {"ZbInfo":{"0xE404":{"Device":"0xE404","Name":"leak_pompe_piscine","IEEEAddr":"0xA4C13889F8BBF6A5","ModelId":"TS0207","Manufacturer":"_TZ3000_t6jriawg","Endpoints":[1],"Config":["A01.4"],"ZoneType":42,"ZoneStatus":0,"Reachable":false,"BatteryPercentage":9,"LastSeen":8929,"LastSeenEpoch":1712733139,"LinkQuality":53}}}
10:41:08.101 RSL: RESULT = {"ZbInfo":"Done"}

arendst, 2024-04-10:
How/Where do you execute the command?

If on the web console it is most likely the cyclic log buffer is full while adding the twelvth entry deleting the first. Try the same using a syslog server.

Agnes56fr, 2024-04-10:
the command is executed via http
http://192.168.0.129/cm?cmnd=zbinfo
I do not know what a syslog server is can you explain ?

Agnes56fr, 2024-04-10:
for more informations
zbstatus list all 12 devices

{"ZbStatus1":[{"Device":"0xE404","Name":"leak_pompe_piscine"},{"Device":"0x4067","Name":"leak_spa"},{"Device":"0x4DEA","Name":"temp_etage"},{"Device":"0xD1A7","Name":"Temp_Exterieure"},{"Device":"0x43B0","Name":"temp_ssol"},{"Device":"0xD76C","Name":"Temp_SalleSport"},{"Device":"0x7493","Name":"Leak_arriere_cuisine"},{"Device":"0xF49F","Name":"Temp_Salon"},{"Device":"0x56D6","Name":"_Prise_1"},{"Device":"0x2EAF","Name":"_Prise_2"},{"Device":"0x5A70","Name":"_Prise_3"},{"Device":"0x9332","Name":"_Prise_4"}]}

arendst, 2024-04-10:
Executing using HTTP or on the web GUI will result in the same experience; the report buffer is full and loses the first entry in favour of the last.

A syslog server is a device able to log messages from all kinds of resources like your Zigbee bridge. 

You'll have to learn with the fact that executing the command will not report all devices. You already found the solution by reporting individual devices.

barbudor, 2024-04-10:
Syslog is a standard protocol to send log messages over the network to a receiving log server
The web console as a limited buffer so @arendst is assuming that it is too short to get all results from the ZbInfo command
Syslog would not have that limitation as each log is immediately sent to the server

If you are running linux, syslog is most of the time already running and listenning on port 514 (see https://tasmota.github.io/docs/Troubleshooting/#syslog-logging)
You could configure tasmota to send the logs to that server using:
`backlog loghost xxx.xxx.xxx.xx; logport 514; syslog 2`

If you are running windows and have Python install, you can run a small syslog equivalent with this [python code](https://gist.github.com/barbudor/2f860f53bcb7131f538ceeb3db4142dc) which use port 5140, so configure tasmota with:
`backlog loghost xxx.xxx.xxx.xx; logport 5140; syslog 2`

Agnes56fr, 2024-04-10:
ok,
can it be a buffer size problem if the result come from a http request ?

arendst, 2024-04-10:
> can it be a buffer size problem if the result come from a http request

The buffer size is indeed the cause BUT ... The buffer size is bound to available RAM. Increasing the buffer wil decrease precious RAM. That's a trade-off I have to make. Increasing the buffer is not the solution as you can imagine; having fifty zigbee devices would need all available RAM. You have to change the way you retreive your data like you already found out by individual ``ZbInfo`` commands.

sfromis, 2024-04-10:
Commands via http is the worst way if you want to see full results. You'll generally get better results in the Tasmota web UI console, as the JavaScript code in the browser can save data it got before it rolled out of the limited buffer space in Tasmota itself. More generally, make sure to already be in the web UI console when the command is issued.

However, for a command producing a lot of output at the same time, the lines may roll off the buffer before being picked up by the browser; the default refresh interval is 2.345 seconds.

If you use serial console instead, you'll generally get all lines as they are produced, without the limitations incurred when accessing results via the web UI.

I'd say that the best strategy is what Tasmota is designed for, using MQTT. The results will be published, without losing lines, and can be picked up that way.

Jason2866, 2024-04-10:
Closing since answered and it is not an issue.

closed 2024-04-10 by Jason2866

Agnes56fr, 2024-04-10:
"You'll generally get better results in the Tasmota web UI console, as the JavaScript code in the browser can save data it got before it rolled out of the limited buffer space in Tasmota itself."

When in the web UI it does exactly the same

s-hadinger, 2024-04-10:
I will eventually try eventsource which should eliminate this problem with the console

Agnes56fr, 2024-04-10:
hello, thank you for your answer,
what is eventsource ?

Agnes56fr, 2024-04-10:
?m=1 output is a lot longer than /cm/cmnd=zbinfo
and ?m=1 is not truncated

Agnes56fr, 2024-04-10:
Thank you all for trying to help me.

I use http://192.168.0.129/cm?cmnd=zbinfo in a php file and analyse the answer to display the status of all devices in a html page (which is static, it does not update as device's status change, it just update when you refresh the page, wich is more than enough considering it display temperatures mostly).

I understood why it could not cut the result anywhere since the size of the buffer is checked device/device
I though I understood that being in the console might create the buffer limit problem, then though it was finally preferable to be in the console (so I'm a bit lost on understanding the buffer problem really)

using eventsource to push devices' changes to a web page sound great, but since it is not implemented in Tasmota, I will not go this way.

instead, I'm about to rewrite my php in order to pull info devices/devices with zbinfo 1, 2, 3...
I thought it was better to make one global request but if you say I can only if it is short enough then I will go device/device.

Agnes56fr, 2024-04-10:
Thanks again for all those informations.

So the ZBInfo command overfilled the internal tasmota buffer.
I also understood that MQTT should be the way.

For me having a MQTT server beside tasmota is kinda double sources of informations since tasmota already keep the informations  and since I can easyly access it with the http request even if not designed for that and all that without having to contact the devices, well, for my application, I could not see any good reason to have another server with MQTT on it.

I thank you very much all for your work and for your answers