arendst/Tasmota #21150: 13.4.0.3 to 13.4.0.4 update breaks the wifi connection

JF62, 2024-04-11:
### PROBLEM DESCRIPTION
Updating to the latest dev version (I expect it's depending on the WIFI refactoring (wifihelper)) breaks the connection to my ESP8266EX devices. After Update they just blinking and don't connect anymore. A revert to 13.4.0.3 and all is fine again

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): SONOFF BASIC ESP8266EX
- [x ] Tasmota binary firmware version number used: 13.0.4.4
  - [x ] Pre-compiled
  - [x] Self-compiled
  - tried both
- [x ] Flashing tools used: VS or WEBINTERFACE OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
Not possible because of no connection to device after update

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
Not possible because of no connection to device after update

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Jutst UPDATE

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
that it will work ;-)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

JF62, 2024-04-11:
Additional info: Upfate to LITE Version works fine, update to DE or Standard TASMOTA version fails

sfromis, 2024-04-11:
Seeing no ESP8266 Wifi trouble after having upgraded two devices to most recent dev build commit d3ab3ce, from 3½ hours ago.

Jason2866, 2024-04-11:
It is the "classic" Sonoff Basic problem. Bad power supply.

JF62, 2024-04-11:
13.4.0.3 all fine and after upgrade it's bad power supply and after downgrade all good again?

Jason2866, 2024-04-11:
Yes.

JF62, 2024-04-11:
Great explanation... Can you tell why? I see just a change in the wifihelper.h and if I revert this change back all is fine

s-hadinger, 2024-04-12:
I will make some additional tests. The changes in wifihelper are not supposed to change anything for esp8266

arendst, 2024-04-12:
FWIW my sonoff basic still works with v13.4.0.4
```
00:00:00.002 HDW: ESP8266EX
00:00:00.054 CFG: Loaded from flash at FA, Count 9552
00:00:00.060 QPC: Count 1
00:00:00.065 TYA: Active=0
00:00:00.069 ROT: Mode 1
00:00:00.070 SRC: Restart
00:00:00.072 Project tasmota - Sonoff3 Version 13.4.0.4(7e455ba-tasmota)-2_7_6(2024-04-11T20:31:09)
00:00:00.175 WIF: Attempting connection...
00:00:01.001 WIF: Connecting to AP2 indebuurt_IoT Channel 11 BSSId 18:E8:29:CA:17:C1 in mode 11n as sonoff3...
00:00:02.511 WIF: Connected
00:00:02.763 HTP: Web server active on sonoff3 with IP address 192.168.2.139
00:00:03.256 RTC: UTC 2024-04-12T09:15:14Z, DST 2024-03-31T02:00:00, STD 2024-10-27T03:00:00
11:15:14.001 RTC: Synced by NTP
11:15:14.532 MQT: Attempting connection...
11:15:14.547 MQT: Connected
11:15:14.550 MQT: tele/sonoff3/LWT = Online (retained)
11:15:14.552 MQT: cmnd/sonoff3/POWER = 
11:15:14.554 MQT: Subscribe to cmnd/sonoff3/#
11:15:14.556 MQT: Subscribe to cmnd/tasmotas/#
11:15:14.557 MQT: Subscribe to cmnd/DVES_0CAB6F_fb/#
11:15:14.565 MQT: tele/sonoff3/INFO1 = {"Info1":{"Module":"Sonoff Basic","Version":"13.4.0.4(7e455ba-tasmota)","FallbackTopic":"cmnd/DVES_0CAB6F_fb/","GroupTopic":"cmnd/tasmotas/"}}
11:15:14.579 MQT: tele/sonoff3/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"sonoff3","IPAddress":"192.168.2.139"}}
11:15:14.590 MQT: tele/sonoff3/INFO3 = {"Info3":{"RestartReason":"Software/System restart","BootCount":313}}
11:15:14.601 MQT: stat/sonoff3/RESULT = {"POWER":"OFF"}
11:15:14.604 MQT: stat/sonoff3/POWER = OFF
11:15:17.295 QPC: Reset
11:15:19.231 APP: Boot Count 313
11:15:19.238 MQT: tele/sonoff3/STATE = {"Time":"2024-04-12T11:15:19","Uptime":"0T00:00:10","UptimeSec":10,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":2,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"Mode":"11n","RSSI":100,"Signal":-16,"LinkCount":1,"Downtime":"0T00:00:04"}}
11:15:19.333 CFG: Saved to flash at F9, Count 9553, Bytes 4096
```

JF62, 2024-04-12:
> But I agree with @Jason2866; the sonoff basic is notorious to stop working at some new version. Your bus stops here. Search the issues and discussions. You're not alone among sonoff basic friends.

But why then the LITE version works fine and the standard don't?

Jason2866, 2024-04-12:
Once again search

closed 2024-04-12 by Jason2866

Joylancer, 2024-04-22:
Ich habe das Problem mit dem Wifi Reconnect bei der letzten Dev auch. 
2 Geräte, einmal 
Sonoff Dual R3 V2 (ESP32-D0WD-V3 v3.0/tasmota32.bin)
 und beim 
Shelly Plus 2PM v0.1.9 (ESP32-U4WDH-S v3.0/tasmota32solo1.bin).
Bemerkt habe ich es im Sonoff Adapter, das beide sich permanent reconnected haben.
Bin jetzt zurück auf die Release-Version (13.4.0), damit keine Probleme.
Fehler bei beiden: Multicast (re)joined
Vielleicht könnt ihr damit etwas anfangen.
Alle anderen Geräte (ESP8266) laufen ohne Probleme.