arendst/Tasmota #21179: Device locator live link is broken in wiki

jsprenkle, 2024-04-14:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Go to: 
https://tasmota.github.io/docs/Tasmota-Device-Locator/#run-the-live-version
Click on the link.
Goes to a provider parking page.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Should go to device locator page

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2024-04-14:
I'm not sure who was hosting the live version. We can certainly find an alternative solution

Jason2866, 2024-04-14:
Since CORS needs to be enabled to work and the project is not maintained anymore i would vote for removing it from the documentation.

sfromis, 2024-04-14:
As mentioned on Discord:
> That very old abandonware is not part of the Tasmota project. If you want to try to resuscitate it, you can get it from Github.

jsprenkle, 2024-04-14:
> Since CORS needs to be enabled to work and the project is not maintained anymore i would vote for removing it from the documentation.

Seems like a reasonable idea if the author isn't maintaining it. I wrote to him via github mail but got no response so far

Jason2866, 2024-04-17:
removed from documentation

closed 2024-04-17 by Jason2866