arendst/Tasmota #21187: Can't switch power outlets after update to tasmota 13.4.0.4 (and 13.4.1.1)

MyUniversalDeveloper, 2024-04-15:
### PROBLEM DESCRIPTION
Can't switch power outlets after update to tasmota 13.4.0.4 (and 13.4.1.1)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Gosund_P1
- [x] Tasmota binary firmware version number used: 13.4.1.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
14:34:30.840 RSL: RESULT = {"NAME":"Gosund_P1","GPIO":[0,3072,544,3104,0,259,0,0,225,226,224,0,32,4704],"FLAG":0,"BASE":18}
14:34:31.077 RSL: RESULT = {"Module":{"0":"Gosund_P1"}}
14:34:31.293 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"3072":"CSE7766 Tx"},"GPIO2":{"544":"LedLink"},"GPIO3":{"3104":"CSE7766 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"259":"Relay_i4"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"225":"Relay2"},"GPIO13":{"226":"Relay3"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"4704":"ADC Input"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
14:35:15.571 RSL: RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":34,"Free":477,"Rules":"ON Button1#State DO Power4 2 ENDON"}}
14:35:15.806 RSL: RESULT = {"Rule2":{"State":"ON","Once":"ON","StopOnError":"OFF","Length":116,"Free":395,"Rules":"ON Analog#A0div10<30 DO Power3 2 BREAK ON Analog#A0div10<60 DO Power2 2 BREAK ON Analog#A0div10<80 DO Power1 2 ENDON"}}
14:35:16.015 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```

- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:35:50.141 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Mehrfachsteckdose","FriendlyName":["Steckdose 1","Steckdose 2","Steckdose 3","USB"],"Topic":"mdosexxxx","ButtonTopic":"0","Power":"1000","PowerLock":"0000","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
14:35:50.159 RSL: STATUS1 = {"StatusPRM":{"Baudrate":4800,"SerialConfig":"8E1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Exception","Uptime":"0T00:25:25","StartupUTC":"2024-04-15T12:10:25","Sleep":50,"CfgHolder":4617,"BootCount":451,"BCResetTime":"2020-07-10T11:53:11","SaveCount":2265,"SaveAddress":"F5000"}}
14:35:50.173 RSL: STATUS2 = {"StatusFWR":{"Version":"13.4.1.1(d957c97-tasmota)","BuildDateTime":"2024-04-15T10:31:40","Boot":31,"Core":"2_7_6","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285N08","CR":"403/699"}}
14:35:50.187 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["StFld5",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C80001000680003C5A0A002800000000","000000C0","00006000","00004000","00000000"]}}
14:35:50.211 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":647,"Free":356,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,!2,3,!4,!5,!6,!7,!8,9,10,!12,!16,!18,!19,!20,!21,!22,!24,26,!27,29,!30,!35,!37,!45,62,!68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
14:35:50.228 RSL: STATUS5 = {"StatusNET":{"Hostname":"mdosexxx","IPAddress":"192.168.xxx.x","Gateway":"192.168.xxx.x","Subnetmask":"255.255.255.0","DNSServer1":"192.168.xxx.x","DNSServer2":"0.0.0.0","Mac":"xxxx","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0}}
14:35:50.244 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-04-15T12:35:50Z","Local":"2024-04-15T14:35:50","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":99,"Sunrise":"06:34","Sunset":"20:28"}}
14:35:50.258 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
14:35:50.319 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-04-15T14:35:50","ANALOG":{"A0":1024},"ENERGY":{"TotalStartTime":"2020-07-10T11:45:37","Total":34.856,"Yesterday":0.073,"Today":0.043,"Power":3,"ApparentPower":16,"ReactivePower":16,"Factor":0.18,"Voltage":230,"Current":0.070}}}
14:35:50.336 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-04-15T14:35:50","Uptime":"0T00:25:25","UptimeSec":1525,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"ON","Wifi":{"AP":1,"SSId":"StFld5","BSSId":"04:B4:FE:D6:31:08","Channel":1,"Mode":"11n","RSSI":64,"Signal":-68,"LinkCount":1,"Downtime":"0T00:00:15"}}}
14:35:50.359 RSL: STATUS12 = {"StatusSTK":{"Exception":28,"Reason":"Exception","EPC":["40235150","00000000","00000000"],"EXCVADDR":"00000079","DEPC":"00000000","CallChain":["40235cba","40258594","40256d08","402394a5","40227322","40252329","4025188c","40221c40","40296706","4025186c","40296706","40221c40","40221c40","402399a8","402529b4","40253d5e","40246b4a","40247119","40251e7c","40221c40","402558a2","40255900","401000e1","40221c78","40251925","40228e7e","401012c2","40228f97","40256b10","4024a121","4021c0a3"]}}

```

- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:00:05.895 WIF: Checking connection...
00:00:05.898 WIF: Connected
00:00:05.938 DMP: 55 5A 02 D8 20 00 07 AC 00 3D 6D 03 77 1F 4E C7 F8 25 8A 50 E1 C2 69 08
00:00:05.964 DMP: 55 5A 02 D8 20 00 07 AC 00 3D 6D 03 77 1F 4E C7 F8 27 22 B6 E1 C2 69 08
00:00:06.019 WIF: Sending Gratuitous ARP
00:00:06.022 NTP: Sync time...
00:00:06.024 WIF: DNS resolved '192.168.xxx.x' (192.168.177.1) in 1 ms
00:00:06.038 RTC: UTC 2024-04-15T12:45:10Z, DST 2024-03-31T02:00:00, STD 2024-10-27T03:00:00
14:45:10.000 RTC: Synced by NTP
14:45:10.022 DMP: 55 5A 02 D8 20 00 07 AC 00 3D 6D 03 77 1F 4E C7 F8 28 BC 1B E1 C2 69 08
14:45:10.074 HTP: Web server active on mdosexxx with IP address 192.168.xxx.x
14:45:10.079 RSL: INFO1 = {"Info1":{"Module":"Gosund_P1","Version":"13.4.1.1(d957c97-tasmota)","FallbackTopic":"cmnd/DVES_C67027_fb/","GroupTopic":"cmnd/tasmotas/"}}
14:45:10.092 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"mdosexxx","IPAddress":"192.168.xxx.x"}}
14:45:10.113 RSL: INFO3 = {"Info3":{"RestartReason":{"Exception":28,"Reason":"Exception","EPC":["40235150","00000000","00000000"],"EXCVADDR":"00000079","DEPC":"00000000","CallChain":["40235cba","40258594","40256d08","402394a5","40227322","40252329","4025188c","40221c40","40296706","4025186c","40296706","40221c40","40221c40","402399a8","402529b4","40253d5e","40246b4a","40247119","40251e7c","40221c40","402558a2","40255900","401000e1","40221c78","40251925","40228e7e","401012c2","40228f97","40256b10","4021f88c","402314d4"]},"BootCount":453}}
14:45:10.129 RSL: RESULT = {"POWER1":"OFF"}
14:45:10.131 RSL: POWER1 = OFF
14:45:10.134 RSL: RESULT = {"POWER2":"OFF"}
[... 12 lines cut ...]
14:45:10.863 DMP: 55 5A 02 D8 20 00 07 AC 00 3D 6D 03 FD DF 4E C7 F8 35 87 43 E1 C2 69 4E
14:45:11.009 DMP: 55 5A 02 D8 20 00 07 AC 00 3D 6D 03 FD DF 4E C7 F8 37 21 A7 E1 C2 69 4E
14:45:11.216 QPC: Reset
14:45:11.222 DMP: 55 5A 02 D8 20 00 07 AC 00 3D 6D 03 FD DF 4E C7 F8 38 BB 0C E1 C2 69 4E
14:45:11.240 DMP: 55 5A 02 D8 20 00 07 AC 00 3D 6D 03 FD DF 4E C7 F8 3A 53 72 E1 C2 69 4E
14:45:11.323 DMP: 55 5A 02 D8 20 00 07 AC 00 3D 6D 03 FD DF 4E C7 F8 3B EC D7 E1 C2 69 4D
14:45:11.429 DMP: 55 5A 02 D8 20 00 07 B1 00 3D 6D 03 FD DF 4E C7 F8 3D 87 3B E1 C2 69 53

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Like described above.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Correct switching of the power outlets.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2024-04-15:
I fear your internal power supply reaches EOL. We see more devices having issues with later Tasmota versions.

I've tested latest dev release on a Sonoff 4channel relay box and it still works fine there.

arendst, 2024-04-15:
For testing purposes you might want to try disabling the analog input and see if the relays are operational from the web GUI.

To disable analog input do change GPIO17 from ``ADC Input`` to ``None`` in menu option ``Configure Template``. ALso disable the rule with command ``rule2 0``.

MyUniversalDeveloper, 2024-04-16:
Hi, thank You for your quick response! I've changed the settings but there was no benefit. The rollback to 13.4.0 works well. Is it possible to get minor versions like 13.4.0.2 or 13.4.0.3?

Jason2866, 2024-04-16:
You find any version here. It is based on day and commit SHA
https://github.com/tasmota/install/releases

arendst, 2024-04-17:
Would be great if you can pinpoint the exact version where it fails.

akerrichardt, 2024-04-17:
Shelly 1, same problem here, no manual control with physical switch.  Everything works when reverted back to 13.4.0.

arendst, 2024-04-17:
Verified toggle switch connected to Shelly 1PM broken with latest versions.

Investigating....

arendst, 2024-04-17:
Try latest dev branch. At least the Shelly 1 should work fine now. Not sure about the initial device.

MyUniversalDeveloper, 2024-04-20:
@arendst With version 13.4.0.4 the failure arrrives. @Jason2866 Thank You for your hint.

Jason2866, 2024-04-20:
@MyUniversalDeveloper Please try latest development version. It should be fixed.

Jason2866, 2024-04-28:
Closing since no feedback.

MyUniversalDeveloper, 2024-05-07:
Sorry for my late feedback. I've tried today the newest (13.4.1.2) firmware with the same failure as before. In my opinion this bug should be reopened (if this is possible).

arendst, 2024-05-07:
Pls provide more logging while pressing the individual switches. Set logging to 4 and press the switches after some seconds. There should be something like 

```
16:32:59.667-022 RUL: ANALOG#A0DIV10<30 performs 'Power3 2'
16:32:59.670-022 SRC: Rule
16:32:59.671-022 CMD: Grp 0, Cmd 'POWER', Idx 3, Len 1, Pld 2, Data '2'
16:32:59.676-019 MQT: stat/wemos14/RESULT = {"POWER3":"ON"}
16:32:59.681-020 MQT: stat/wemos14/POWER3 = ON
16:33:00.546-022 CFG: Saved to flash at FA, Count 35, Bytes 4096
16:33:05.156-022 RUL: ANALOG#A0DIV10<30 performs 'Power3 2'
16:33:05.159-022 SRC: Rule
16:33:05.161-022 CMD: Grp 0, Cmd 'POWER', Idx 3, Len 1, Pld 2, Data '2'
16:33:05.168-019 MQT: stat/wemos14/RESULT = {"POWER3":"OFF"}
16:33:05.176-020 MQT: stat/wemos14/POWER3 = OFF
16:33:05.543-022 CFG: Saved to flash at F9, Count 36, Bytes 4096
```

And pls verify if the main button still works.

mediarath, 2024-05-07:
I have just flashed a P1 with 13.4.1.2. Same template and same rule and cannot reproduce the error. My P1 is fine. At least with 13.4.1.2

Jason2866, 2024-05-07:
@mediarath Thx for testing. So the issue is specific to this one device.

Jason2866, 2024-05-07:
Moving to discussions, since not a Tasmota issue