arendst/Tasmota #21200: Range Extender driver `xdrv_58_range_extender` does not compile with core 3.0.0

Jason2866, 2024-04-16:
Details of failing can be seen in this GH Actions run https://github.com/Jason2866/Tasmota-build/actions/runs/8704050149/job/23871521878

@joba-1 can you have a look and adopt to IDF5.1 / Arduino core 3.0.0?

joba-1, 2024-04-16:
At first glance:  tcpip_adapter API discontinued and replaced by ESP-NETIF :(
See https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32/migration-guides/release-5.x/5.0/networking.html?highlight=tcp#tcp-ip-adapter

No idea how much changed. Maybe it is more or less search&replace ... and testing.

Could you please make it easy for me and send a platformio.ini or (a link to) instructions to compile locally with this new stuff? What I have working is VS Code on Linux with PlatformIO Extension and a cloned tasmota. I guess I just have to replace some platform or framework entries?

Jason2866, 2024-04-16:
Do a fresh clone of Tasmota in a different folder. Don't use your existing setup it may conflicts since the platformio setup has changed. You can use this env
```
[env:tasmota32-rangeextender]
extends                     = env:tasmota32_base
build_flags                 = ${env:tasmota32_base.build_flags}
                              -DFIRMWARE_TASMOTA32
                              -DUSE_WIFI_RANGE_EXTENDER
                              -DUSE_WIFI_RANGE_EXTENDER_NAPT
lib_ignore                  =
                              TTGO TWatch Library
                              Micro-RTSP
```

sillyfrog, 2024-05-23:
I noticed this has been marked stale. I have this on my radar, but it'll be a little bit before I'll have time to look at it properly... Let me know if you get very far in the meantime. Thanks!

joba-1, 2024-05-23:
sure, I‘ll let you know of any progress (or problems, in case you have an idea). Earliest next week, nothing new so far.

sfromis, 2024-06-17:
Unstale, builds still failing

sfromis, 2024-07-12:
Any hope of getting this feature refactored to match the API change?

sfromis, 2024-08-06:
*unstale*

sfromis, 2024-08-31:
Tickle

fdaubercy, 2024-09-12:
Hi.
I always have to run my rangeextender with a version older than the current one. Compiling with this option still doesn't work.
Frédéric

Jason2866, 2024-09-12:
@fdaubercy Thats why this issue is open ;-)

Jason2866, 2024-09-27:
Closing should be solved with #22205

closed 2024-09-27 by Jason2866

sillyfrog, 2024-09-28:
Thank you all (@Jason2866  @pharaocurt ) for helping get this sorted! (Sorry I never had a chance to get back into it).