arendst/Tasmota #21204: ESP32-C3 ETH01 EVO DM9051 wrong ETH_Type declaration in my_user_config.h

androegg, 2024-04-17:
### PROBLEM DESCRIPTION
In my_user_config.h under 
//#define USE_ETHERNET                             // Add support for ethernet (+20k code)
in detail 
#define ETH_TYPE  declaration  4 = ETH_PHY_DM9051 is wrong should be 7 = ETH_PHY_DM9051
If #define ETH_TYPE 4 Module EthType is set to 8 and ethernet dosent work at all
ETH_TYPE 7 works perfect

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): ESP32-C3 ETH01 EVO DM9051
- [x] Tasmota binary firmware version number used: Tasmota 13.4.0.3 (arduino30)](https://bit.ly/tasmota)
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: ESPTool
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2024-04-17:
Correct. Has been changed since Core3. Was 4 in Core3, Is now 7 in core3 (and latest dev)

arendst, 2024-04-17:
Try latest dev release where ETH_PHY_DM9051 no has permamently been changed from 4 to 10(!)

closed 2024-04-18 by androegg

arendst, 2024-05-05:
As of v13.4.1.2 it has been restored from 10 to 4!!!