arendst/Tasmota #21205: Core I2S Audio not working

Jason2866, 2024-04-17:
Follow up of https://github.com/arendst/Tasmota/discussions/16074#discussioncomment-9143268

Did some short tests.
- I2S micro not working (INMP441)
- mp3 stram not working

Crahes happened too.

@s-hadinger @gemu2015

Staars, 2024-04-18:
Meanwhile @rug1024 could try the following things (with `seriallog 3`):

1. Check general microphone functionality with `i2srec`, e.g.:
   ` i2srec /test.mp3` 
    (then after a few seconds)
    `i2srec` (to stop recording and finish writing)
    Download and test MP3 file.

2. Provide console output while accessing `http://IP:81/stream.mp3` from your browser.
    Please also check output of `i2sstream` and test behaviour of switching stream server on/off with  `i2sstream 0`/`i2sstream 1`.

rug1024, 2024-04-18:
On executing i2srec the device instantly crashed and does not reboot anymore.
Power cycle also not successfull, think it needs to be reflashed with Esptool.

Staars, 2024-04-19:
Okay, I have found my INMP441 and could fix the crashes with `i2srec`.
It was possible to record raw audio in Berry, but the MP3 encoder ist totally broken at the moment, so no luck with `i2srec` for now. This makes it impossible to stream MP3 for obvious reasons too, although there was another bug that would crash the ESP before the start of the (non working) stream.
There are a few more glitches.

s-hadinger, 2024-04-21:
Can you guys describe your configuration (especially template) when the crash happens?

Staars, 2024-04-21:
The crashes in `i2srec` will happen in every mic-only-configuration at the moment, because these calls can not work  without an output instance.

https://github.com/arendst/Tasmota/blob/cb70ff7a778e2e38416faefd49575144e9edb9c7/tasmota/tasmota_xdrv_driver/xdrv_42_0_i2s_audio_idf51.ino#L436
https://github.com/arendst/Tasmota/blob/cb70ff7a778e2e38416faefd49575144e9edb9c7/tasmota/tasmota_xdrv_driver/xdrv_42_0_i2s_audio_idf51.ino#L462

The i2sstream does not start the mic at the moment and crashes without starting it by hand with`i2smic 1`.

But I still only get garbage MP3 data from the encoder ... which I do not understand yet.

Staars, 2024-04-25:
After more bugfixing I have the driver working including MP3 streaming, but the INMP441 is a special animal, that seems to need an unusual I2S configuration. ATM this microphone produces garbled/distorted sound, but at least it works somehow. I have found more infos in other projects, but need to test this myself.

The good thing is, that the new driver is very flexible and can be extented quite easily to reflect very different configurations with one firmware image. You can already exchange the connected Firmware (i.e. from Standard to PDM), reconfigure it and use it without reflashing the ESP32.
IMHO it is a huge step forward.

rug1024, 2024-04-25:
Great news! Thanks for the effort!
The distortion you report has occurred here too, also in previous Tasmota-builds (afaik since 12.x).
Found out just yet: It's only there when streaming live, when recording via i2srec it's not.
I attached a sample of what it sounds like (recorded with my mobile in front of the speaker).
[audio.zip](https://github.com/arendst/Tasmota/files/15106525/audio.zip)
Next thing: the distortion is only with my ESP32**S3**, on two other ESP32**S2** the stream is clean (Tasmota 13.4.0 Master, Core/SDK Version: 2.0.14 / 4.4.6.240105).
Since these mics don't cost much, what other types can you recommend? Willing to buy and test.

Staars, 2024-04-25:
Interesting point about the different behavior of different SOC's.

I will have to check this too as currently the INMP441 is connected to a S3. In theory it should not matter, but I would not be too surprised to find the same.

My PDM micro is a cheap clone of this:
https://learn.adafruit.com/adafruit-pdm-microphone-breakout/overview

But is not extensively tested by me so far. Initial impression is good.

Staars, 2024-04-25:
INMP441 works here now for MP3 audio recording and streaming on the S3 without distortion and the expected (mediocre) sound quality.
I hope this is repeatable on other platforms too.

Please allow some time for testing and code refactoring as we need some new variables in the driver setting.

rug1024, 2024-04-25:
Great news! Ordered myself 2 of [these](https://www.ebay.de/itm/254522070947) PDM Mics for testing (like the JST-SH plug).
Will compare to INMP441 and report.

rug1024, 2024-04-26:
@Staars 
Can you please post your compiler options?
Not quite getting the PDM Mic working yet.

Staars, 2024-04-26:
Current state of changes here:
https://github.com/Staars/Tasmota/tree/i2s

There are more refactoring steps ahead, but with the next PR I would like to change as few things as possible.

```[env:tasmota32s3-i2s]
extends                     = env:tasmota32_base
board                       = esp32s3-qio_opi
build_flags                 = ${env:tasmota32_base.build_flags}
                              -DFIRMWARE_TASMOTA32
                              -DUSE_I2S
                              -DUSE_I2S_AUDIO
                              -DUSE_I2S_MIC
                              -DUSE_SHINE
                              -DMP3_MIC_STREAM
                              -DUSE_I2S_AUDIO_BERRY
                              -DOTA_URL='""'
lib_extra_dirs              = lib/libesp32, lib/libesp32_div, lib/lib_basic, lib/lib_ssl, lib/lib_i2c, lib/lib_audio
lib_ignore                  = Micro-RTSP
```

Out of the box this should work with an PDM microphone.
`i2sconfig {"Rx":{"Mode":0}}` would change to standard mode with default settings for INMP441 with left channel.
`i2sconfig {"Rx":{"SlotMask":2}}` would change this to right channel.

`I2smgain` should be set to something like 30-60. 

`i2sconfig {"Sys":{"Mp3Preallocate":1}}` is needed for both. Correct pin settings are of course needed too.

rug1024, 2024-04-27:
Will need some time for testing and trying to get this running.
Just wondering what that is for:
[image]

Staars, 2024-04-27:
Forget the current docs, the aim for the new driver was to configure "everything" at runtime.

So things like PDM, left/right, mono/stereo, ... do not need a different firmware. You will be even able to swap out the microphone to a total different type later or back and forth without reflashing.

rug1024, 2024-04-27:
I tried to compile with Gitpod using your fork, unfortunately this does not work very well.
Since I also use ESP32S3 for testing, could you send me compiled binary just to find out if my mic is working at all?

Staars, 2024-04-27:
Then please head over to Discord and find me there or use Tasmotas compiling-development channel.

Jason2866, 2024-04-29:
Tested with M5stack core 2 and i2rec works. saytime does work too.

Jason2866, 2024-05-06:
@Staars Can we close this? Everything I tested is working.

rug1024, 2024-05-06:
I can test again this evening. When we wrote in Discord it was not working. PDM and INMP441 both not working. With 13.4.0 Master the INMP441 still works fine.

Jason2866, 2024-05-06:
@gemu2015 not related to i2s but parallel display do work with core 3.0.0 and uDisplay driver.

s-hadinger, 2024-05-06:
Unfortunately I don't have any PAR8/PAR16 display... So I can't help

sfromis, 2024-05-17:
I was trying to get I2S audio to work with small 8Ω speaker connected to a board having a MAX98357A. Configured the I2S gpios for BCLK, WS and DOUT to match, and left the SD and GAIN pins unconnected, as indicated on the [external DAC wiring](https://tasmota.github.io/docs/I2S-Audio/#audio-output) suggestions in the docs. The board schematics indicate that GAIN is pulled down, and SD connected directly to the ESP32-S3, no pull-up/down, so unconfigured would be floating.

No luck with commands `I2SSay` or `I2STime`.

As MAX98357A specs says that SD_MODE pulled low mans shutdown, I decided to config the ESP32-S3 pin as `Output Hi` and now the test cases worked (not great voice quality, but I'm not very demanding).

Not knowing much about I2S audio, I'm wondering if this suggestion should just be made part of the docs, or if there are complications needing extra considerations.

Jason2866, 2024-05-17:
@sfromis imho mentioning `SD_MODE` needs a defined state should be in the docs.

sfromis, 2024-05-17:
Did a small test getting a WebRadio MP3 stream to play. Got a few seconds of garbled sound first, but then normal audio. Next step was interrupting it with the command `I2STime` (saw no clean way of stopping a stream), and got:
```
[14:34:11.626] 14:34:11.037 CMD: I2SWr http://listen.livestreamingservice.com/181-jazzmix_128k.mp3
[14:34:12.517] 14:34:11.928 I2S: Tx i2s_channel_enable err=0x0000
[14:34:13.041] 14:34:11.934 MQT: t337blng/stat/I2SWR = {"I2SWR":"http://listen.livestreamingservice.com/181-jazzmix_128k.mp3"}
[14:34:13.041] 14:34:12.452 I2S: updateClockConfig i2s_channel_disable err=0x0000
[14:34:13.043] 14:34:12.454 I2S: updateClockConfig i2s_channel_reconfig_std_clock err=0x0000
[14:34:13.044] 14:34:12.455 I2S: updateClockConfig i2s_channel_enable err=0x0000
[14:34:57.870] 14:34:57.282 CMD: i2stime
[14:34:57.880] 14:34:57.290 I2S: updateClockConfig i2s_channel_disable err=0x0000
[14:34:57.880] 14:34:57.291 I2S: Could not write samples (count=1): 259
[14:34:57.880] 14:34:57.292 I2S: Could not write samples (count=1): 259
[14:34:57.881] 14:34:57.293 I2S: Could not write samples (count=1): 259
[14:34:57.883] 14:34:57.294 I2S: Could not write samples (count=1): 259
```
Last message in a loop until I killed it (power off). First set of time stamps from my terminal program.

Staars, 2024-05-20:
I ordered some cheap I2S stuff („the usual suspects from Ali“) to be able to do more tests myself.
This includes the MAX98357A.
I still was not able to find a failing configuration with the hardware I own.

The webradio does work now much more robust than with core 2.x, but you will always run into trouble with  non optimal network conditions - which may happen outside your home.

Hopefully I can do more tests in the next 1-2 weeks to confirm, what works and how to configure it.

sfromis, 2024-05-20:
I'm indeed experiencing `I2SWr` working better now.

Next step is trying to get the microphone to work. Schematics shows that my board has a MEMS microphone, ICS-43434 with this pinout:
```
      39: Select  (pulled down, 10K)
      40: WS
      41: Data
      42: BCLK
```
No select pin in Tasmota gpio configs, I let it be floating for now.

And my Tasmota template includes:
```
Gpio.00	IO	1	User
Gpio.01	AO	7840	I2S WS1
Gpio.02	AO	7808	I2S BCLK1
Gpio.03	AO	7776	I2S DOUT1
Gpio.04	AO	3840	Output Hi  That's the mentioned SD_MODE
Gpio.40	IO	7841	I2S WS2
Gpio.41	IO	7873	I2S DIN2
Gpio.42	IO	7809	I2S BCLK2
```
This means that I have a `I2S DIN2` without having `I2S DIN1`, I was not assuming this to be bad?

At boot, I have these messages:
```
I2S: config loaded from file '/.drvset042'
I2S: Tx i2s_channel_enable err=0x0000
```
Is error code 0 an error?

Where it gets "annoying" is when doing a test with `I2SRec /test.mp3`, without having changed any `I2SConfig` settings:
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core  1 register dump:
PC      : 0x420fdf1e  PS      : 0x00060930  A0      : 0x820fdfae  A1      : 0x3fcac490  
A2      : 0x00000000  A3      : 0x3c1afb13  A4      : 0x00000016  A5      : 0x00000001  
A6      : 0x0000004c  A7      : 0x3fcac407  A8      : 0x820fdf1e  A9      : 0x3fcac430  
A10     : 0x00000003  A11     : 0x3fcb36f0  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x3fce9a70  A15     : 0x3c2c0000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000036  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffc  
Backtrace: 0x420fdf1b:0x3fcac490 0x420fdfab:0x3fcac4b0 0x42030292:0x3fcac4e0 0x420071e3:0x3fcac500 0x4202f9bc:0x3fcac530 0x42064cef:0x3fcac590 0x42009a3f:0x3fcac5b0 0x42009b1e:0x3fcac6a0 0x42019679:0x3fcac6f0 0x42063599:0x3fcac740 0x42072319:0x3fcac760 0x42072e99:0x3fcac780 0x420748fe:0x3fcac7b0 0x42016f78:0x3fcac830 0x42064cef:0x3fcac880 0x4200f030:0x3fcac8a0 0x42009ce5:0x3fcac8c0 0x420daab2:0x3fcac930
```

Any noob errors?

sfromis, 2024-05-20:
Yes, same S3 [Bling from UnexpectedMaker](https://github.com/UnexpectedMaker/bling) as when testing audio output. As it has completely separate I2S pins for amplifier and microphone, it does not sound like full duplex makes sense?

Anyway, tried setting `I2SConfig {"Sys":{"Duplex":1}}`, (only place where I saw duplex on the docs page) which would not "stick", presumably due to having separate pins configured?

Trying to record after having set `I2SConfig {"Rx":{"Mode":0}}` I'm still seeing a crash with LoadProhibited.
`"Rx":{"SampleRate":32000,"Gain":30,"Mode":0,"SlotMask":1,"SlotWidth":32,"Channels":1,"DCFilterAlpha":32511,"LowpassAlpha":17719,"APLL":1,"WsWidth":32,"WsPol":0,"BitShift":1,"LeftAlign":1,"BigEndian":0,"LsbOrder":0,"DMAFrame":512,"DMADesc":3}}`
(All configs are unchanged from first boot, except now for Rx:Mode).

Staars, 2024-05-26:
`EXCVADDR: 0x00000036` looks like the old bug. You are really using the latest development version?

Yes, duplex can not work with the Bling.

I now have a setup with microphone and output (INMP441 and PCM5102) which works, but I already found a few things and glitches to change in the future.

sfromis, 2024-05-26:
Yes, I was using latest development at that time. I saw no change after the fix for building for ESP32-S3, again crashing.

Now, few days later, I've refreshed the source as of yesterday, 20240525-195146, and updated.

The same `I2SRec /test.mp3` apparently fares better now, looks like trying to do something. But output file size is 0. My expectation was that if I needed tweaks for microphone gain or something, then it would still save something in the file. Is this a good assumption?
```
16:21:34.395 CMD: i2srec /test.mp3
16:21:34.396 SRC: WebConsole from 192.168.86.39
16:21:34.397 CMD: Grp 0, Cmd 'I2SREC', Idx 1, Len 9, Pld -99, Data '/test.mp3'
16:21:34.398 I2S: try late buffer allocation for mp3 encoder
16:21:34.399 I2S: startRx called
16:21:34.400 I2S: mic init rx_channels:1 rx_running:0 rx_handle:0x3c2c2910
16:21:34.401 I2S: RX channel enable err:0x0000
16:21:34.545 I2S: samples 1152, bytesize 2304, time: 36
16:21:34.548 MQT: t337blng/stat/I2SREC = {"I2SRec":"/test.mp3"}
16:21:34.550 !! 2048, 2304
16:21:34.585 !! 2048, 2304
16:21:34.621 !! 2048, 2304
16:21:34.960 !! 2048, 2304
16:21:34.996 !! 2048, 2304
16:21:35.032 !! 2048, 2304
16:21:35.068 !! 2048, 2304
16:21:35.156 !! 2048, 2304
16:21:35.192 !! 2048, 2304
16:21:35.228 !! 2048, 2304
16:21:35.264 !! 2048, 2304
16:21:35.300 !! 2048, 2304
16:21:35.336 !! 2048, 2304
16:21:35.372 !! 2048, 2304
16:21:35.422 !! 2048, 2304
16:21:35.685 !! 1536, 2304
16:21:35.721 !! 2048, 2304
16:21:35.757 !! 2048, 2304
16:21:35.947 !! 2048, 2304
16:21:35.983 !! 2048, 2304
16:21:36.270 I2S: MP3 file closed
16:21:36.270 I2S: calling stopRx() rx_running:1 rx_handle:0x3c2c2910
16:21:36.271 I2S: stopRx i2s_channel_disable err=0x0000
16:21:36.272 mp3task result code: 0
```

My template tries to have the MEMS microphone on I2S 2, with the amplifier on I2S 1. These pins (except for "boring" Button/SPI/neopixel pins):
```
Gpio.01	AO	7840	I2S WS1
Gpio.02	AO	7808	I2S BCLK1
Gpio.03	AO	7776	I2S DOUT1
Gpio.04	AO	3840	Output Hi
Gpio.40	IO	7841	I2S WS2
Gpio.41	IO	7873	I2S DIN2
Gpio.42	IO	7809	I2S BCLK2
```
It did not help removing the amplifier, with only the microphone as I2S 1.

An additional test shows that the I2SPlay command no longer produced any sound, after having added the microphone on I2S 2. Worked when only having I2S 1, but I did notice that the I2SStop command did not succeed, instead giving `{"I2SStop":"I2S output not configured"}`

And my older issue about loop/crash trying the `I2STime` command while `I2SPlay` was playing sound reared it's head again.
```
[16:32:32.516] 16:32:31.707 I2S: updateClockConfig i2s_channel_disable err=0x0000
[16:32:32.516] 16:32:31.707 I2S: updateClockConfig i2s_channel_reconfig_std_clock err=0x0000
[16:32:32.518] 16:32:31.709 I2S: Could not write samples (count=1): 259
[16:32:32.518] 16:32:31.709 I2S: Could not write samples (count=1): 259
[16:32:32.519] 16:32:31.710 I2S: Could not write samples (count=1): 259
[16:32:32.519] 16:32:31.710 I2S: Could not write samples (count=1): 259
```

I did double-check my timestamps for both source download and build.

Staars, 2024-05-26:
Okay, thanks for testing.
`16:21:34.550 !! 2048, 2304` this is the old setting for DMAFrame. The hidden .drvset file does not get updated automatically (no idea, if there is a 100% best way to update changes there automatically with driver updates - probably not).

`i2sconfig {"Rx":{"DMAFrame":576}}` - should be the minimum here now, but you can try a higher value.

I will take a look at the issue for i2splay preventing a start of other command. I did not touch this section of the driver yet myself.

The crash of the ESP32-S3 is completely unclear to me as I have a S3 here as testing device working happily.

sfromis, 2024-05-26:
The general approach around updates of defaults for user-configurable values is that the currently configured values are not affected by the change of defaults, unless the user chooses to do a device reset, to get all back to the defaults.

Or by removing the `.drvset042` file, to have it reinitialized. After that I indeed got `"DMAFrame":576`. Now I saw `17:20:27.757 !! 1152, 2304` but still the test.mp3 file having a size of 0 after the command completed with closing the file and stopRx.

No improvement with `I2SConfig {"Rx":{"DMAFrame":2048}}`, also not with `{"Rx":{"Mode":0}}`. Is my assumption about gain (or similar) changes only useful if it did put *sometthing* into the output mp3 file a safe one?

At least no more a crash with `I2SRec` and the crash I mentioned with the loop was after it had been looping some time. Could be a watchdog....

gemu2015, 2024-06-15:
i again did a short audio mp3play test with core3 on a plain esp32 4M with the following issues:
1. after completion of the sound, audio power stays on about 4 seconds giving some strange noise
2. playing audio in a loop crashes at random times with various exception codes.
on core 2 this works without any issues

```
>D

>BS
=>i2splay /Startup.mp3
>S
if upsecs%5==0 {
	=>i2splay /Glass.mp3
	print heap %heap%
	print stack %stack%
}
```

[sounds.zip](https://github.com/user-attachments/files/15846293/sounds.zip)

gemu2015, 2024-07-12:
@s-hadinger please review
finally i identified the bugs in audio out which caused all audio output to crash after various repeats.
within bool TasmotaI2S::beginTx(void)
this causes the crashes
```
for(int i = 0;i < 6;i++){
      i2s_channel_preload_data(_tx_handle, zero_buffer, sizeof(zero_buffer), &sz); // preload DMA buffer with silence
}
```
as stated in the espressif docs "i2s_channel_preload_data" may never be called when channel is already started which it is in this case
commenting out this line fixed the crashes and has no audible consequences

the noise after each sound (about 3 seconds) comes from
this 
```
audio_i2s_mp3.task_running = true;
  while (audio_i2s_mp3.mp3->isRunning() && audio_i2s_mp3.task_running) {
    if (!audio_i2s_mp3.mp3->loop()) {
        audio_i2s_mp3.task_running == false;
        break;
    }
    vTaskDelay(pdMS_TO_TICKS(1));
  }

```
the break is needed, otherwise isRunning() is called once again which caused the delay with noise, same problem with web radio main loop

Jason2866, 2024-07-12:
The issue is probably something different. I had/have the same problems as gemu. After enabling `I2S_ISR_IRAM_SAFE` in sdkconfig and build the Arduino Libs with this setup, the issue is gone. My S3 works perfectly fine now. No changes in code needed. No crash and Audio recording does work as expected.
Will provide the changed framework soon.

rug1024, 2024-07-12:
Great news. Thank you @ all for your work. Hope that ESP32 S2 will also work again too.

Jason2866, 2024-07-12:
Changed framework is now available and active in branch development

gemu2015, 2024-07-13:
@Jason2866 
sorry, keeps crashing.
those 2 bugs i found are really bugs and have to be solved.
with my fixes above no more crashes at all and no more noise at end of sound

Jason2866, 2024-07-13:
@gemu2015 Which device MCU do you have/use? Can't reproduce the issue with the devices i have.

gemu2015, 2024-07-13:
i get the error with an esp32 and esp32-s3 test device compiled with the latest dev
simply play an mp3 file with i2splay and repeat several times or repeat is2say hello several times e.g. in console ( or create a script to repeat the commands)
at least after a few iterations it crashes 
and after each mp3 sound you hear an annoying sound for several seconds.

the results are exactly the same on both devices.

sfromis, 2024-07-13:
As a test, I let this run for a few minutes, without seeing any crashes, on an Unexpected Maker Bling! with ESP32-S3, and latest dev built yesterday. While there was a bit of "crackling" a fraction of a second at end of each "hello", nothing to break the silence between each time.
`tasmota.add_cron('*/5 * * * * *', /->tasmota.cmd('i2ssay hello'),'h')`

```
17:22:50.248 SRC: Berry
17:22:50.249 CMD: Grp 0, Cmd 'I2SSAY', Idx 1, Len 5, Pld -99, Data 'hello'
17:22:50.250 I2S: I2SPrepareTx out=0x3fced6ec
17:22:50.251 I2S: Calling beginTX (tx_handle:1070520212 already_running:0)
17:22:50.252 I2S: Tx i2s_channel_enable err=0x0000
17:22:50.253 I2S: begin _tx_running succeeded
17:22:50.254 I2S: SetGain: 0.11
17:22:50.256 I2S: SetRate: 22050 was 22050 tx_running=1
17:22:50.257 I2S: SetBitsPerSample: 8
17:22:50.258 I2S: SetChannels: 1
17:22:50.259 I2S: Calling beginTX (tx_handle:1070520212 already_running:1)
17:22:50.721 I2S: calling stopTx() tx_running:1 tx_handle:0x3fced794
17:22:50.722 I2S: stopTx i2s_channel_disable err=0x0000
17:22:50.726 MQT: xxxxxxxxx/stat/I2SSAY = {"I2SSay":"hello"}
```

Jason2866, 2024-07-13:
@gemu2015 Last changes merged which where already in the pipline. The major issue was the ISR problem!

gemu2015, 2024-07-13:
no more crashes with this version, but audio power off is missing after mp3play,

closed 2024-07-13 by Jason2866

rug1024, 2024-08-22:
I know this issue has been closed, but audio-streaming from ESP to Browser it is still not working.
I have ESP32-S2 with INMP441 Mic and I use I2S-Streaming (Port 81). Last working version here is 13.4.0 Master.
With latest build not even the server on port 81 is opened.

Jason2866, 2024-08-22:
@rug1024 Please open a new issue. Providing exactly all data and which are asked in the template. Additionally your exact setup (wiring) GPIOs and options set.