arendst/Tasmota #21252: CORS command not working

LAMERCZ, 2024-04-22:
### PROBLEM DESCRIPTION
CORS command (mentioned here: https://tasmota.github.io/docs/Commands/#wi-fi) cannot be invoked.
Console returns: RESULT = {"Command":"Unknown"}

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff T1 1CH
- [X] Tasmota binary firmware version number used: 13.4.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: esptool
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
00:07:46.222 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
00:07:46.430 RSL: RESULT = {"Module":{"28":"Sonoff T1 1CH"}}
00:07:46.683 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
IMHO TMI - sensitive info and not relevant.

```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
00:10:07.248 CMD: CORS
00:10:07.250 SRC: WebConsole from 192.168.222.10
00:10:07.253 CMD: Grp 0, Cmd 'CORS', Idx 1, Len 0, Pld -99, Data ''
00:10:07.257 RSL: RESULT = {"Command":"Unknown"}
00:10:10.391 WIF: Checking connection...
00:10:14.116 CMD: CORS *
00:10:14.118 SRC: WebConsole from 192.168.222.10
00:10:14.121 CMD: Grp 0, Cmd 'CORS', Idx 1, Len 1, Pld -99, Data '*'
00:10:14.125 RSL: RESULT = {"Command":"Unknown"}

```

### TO REPRODUCE
Just run the command ...

### EXPECTED BEHAVIOUR
CORS should be set according to documentation. I found mention on DISCORD it was previously disable, but it was a year old message, so I guess if it is in DOC, it should be working, right?

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
N/A

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-04-22:
No issue there. Commands are only available when a particular feature is included in the build you use. And the insecure `USE_CORS` is very deliberately not included in standard precompiled builds.

LAMERCZ, 2024-04-22:
Ok, but it should be mentioned in DOC.

Jason2866, 2024-04-23:
Command removed from Documentation.

closed 2024-04-23 by Jason2866

Jason2866, 2024-04-23:
CORS is NOT safe. That's why it is removed. Documentation is Open Source too. You are invited to do an PR. Doing by just clicking the pencil and doing changes.

sfromis, 2024-04-23:
If you access Internet, or run any downloaded programs, from some computer on your local Wifi, it is **not** "safe" with CORS in Tasmota.

LAMERCZ, 2024-04-23:
If I understood CORS correctly, it just add HTTP response header "Access-Control-Allow-Origin". It is up to application if it even recognize this access control .... if not, then nothing happens either way, right? And AFAIK this is only related to web browsers, so for example tasmota works for me on firefox (windows and android), but not in chrome or edge (maybe there is some option to allow cross domain, I don't use those so I didn't dig into that).

Just to be absolutely sure, because you scared me a little:) My understanding the CORS is just about access control header, so if I install addon, greasemonkey script or whatever to my browser to add this header, it will be the same as allowing CORS on tasmota, or am I incorrect here? Because malware/virus or whatever won't follow "Access-Control-Allow-Origin" header rules, so it has nothing to do with lowered security, right?

Only possible scenario I see is if I visit some page with my browser (Firefox which allows CORS and recognize "Access-Control-Allow-Origin" correctly), which will have some script and call for example http://192.168.30.108/cm?cmnd=Power and find out if I have my bedroom light on or off;-D Which is not that bad IMHO;-D BTW, I have also button on my page http://192.168.30.108/cm?cmnd=Power%20TOGGLE which works regardless CORS .... Which would be much more annoying and possible dangerous .... but as I said, it works just fine without CORS. Only what is not working (without CORS) is getting info about ON/OFF state, which is annoying but not crucial ... 
So I don't see anything dangerous here or am I missing something?
Just for a reference (I rather changed IP ranges completely;-D), here is my HTML page (hosted on my VPS and public domain name - anyone can access this page, but since it refers to local IP ranges, it won't do anything - if not connected directly to my home wifi):

```
<meta charset="UTF-8">
<body style="background-color:black;font-color:white;">
<style>
a:link{
  color:white;
}
a:visited{
  color:white;
}
a:hover{
  color:orange;
}
a:focus{
  color:green;
}
a:active{
  color:red;
}
</style>

<font color="white">
<script src="https://ajax.googleapis.com/ajax/libs/jquery/3.4.0/jquery.min.js"></script>
</head>
<span style="font-family:arial; font-weight:bold; font-size:35px;">SONOFF zařízení
<br>----------</span><br><br>
<span style="font-family:arial; font-weight:bold; font-size:25px;">

<button type="button" onclick="$.get('http://192.168.30.106/cm?cmnd=Power%20TOGGLE',function( data ) {$('#zas_pow_1x1').html('&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;    '+data.POWER);});"><span style="font-family:arial; font-weight:bold; font-size:25px;">PRUSA MINI+</span></button><a href="http://192.168.30.106" target="_blank"><span id="zas_pow_1x1">&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;    OFFLINE</span></a>&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<a href="http://192.168.30.52">PRUSA MINI+ WEB</a>
<br><br>
----------<br><br>
<button type="button" onclick="$.get('http://192.168.30.100/cm?cmnd=Power1%20TOGGLE',function( data ) {$('#dvojv_1x1').html('&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;    '+data.POWER1);});"><span style="font-family:arial; font-weight:bold; font-size:25px;">Světlo Tomáš zářivky</span></button><a href="http://192.168.30.100" target="_blank"><span id="dvojv_1x1">&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;    OFFLINE</span></a><br><br>

<button type="button" onclick="$.get('http://192.168.30.100/cm?cmnd=Power2%20TOGGLE',function( data ) {$('#dvojv_1x2').html('&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;    '+data.POWER2);});"><span style="font-family:arial; font-weight:bold; font-size:25px;">Světlo Tomáš LED</span></button><a href="http://192.168.30.100" target="_blank"><span id="dvojv_1x2">&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;    OFFLINE</span></a><br><br>

----------<br><br>
<button type="button" onclick="$.get('http://192.168.30.105/cm?cmnd=Power%20TOGGLE',function( data ) {$('#zas_26x2').html('&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;    '+data.POWER);});"><span style="font-family:arial; font-weight:bold; font-size:25px;">KLIMA</span></button><a href="http://192.168.30.105" target="_blank"><span id="zas_26x2">&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;    OFFLINE</span></a><br><br>
<button type="button" onclick="$.get('http://192.168.30.107/cm?cmnd=Power%20TOGGLE',function( data ) {$('#zas_27x2').html('&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;    '+data.POWER);});"><span style="font-family:arial; font-weight:bold; font-size:25px;">MACBOOK</span></button><a href="http://192.168.30.107" target="_blank"><span id="zas_27x2">&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;    OFFLINE</span></a><br><br>
----------<br><br>
<button type="button" onclick="$.get('http://192.168.30.103/cm?cmnd=Power1%20TOGGLE',function( data ) {$('#dvojz_1x1').html('&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;    '+data.POWER1);});"><span style="font-family:arial; font-weight:bold; font-size:25px;">ASV</span></button><a href="http://192.168.30.103" target="_blank"><span id="dvojz_1x1">&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;    OFFLINE</span></a><br><br>
<button type="button" onclick="$.get('http://192.168.30.103/cm?cmnd=Power2%20TOGGLE',function( data ) {$('#dvojz_1x2').html('&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;    '+data.POWER2);});"><span style="font-family:arial; font-weight:bold; font-size:25px;">Lampička</span></button><a href="http://192.168.30.103" target="_blank"><span id="dvojz_1x2">&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;    OFFLINE</span></a><br><br>
----------<br><br>
[... cut ...]
$.get('http://192.168.30.101/cm?cmnd=Power',function( data ) {$('#zas_pow_2x1').html("&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;    "+data.POWER);});
$.get('http://192.168.30.103/cm?cmnd=Power1',function( data ) {$('#dvojz_1x1').html("&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;    "+data.POWER1);});
$.get('http://192.168.30.103/cm?cmnd=Power2',function( data ) {$('#dvojz_1x2').html("&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;    "+data.POWER2);});
$.get('http://192.168.30.110/cm?cmnd=Power',function( data ) {$('#vyp_1').html("&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;    "+data.POWER);});
</script>
</font>
</body>
```

Thanks.