arendst/Tasmota #21284: Incorrect power measurement on Shelly Puls PM Mini

guiumb, 2024-04-26:
### PROBLEM DESCRIPTION
I have a new device Shally Puls PM Mini with tasmota 13.4.1.2 self compiled firmware connected to 60W bulb lamp with current clamp connection in between.
I read 0.080A on the Shelly and 0.220A on the clamp and also the power is not correct on the Shelly.
As I flashed the unit with Tasmota as soon as I bought it, I have no reference with the original firmware. This is my 1st shelly plus device and I cannot verify with any other.
I have also upgraded the device with pre-compiled binary but the issue persists.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly Plus PM Mini (SNPM-001PCEU16)
- [x] Tasmota binary firmware version number used: 13.4.1.2
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: platformio
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
16:10:23.558 CMD: Backlog Template; Module; GPIO 255
16:10:23.583 MQT: stat/shellybagno/RESULT = {"NAME":"Shelly PlusPMMini","GPIO":[576,32,0,4736,0,0,3200,8161,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
16:10:23.811 MQT: stat/shellybagno/RESULT = {"Module":{"0":"Shelly PlusPMMini"}}
16:10:24.062 MQT: stat/shellybagno/RESULT = {"GPIO0":{"576":"LedLink_i"},"GPIO1":{"32":"Button1"},"GPIO2":{"0":"None"},"GPIO3":{"4736":"ADC Temp1"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"3200":"Serial Tx"},"GPIO7":{"8161":"BL0942 Rx2"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"}}
```

- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
No rules:
16:19:40.894 CMD: Backlog Rule1; Rule2; Rule3
16:19:40.920 MQT: stat/shellybagno/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
16:19:41.151 MQT: stat/shellybagno/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
16:19:41.401 MQT: stat/shellybagno/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
```

- [x] Provide the output of this command: `Status 0`:
```lua
16:20:42.520 CMD: status 0
16:20:42.525 MQT: stat/shellybagno/STATUS = {"Status":{"Module":0,"DeviceName":"Presa Bagno","FriendlyName":["Presa Bagno"],"Topic":"shellybagno","ButtonTopic":"0","Power":"1","PowerLock":"0","PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
16:20:42.532 MQT: stat/shellybagno/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32c3.bin","RestartReason":"Software reset digital core","Uptime":"0T00:00:17","StartupUTC":"2024-04-26T14:20:25","Sleep":50,"CfgHolder":4617,"BootCount":6,"BCResetTime":"2024-04-26T11:44:34","SaveCount":16}}
16:20:42.538 MQT: stat/shellybagno/STATUS2 = {"StatusFWR":{"Version":"13.4.1.2(tasmota32)","BuildDateTime":"2024-04-26T11:39:33","Core":"3_0_0","SDK":"5.1.3.240421","CpuFrequency":160,"Hardware":"ESP32-C3 v0.4","CR":"477/699"}}
16:20:42.546 MQT: stat/shellybagno/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["guilan",""],"TelePeriod":60,"Resolution":"FE8181C0","SetOption":["0020A00B","2805C80000000600003C5A0A192800000000","0000A280","00006000","00006000","00000001"]}}
16:20:42.561 MQT: stat/shellybagno/STATUS4 = {"StatusMEM":{"ProgramSize":1945,"Free":806,"Heap":153,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":80,"FlashMode":"DIO","Features":["0809","9F8AD4DF","0005A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80B1","00000005"],"Drivers":"1,2,3,!4,!5,!8,9,10,11,!14,!16,!17,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,!71,82,!86,!87,!88,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
16:20:42.570 MQT: stat/shellybagno/STATUS5 = {"StatusNET":{"Hostname":"shellybagno-2612","IPAddress":"10.0.0.203","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer1":"10.0.0.1","DNSServer2":"0.0.0.0","Mac":"54:32:04:BB:6A:34","IP6Global":"","IP6Local":"fe80::5632:4ff:febb:6a34%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"10.0.0.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":19.0}}
16:20:42.581 MQT: stat/shellybagno/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.150","MqttPort":1883,"MqttClientMask":"shellybagno","MqttClient":"shellybagno","MqttUser":"umbyha","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
16:20:42.592 MQT: stat/shellybagno/STATUS7 = {"StatusTIM":{"UTC":"2024-04-26T14:20:42Z","Local":"2024-04-26T16:20:42","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":99,"Sunrise":"06:18","Sunset":"20:24"}}
16:20:42.603 MQT: stat/shellybagno/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0,0,0,0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
16:20:42.616 MQT: stat/shellybagno/STATUS10 = {"StatusSNS":{"Time":"2024-04-26T16:20:42","ANALOG":{"Temperature1":52.223},"ENERGY":{"TotalStartTime":"2024-04-26T11:45:22","Total":0.01930,"Yesterday":0.01930,"Today":0.00000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":50,"Voltage":216,"Current":0.000},"ESP32":{"Temperature":58.400},"TempUnit":"C"}}
16:20:42.628 MQT: stat/shellybagno/STATUS11 = {"StatusSTS":{"Time":"2024-04-26T16:20:42","Uptime":"0T00:00:17","UptimeSec":17,"Heap":152,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":23,"Objects":300},"Wifi":{"AP":1,"SSId":"guilan","BSSId":"3C:A6:2F:08:7B:F7","Channel":1,"Mode":"HT20","RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```

- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
16:58:10.936 CMD: weblog 4
16:58:10.940 MQT: stat/shellybagno/RESULT = {"WebLog":4}
16:58:11.131 CFG: Saved, Count 19, Bytes 4096
16:58:11.725 BL9: Rx 55000000767130097C014900002F02001E4E00200100AE
16:58:12.783 BL9: Rx 558A0200E56930A57C013A00002F02001E4E002001002E
16:58:13.748 BL9: Rx 550D0500E455306A7B018000002F02001E4E00020100D1
16:58:14.748 BL9: Rx 55EE0300D15F30987D016100002F02001E4E00000100EC
16:58:15.754 BL9: Rx 551A0200A36630C77B013300002F02001E4E00220100C7
16:58:16.712 BL9: Rx 551706006B6730267D01AF00002F02001E4E0020010022
16:58:17.714 BL9: Rx 55A50400386630E07C018200002F02001E4E002001003E
16:58:18.715 BL9: Rx 552F0500EC5D30077D017A00002F02001E4E00220100E6
16:58:19.721 BL9: Rx 551A0500E35E304D7A018900002F02001E4E00020100D1
16:58:20.722 BL9: Rx 55F20500864130187C01B800002F02001E4E0002010077
16:58:21.727 BL9: Rx 557D0600415330697B01E300002F02001E4E00000100A5
16:58:22.726 BL9: Rx 55B50100FB7030AA7E012600002F0200164E000001001C
16:58:23.731 BL9: Rx 555A0400528230807D017100002F02001E4E00200100C3
[... 24 lines cut ...]
16:58:44.735 BL9: Rx 55974E00AD2E319A7E016F27002F0200264E00200100EC
16:58:45.741 BL9: Rx 553D4E002F2A31FF7D014027002F0200264E0020010093
16:58:46.741 BL9: Rx 55164F000C2331347E01A627002F02001E4E002001004F
16:58:47.744 BL9: Rx 55754E006D1C31BE7E014F27002F02001E4E0000010084
[... cut ...]
16:58:52.711 BL9: Rx 554B4E00BF8230DE7E01BF26002F02001E4E0020010048
16:58:53.714 BL9: Rx 553E4E00918E30778101C426002F02001E4E00200100D6
16:58:54.714 BL9: Rx 551F4E00C98F303F8201B526002F02001E4E0020010002
16:58:55.721 BL9: Rx 55414E000B8030FB7F017C26002F02001E4E000001004D
16:58:56.721 BL9: Rx 55E84D00C29830D080019E26002F02001E4E00000100E0
```

### TO REPRODUCE

### EXPECTED BEHAVIOUR

### SCREENSHOTS

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-04-26:
Did you try the calibration?
https://tasmota.github.io/docs/Power-Monitoring-Calibration/

guiumb, 2024-04-26:
Yes, but the value is sitll incorect.
```

20:38:34.709 MQT: tele/shellybagno/SENSOR = {"Time":"2024-04-26T20:38:34","ANALOG":{"Temperature1":53.273},"ENERGY":{"TotalStartTime":"2024-04-26T11:45:22","Total":0.07622,"Yesterday":0.01930,"Today":0.05692,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":50,"Voltage":205,"Current":0.000},"ESP32":{"Temperature":58.400},"TempUnit":"C"}
20:39:20.563 CMD: powerset 60.0
20:39:20.567 MQT: stat/shellybagno/RESULT = {"PowerSetCal":165}
20:39:34.722 MQT: tele/shellybagno/STATE = {"Time":"2024-04-26T20:39:34","Uptime":"0T04:19:09","UptimeSec":15549,"Heap":154,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":23,"Objects":300},"Wifi":{"AP":1,"SSId":"guilan","BSSId":"3C:A6:2F:08:7B:F7","Channel":1,"Mode":"HT20","RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:04"}}
20:39:34.731 MQT: tele/shellybagno/SENSOR = {"Time":"2024-04-26T20:39:34","ANALOG":{"Temperature1":53.273},"ENERGY":{"TotalStartTime":"2024-04-26T11:45:22","Total":0.07643,"Yesterday":0.01930,"Today":0.05713,"Period":0,"Power":17,"ApparentPower":17,"ReactivePower":0,"Factor":1.00,"Frequency":50,"Voltage":205,"Current":0.085},"ESP32":{"Temperature":58.400},"TempUnit":"C"}
20:40:34.698 MQT: tele/shellybagno/STATE = {"Time":"2024-04-26T20:40:34","Uptime":"0T04:20:09","UptimeSec":15609,"Heap":154,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":23,"Objects":300},"Wifi":{"AP":1,"SSId":"guilan","BSSId":"3C:A6:2F:08:7B:F7","Channel":1,"Mode":"HT20","RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:04"}}
20:40:34.706 MQT: tele/shellybagno/SENSOR = {"Time":"2024-04-26T20:40:34","ANALOG":{"Temperature1":53.273},"ENERGY":{"TotalStartTime":"2024-04-26T11:45:22","Total":0.07712,"Yesterday":0.01930,"Today":0.05782,"Period":1,"Power":17,"ApparentPower":17,"ReactivePower":0,"Factor":1.00,"Frequency":50,"Voltage":206,"Current":0.085},"ESP32":{"Temperature":58.400},"TempUnit":"C"}
20:40:53.711 CMD: voltageset 220
20:40:53.716 MQT: stat/shellybagno/RESULT = {"VoltageSetCal":14403}
20:41:34.720 MQT: tele/shellybagno/STATE = {"Time":"2024-04-26T20:41:34","Uptime":"0T04:21:09","UptimeSec":15669,"Heap":153,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":23,"Objects":300},"Wifi":{"AP":1,"SSId":"guilan","BSSId":"3C:A6:2F:08:7B:F7","Channel":1,"Mode":"HT20","RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:04"}}
20:41:34.729 MQT: tele/shellybagno/SENSOR = {"Time":"2024-04-26T20:41:34","ANALOG":{"Temperature1":53.315},"ENERGY":{"TotalStartTime":"2024-04-26T11:45:22","Total":0.07785,"Yesterday":0.01930,"Today":0.05855,"Period":1,"Power":19,"ApparentPower":19,"ReactivePower":0,"Factor":1.00,"Frequency":50,"Voltage":221,"Current":0.086},"ESP32":{"Temperature":59.400},"TempUnit":"C"}
```

sfromis, 2024-04-26:
What you post shows that you did **not** perform the calibration

guiumb, 2024-04-27:
Sorry for the incomplete cut and paste: this is the complete one.
```
09:23:57.524 CMD: powerset 58
09:23:57.528 MQT: stat/shellybagno/RESULT = {"PowerSetCal":153}
09:24:01.784 CMD: voltageset 213
09:24:01.789 MQT: stat/shellybagno/RESULT = {"VoltageSetCal":14318}
09:24:06.545 CMD: currentset 220
09:24:06.549 MQT: stat/shellybagno/RESULT = {"CurrentSetCal":251213}
09:24:32.690 MQT: tele/shellybagno/STATE = {"Time":"2024-04-27T09:24:32","Uptime":"0T17:04:07","UptimeSec":61447,"Heap":154,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":2,"Berry":{"HeapUsed":23,"Objects":300},"Wifi":{"AP":1,"SSId":"guilan","BSSId":"3C:A6:2F:08:7B:F7","Channel":1,"Mode":"HT20","RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:04"}}
09:24:32.698 MQT: tele/shellybagno/SENSOR = {"Time":"2024-04-27T09:24:32","ANALOG":{"Temperature1":52.766},"ENERGY":{"TotalStartTime":"2024-04-26T11:45:22","Total":0.08050,"Yesterday":0.05938,"Today":0.00182,"Period":1,"Power":16,"ApparentPower":16,"ReactivePower":0,"Factor":1.00,"Frequency":50,"Voltage":214,"Current":0.075},"ESP32":{"Temperature":58.400},"TempUnit":"C"}
```

My doubt is that currentcal is higher than 32000.

guiumb, 2024-04-27:
I would like to apologise to everyone: the incorrect readings were caused by incorrect wiring. Thank you for your patience @sfromis.

closed 2024-04-27 by guiumb