arendst/Tasmota #21297: ESP32 and PZEM-004T v3 working intermittently

fmarzocca, 2024-04-28:
### PROBLEM DESCRIPTION
I have connected a PZEM-004t-v3 to ESP32 (Tasmota 13.4.0(solo1)). R8 where lowered.

It works intermittently for few hours, then it display all zeros in the energy fields. I have double-checked the wiring and it is consistent. I am using hardware serial and connected PZEM to ESP's RX/TX GPIOs.

When not displaying energy data, there is no error in WebLog 4.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): ESP32 DevKit
- [ ] Tasmota binary firmware version number used: 13.4.0(solo1)
- [ ]   - [X] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:54:57.153 MQT: stat/sen6-pdu/RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
15:54:57.392 MQT: stat/sen6-pdu/RESULT = {"Module":{"1":"ESP32-DevKit"}}
15:54:57.650 MQT: stat/sen6-pdu/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"1472":"PZEM0XX Tx"},"GPIO2":{"0":"None"},"GPIO3":{"1536":"PZEM016 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"227":"Relay4"},"GPIO14":{"226":"Relay3"},"GPIO15":{"235":"Relay12"},"GPIO16":{"234":"Relay11"},"GPIO17":{"233":"Relay10"},"GPIO18":{"232":"Relay9"},"GPIO19":{"231":"Relay8"},"GPIO20":{"0":"None"},"GPIO21":{"230":"Relay7"},"GPIO22":{"229":"Relay6"},"GPIO23":{"228":"Relay5"},"GPIO24":{"0":"None"},"GPIO25":{"224":"Relay1"},"GPIO26":{"225":"Relay2"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

15:55:30.372 MQT: stat/sen6-pdu/STATUS = {"Status":{"Module":1,"DeviceName":"Sen6-PDU","FriendlyName":["Sen6-PDU2","Sen6-PDU3","Sen6-PDU","","","","",""],"Topic":"sen6-pdu","ButtonTopic":"0","Power":3096,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
15:55:30.382 MQT: stat/sen6-pdu/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32solo1.bin","RestartReason":"Vbat power on reset","Uptime":"0T20:19:09","StartupUTC":"2024-04-27T17:36:21","Sleep":50,"CfgHolder":4617,"BootCount":90,"BCResetTime":"2024-02-28T13:45:53","SaveCount":1122}}
15:55:30.392 MQT: stat/sen6-pdu/STATUS2 = {"StatusFWR":{"Version":"13.4.0(solo1)single-core","BuildDateTime":"2024-02-14T16:17:44","Core":"2_0_14","SDK":"4.4.6.240105","CpuFrequency":240,"Hardware":"ESP32-D0WD-V3 v3.1","CR":"457/699"}}
15:55:30.397 MQT: stat/sen6-pdu/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Vodafone-saltydog","VG_Network_2Ghz"],"TelePeriod":0,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
15:55:30.411 MQT: stat/sen6-pdu/STATUS4 = {"StatusMEM":{"ProgramSize":1899,"Free":20,"Heap":150,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"162085","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80B1","00000004"],"Drivers":"1,2,3,!4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,82,!86,!87,!88,121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
15:55:30.419 MQT: stat/sen6-pdu/STATUS5 = {"StatusNET":{"Hostname":"sen6-PDU","IPAddress":"10.0.0.108","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer1":"10.0.0.1","DNSServer2":"0.0.0.0","Mac":"E8:6B:EA:C9:C9:48","IP6Global":"fd00::ea6b:eaff:fec9:c948","IP6Local":"fe80::ea6b:eaff:fec9:c948%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"10.0.0.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":16.0}}
15:55:30.428 MQT: stat/sen6-pdu/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.0.26","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C9C948","MqttUser":"DVES_USER","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:55:30.434 MQT: stat/sen6-pdu/STATUS7 = {"StatusTIM":{"UTC":"2024-04-28T13:55:30Z","Local":"2024-04-28T15:55:30","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":99,"Sunrise":"06:33","Sunset":"21:01"}}
15:55:30.438 MQT: stat/sen6-pdu/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0,0,0,0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
15:55:30.452 MQT: stat/sen6-pdu/STATUS10 = {"StatusSNS":{"Time":"2024-04-28T15:55:30","ENERGY":{"TotalStartTime":"2024-04-22T09:39:11","Total":2.826,"Yesterday":0.000,"Today":0.074,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Frequency":0,"Voltage":0,"Current":0.000}}}
15:55:30.460 MQT: stat/sen6-pdu/STATUS11 = {"StatusSTS":{"Time":"2024-04-28T15:55:30","Uptime":"0T20:19:09","UptimeSec":73149,"Heap":153,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"Berry":{"HeapUsed":4,"Objects":46},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"ON","POWER5":"ON","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF","POWER9":"OFF","POWER10":"OFF","POWER11":"ON","POWER12":"ON","Wifi":{"AP":2,"SSId":"VG_Network_2Ghz","BSSId":"04:B4:FE:09:BC:D1","Channel":7,"Mode":"11n","RSSI":100,"Signal":-50,"LinkCount":2,"Downtime":"0T00:00:05"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
15:55:30.372 MQT: stat/sen6-pdu/STATUS = {"Status":{"Module":1,"DeviceName":"Sen6-PDU","FriendlyName":["Sen6-PDU2","Sen6-PDU3","Sen6-PDU","","","","",""],"Topic":"sen6-pdu","ButtonTopic":"0","Power":3096,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
15:55:30.382 MQT: stat/sen6-pdu/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32solo1.bin","RestartReason":"Vbat power on reset","Uptime":"0T20:19:09","StartupUTC":"2024-04-27T17:36:21","Sleep":50,"CfgHolder":4617,"BootCount":90,"BCResetTime":"2024-02-28T13:45:53","SaveCount":1122}}
15:55:30.392 MQT: stat/sen6-pdu/STATUS2 = {"StatusFWR":{"Version":"13.4.0(solo1)single-core","BuildDateTime":"2024-02-14T16:17:44","Core":"2_0_14","SDK":"4.4.6.240105","CpuFrequency":240,"Hardware":"ESP32-D0WD-V3 v3.1","CR":"457/699"}}
15:55:30.397 MQT: stat/sen6-pdu/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Vodafone-saltydog","VG_Network_2Ghz"],"TelePeriod":0,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","000 [... cut ...] hz","BSSId":"04:B4:FE:09:BC:D1","Channel":7,"Mode":"11n","RSSI":100,"Signal":-50,"LinkCount":2,"Downtime":"0T00:00:05"}}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2024-04-28:
Can't reproduce the issue. Having a Esp32 with a PZEM-04 module running since months without any problems. Have you replaced the hardware parts already?

Jason2866, 2024-04-28:
Moving to discussions since it is no Tasmota issue.