arendst/Tasmota #21305: Add a new ZbState called ZbLoaded to indicate that all ZbDevices and ZbLoads have been loaded successfully.

tasict, 2024-04-30:
I hope to add a new state to indicate that all ZbDevices and ZbLoads have been loaded successfully.

sfromis, 2024-04-30:
I's say that it should be safe to use the existing `{"ZbState":{"Status":0,"Message":"Started"}}` with an extra one-second interval for the plugin data to be loaded from the local file system.

tasict, 2024-04-30:
No, at this point, the ZB device and zb driver have not actually finished loading yet. If ZbState#Status=0 is added in the rule, calling Zbinfo at this time will fail to display information properly. It requires a delay of about 2 to 3 seconds before calling zbinfo. This is exactly why there is a need to add a new state in reality.

s-hadinger, 2024-04-30:
Got it. I will add it

s-hadinger, 2024-04-30:
Or move Started event after the plug-ins are loaded, which sounds better

tasict, 2024-04-30:
Indeed, that would be perfect!

PR #21320 (2024-05-02): Zigbee startup event triggered after plugins are loaded
## Description:

Now the Zigbee startup event are sent after plugins are loaded

```
19:51:39.364 MQT: tele/tasmota/ZBBridge/RESULT = {"ZbState":{"Status":51,"IEEEAddr":"0x00124B002A2E916B","ShortAddr":"0x0000","DeviceType":7,"DeviceState":9,"NumAssocDevices":2,"AssocDevicesList":["0xE7D6","0x040B"]}}
19:51:39.937 ZIG: Zigbee device information found in File System (12 devices - 616 bytes)
19:51:39.957 ZIG: Zigbee device data in File System (441 bytes)
19:51:39.966 ZIG: ZbLoad '<internal_plugin>' loaded successfully
19:51:40.064 ZIG: ZbLo [... cut]