arendst/Tasmota #21311: A4988_Stepper , Motor commands doesnt give feedback while called without argument. Motor MIS not working ?

ouinouin, 2024-05-01:
### PROBLEM DESCRIPTION
Motor commands doesnt give feedback while called without argument when stepper motor is confgured
MotorMIS commmand not working ? 

### REQUESTED INFORMATION
- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): atom lite
- [ ] Tasmota binary firmware version number used: 14.4.0
  - [ ] Pre-compiled
  - [x ] Self-compiled
- [x] Flashing tools used: webusb
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
`11:57:11.774 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
11:57:11.987 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
11:57:12.246 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"3744":"A4988 STP"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"3776":"A4988 ENA"},"GPIO23":{"3712":"A4988 DIR"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}`
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
issue some of the motor commands without argument, unlike  many if not all commands in tasmota, this doesnt gives back the actual value (letting user think something is wrong)

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
have a feedback of the actual set value while issuing the command without argument.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
```
11:40:45.310 CMD: MotorMIS
11:40:45.314 RSL: RESULT = {"Command":"Error"}
11:40:53.311 CMD: MotorSPR 1
11:40:53.315 RSL: RESULT = {"MotorSPR":"Done"}
11:40:56.561 CMD: MotorSPR 200
11:40:56.565 RSL: RESULT = {"MotorSPR":"Done"}
11:41:16.662 CMD: motorrpm
11:41:16.666 RSL: RESULT = {"Command":"Error"}
11:41:19.612 CMD: motorrpm 20
11:41:19.616 RSL: RESULT = {"MotorRPM":"Done"}
11:41:58.464 CMD: motormis
11:41:58.468 RSL: RESULT = {"Command":"Error"}
11:42:01.164 CMD: motormis 1
11:42:01.168 RSL: RESULT = {"Command":"Error"}
11:42:27.216 CMD: motormsi
11:42:27.224 RSL: RESULT = {"Command":"Unknown"}
11:42:29.969 CMD: motormsi 1
11:42:29.976 RSL: RESULT = {"Command":"Unknown"}
11:42:35.819 CMD: motorMIS 1
11:42:35.824 RSL: RESULT = {"Command":"Error"}
11:42:38.472 CMD: motorMIS30
11:42:38.477 RSL: RESULT = {"Command":"Error"}
```

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
on my experiments, i wasnt able to do anything with the motorMIS command, , returning error with or without arguments.

**(Please, remember to close the issue when the problem has been addressed)**

ouinouin, 2024-05-01:
[image]
could the problem with the command motorMIS be because of a difference of variable name between the header and the cpp file ?  : OneToSixteen in onse side oneToSixteen on the other side (moreover the naming is quite wrong since drv 8825 can do microsteppings up to 1/32. :-)

Jason2866, 2024-05-01:
Will change since it is not align. Tbh idk if this will solve.

Jason2866, 2024-05-19:
Did the change https://github.com/arendst/Tasmota/commit/50979c328c11e48bbbd9b2abaa814843d448f281
solves the issue?

ouinouin, 2024-05-19:
@Jason2866 : i m not yet around my system, i ll give a try witin 2 weeks and post the outcome.
> Did the change [50979c3](https://github.com/arendst/Tasmota/commit/50979c328c11e48bbbd9b2abaa814843d448f281) solves the issue?

ouinouin, 2024-06-12:
after getting external help, i could see that : motorMIS command response depends from the connection of all 3 pins to drive the microstep settings on the template, if you connect only one or two, this doesnt work.

for the motor command, there is no answer programmed,feedback i was misinterpretating the way tasmota cuts the commands between a prefix and a suffix, and was thinking that the prefix had to  always be a valid command.

so finally not really a bug maybe just mising details in doc, i ll close it.

closed 2024-06-12 by ouinouin