arendst/Tasmota #21314: Berry script not returning sensors values in a given condition

rp2019433, 2024-05-01:
### PROBLEM DESCRIPTION

Board: ESP32

Sensors attached to ESP32:
* Analog1
* Analog2
* BME680 
* SGP40 

[image]

When running the Berry script below in version 13.4.0, Tasmota does not return information for any sensor.

```
import json
var sensors = json.load(tasmota.read_sensors())
print(sensors)
```

[image]

If I downgrade version to 13.0.0, Tasmota returns some information, but **without SGP sensor**:

[image]

If I remove BME680 and SGP40, and run Berry script in version 13.4.0, Analog sensors values are returned:

[image]

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
15:39:40.263 CMD: Backlog Template; Module; GPIO 255
15:39:40.288 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
15:39:40.531 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
15:39:40.760 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"640":"I2C SDA1"},"GPIO22":{"608":"I2C SCL1"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"4705":"ADC Input2"},"GPIO35":{"4704":"ADC Input1"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Described above.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

When running commands below, all sensors values should be returned.

```
import json
var sensors = json.load(tasmota.read_sensors())
print(sensors)
```

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

Screenshots inserted above.

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-05-01:
That `nil` could be a symptom of invalid JSON, but you did not show the output from `tasmota.read_sensors()` without also doing the `json.load()`.

rp2019433, 2024-05-01:
Hi. 
I forgot to mention that. ` tasmota.read_sensors()` without `json.load()` works properly.

```
import json
var sensors=tasmota.read_sensors()
print(sensors)
```

[image]

Jason2866, 2024-05-02:
Closing since already fixed in development branch

closed 2024-05-02 by Jason2866