arendst/Tasmota #21350: Web-UI does not load (v13.4.1.2 on ESP8266EX)

VanillaFord, 2024-05-04:
### PROBLEM DESCRIPTION

Upgraded two Shelly1PM from 13.2.0 to 13.4.1.2 (minimal) via Web/local download
- step 1: upgrade to minimal version -> works ok
- step 2: upgrade to full version (tasmota-sensors) works, but Web-UI will not load.

the device does run fine and sensor readings are published via mqtt, though.
After downgrade to v 13.4.0, using mqtt commmands, the Web-UI is back like normal.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Shelly 1PM (ESP8266EX)
- [X] Tasmota binary firmware version number used: 13.4.1.2
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: OTA / Web-UI
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

upgrade ESP8266EX (Shelly1PM) to v13.4.1.2

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Web-UI should be accessible via Browser in v 13.4.1.2

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

001Tycoon, 2024-05-05:
Same problem here with an ESP8285N08 power outlet.
Update from 13.4.1.2 2024.04.30 to 13.4.1.2 2024.05.05 tasmota-DE.bin.gz
A restart when accessing the website
tasmota-minimal.bin.gz everything is fine.

barbudor, 2024-05-05:
Reproduced too
Development version is impacted since Saturday 17:00 but the faulty commit still need to be identified.
I've seen the issue on both ESP8266 and ESP32
The ESP actually crashes on LoadProhibited when trying to send a part of the web page.
Latest dev binaries not impacted are here https://github.com/tasmota/install/releases/tag/2024-05-04.d.599526a90c0bb54f6547c26ff4332577654549c5

s-hadinger, 2024-05-05:
The dev team is on it. Please wait a few more hours

001Tycoon, 2024-05-05:
c4e847d Thanks, the version works again

s-hadinger, 2024-05-05:
Fixed

closed 2024-05-05 by s-hadinger