arendst/Tasmota #21379: KNX Support : Latest commit in development branch introduced a regression when using DHT11 sensor

pbrinette, 2024-05-06:
### PROBLEM DESCRIPTION

The last commit to solve the issue https://github.com/arendst/Tasmota/issues/20834 introduce a regression : 

Since this commit, the whole KNX configuration is lost after reboot when configuring a DHT11 sensor as KNX Source.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ✅ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ✅ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ✅ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ✅ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ✅ ]  Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ✅ ] Device used (e.g., Sonoff Basic): ESP32-DevKit + DHT11 Sensor
- [ ✅ ] Tasmota binary firmware version number used:  [Tasmota 13.4.1.2]
  - [ ] Pre-compiled
  - [ ✅ ] Self-compiled
- [ ✅ ] Flashing tools used: Vscode/plateformIO
- [ ✅  ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
21:49:20.908 CMD: Backlog Template; Module; GPIO 255
21:49:20.935 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
21:49:21.158 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
21:49:21.416 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"1184":"DHT11"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [ ❎ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ✅ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

21:54:32.323 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_26128C","ButtonTopic":"0","Power":"1","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
21:54:32.329 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:17:14","StartupUTC":"2024-05-06T20:37:18","Sleep":50,"CfgHolder":4617,"BootCount":12,"BCResetTime":"2024-04-09T22:43:26","SaveCount":43}}
21:54:32.332 RSL: STATUS2 = {"StatusFWR":{"Version":"13.4.1.2(tasmota32)","BuildDateTime":"2024-05-06T22:34:55","Core":"3_0_0","SDK":"5.1.3.240430","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.1","CR":"428/699"}}
21:54:32.335 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["OUIFI",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["02008009","2805C80001000600001234567800000000","00000080","00006000","00004000","00000000"]}}
21:54:32.354 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":2015,"Free":864,"Heap":144,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80B1","00000004"],"Drivers":"1,2,!3,!4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,82,!86,!87,!88,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
21:54:32.363 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-26128C-4748","IPAddress":"192.168.0.40","Gateway":"192.168.0.254","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.254","DNSServer2":"0.0.0.0","Mac":"08:D1:F9:26:12:8C","IP6Global":"2a01:e0a:1234:5678:1234:5678:1234:5678","IP6Local":"fe80::ad1:f9ff:fe26:128c%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.0.254","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
21:54:32.370 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_26128C","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:54:32.375 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-05-06T20:54:32Z","Local":"2024-05-06T21:54:32","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"05:20","Sunset":"20:12"}}
21:54:32.382 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-05-06T21:54:32","DHT11":{"Temperature":0.2,"Humidity":2.0,"DewPoint":-41.2},"TempUnit":"C"}}
21:54:32.386 BRY: GC from 8025 to 4212 bytes, objects freed 11/45 (in 1 ms) - slots from 51/91 to 50/91
21:54:32.391 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-05-06T21:54:32","Uptime":"0T00:17:14","UptimeSec":1034,"Heap":142,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":4,"Objects":45},"Wifi":{"AP":1,"SSId":"OUIFI","BSSId":"DC:00:B0:FE:D2:C0","Channel":1,"Mode":"HT20","RSSI":52,"Signal":-74,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [ ✅ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
21:53:09.309 CMD: weblog 4 
21:53:09.310 SRC: WebConsole from 192.168.0.36
21:53:09.312 CMD: Grp 0, Cmd 'WEBLOG', Idx 1, Len 2, Pld 4, Data '4 '
```

### TO REPRODUCE

#### If I **don't configure the sensor** as a KNX object, the **configuration is persistant** after the reboot.

I.E : Configure 2 GA 
 
```
13:06:06.021 HTP: Configure KNX
13:06:06.023 KNX: Enabled: 1, Communication Enhancement: 0
13:06:06.024 KNX: Physical Address: 0.2.178
13:06:06.026 KNX: GA: 2
13:06:06.027 KNX: GA #1: KNX TX 1 to 0/6/110
13:06:06.029 KNX: GA #2: KNX TX 2 to 0/6/111
13:06:06.030 KNX: CB: 0
13:06:06.031 HTP: Configuration
```
Configuration is still present after reboot

#### When I configure the DHT11 sensor to send data on a  KNX GA, the **configuration is lost** after the reboot.

Adding 2 GA for Temperature and Humidity sensors

```
13:10:29.454 HTP: Configure KNX
13:10:29.456 KNX: Enabled: 1, Communication Enhancement: 0
13:10:29.458 KNX: Physical Address: 0.2.178
13:10:29.460 KNX: GA: 4
13:10:29.461 KNX: GA #1: KNX TX 1 to 0/6/110
13:10:29.462 KNX: GA #2: KNX TX 2 to 0/6/111
13:10:29.464 KNX: GA #3: Temperature to 0/6/106
13:10:29.466 KNX: GA #4: Humidity to 0/6/107
13:10:29.467 KNX: CB: 0
13:10:29.468 HTP: Configuration
```

This configuration works as expected until reboot . 

After reboot, **All KNX configuration is lost** except the physical address 

```
13:15:21.965 HTP: Configure KNX
13:15:21.968 KNX: Enabled: 1, Communication Enhancement: 0
13:15:21.970 KNX: Physical Address: 0.2.178
13:15:21.972 KNX: GA: 0
13:15:21.973 KNX: CB: 0
13:15:21.975 HTP: Configuration
```

### EXPECTED BEHAVIOUR

The KNX configuration should persist after the reboot even using a DHT11 sensor. 

### SCREENSHOTS

#### Configuration BEFORE reboot when using a DHT11 sensor 
[image]

#### Configuration AFTER reboot when using a DHT11 sensor
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2024-05-06:
Thanks @pbrinette 

In the meantime I have identified the reason
In the previous PR, I removed some code which was making the KNX driver limited to a short list of T&H sensor based on direct GPIO. My change would have allowed to support any T&H sensor including I2C sensors. Some configuration was done later, on the 1st Teleperiod.
But this was not counting on a configuration test performed at the start that was erasing the configuration if some GA was found for sensors that were not yet declared.

I already submit a PR #21378 where I remove totally the test.
I think this is safe to assume that if the user has configured a GA for a temperature sensor (or humidity), then he has connected such sensor.

closed 2024-05-10 by arendst

barbudor, 2024-05-10:
@pbrinette original behavior has now been reverted
That should solve your configuration erasure problem