arendst/Tasmota #21408: Display SSD1306 not Working after Upgrade to Tasmota 14.0.0 an Universal Driver

helmar74, 2024-05-14:
### PROBLEM DESCRIPTION
After Updating my ESP8266 with Display SSD1306 to Tasmota 14.0.0 the display isn't working anymore.
I compiled myself, as I also want to use SR04 Sensor and display the measured value on the display.
Here's my user_conf_override.h
```
#undef USE_DOMOTICZ

#undef USE_HOME_ASSISTANT

#undef USE_EMULATION_HUE

#undef USE_EMULATION_WEMO

#undef ROTARY_V1

#undef USE_SHUTTER

#undef USE_DEEPSLEEP

#undef USE_EXS_DIMMER

#undef USE_WS2812_HARDWARE
#define USE_WS2812_HARDWARE  NEO_HW_SK6812     // Hardware type (NEO_HW_WS2812, NEO_HW_WS2812X, NEO_HW_WS2813, NEO_HW_SK6812, NEO_HW_LC8812, NEO_HW_APA106, NEO_HW_P9813)

#undef USE_WS2812_CTYPE
#define USE_WS2812_CTYPE     NEO_GRBW           // Color type (NEO_RGB, NEO_GRB, NEO_BRG, NEO_RBG, NEO_RGBW, NEO_GRBW)

#undef USE_ADC_VCC
#define USE_ADC_VCC                              // Display Vcc in Power status. Disable for use as Analog input on selected devices

//#undef USE_DISPLAY
//#define USE_DISPLAY                            // Add I2C Display Support (+2k code)

#undef USE_SPI
#define USE_SPI                                  // Hardware SPI using GPIO12(MISO), GPIO13(MOSI) and GPIO14(CLK) in addition to two user selectable GPIOs(CS and DC)

//#undef USE_DISPLAY_ILI9341                  // [DisplayModel 4] Enable ILI9341 Tft 480x320 display (+19k code)

#undef USE_DISPLAY
#define USE_DISPLAY                            // Add SPI Display support for 320x240 and 480x320 TFT
// Universal Display Driver
#define USE_UNIVERSAL_DISPLAY
// Universal Display Driver braucht Script
#undef USE_RULES 
#define USE_SCRIPT

#undef USE_SR04
#define USE_SR04                                 // Add support for HC-SR04 ultrasonic devices (+1k code)
```

I created a Script on the Device which looks like this:
```
>D
:H,SSD1306,128,64,1,I2C,3c,*,*,*
:S,0,2,1,0,30,20
:I
AE
D5,80
A8,3F
D3,00
40
8D,14
20,00
A1
C8
DA,12
81,9F
D9,F1
DB,40
A4
A6
AF
:o,AE
:O,AF
:A,00,10,40,00,00
:i,A6,A7
>B
->displayreinit
#
```

I also configured a GPIO with "A" Option

After that the display is dark
I tried script> dt[] commands in console, but nothing shows up

What I'm doing wrong?

Thanks for you help in advance
### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Wemos Mini D with SSD1306 and SR04
- [X] Tasmota binary firmware version number used: 14.0.0
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: PlatformIO
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-05-14:
The [docs says](https://tasmota.github.io/docs/Universal-Display-Driver/#step-5-configure-the-displayini-descriptor-file) that you can have the `display.ini` content in a special `>d` section in the script, which is different from what you used, `>D`.

helmar74, 2024-05-14:
so i have to use "lower case" <d ?

sfromis, 2024-05-14:
Yes, lower case `>d` is what the docs says.

helmar74, 2024-05-14:
OK, change the script to:
```
>D
>d
:H,SSD1306,128,64,1,I2C,3c,*,*,*
:S,0,2,1,0,30,20
:I
AE
D5,80
A8,3F
D3,00
40
8D,14
20,00
A1
C8
DA,12
81,9F
D9,F1
DB,40
A4
A6
AF
:o,AE
:O,AF
:A,00,10,40,00,00
:i,A6,A7

>B
->displayreinit
#
```

but it still not wokring...

helmar74, 2024-05-14:
Config of Device:
[image]

sfromis, 2024-05-14:
Better follow the docs...
> Set an unused GPIO to Option A3

That's the dropdown to the right, for the `3`.

helmar74, 2024-05-14:
ok got it running now. Sorry for being so stupid ;-)
just another question.
I used Displaytext commands with the old driver. is this still possible with universal display driver?
My display is showing the values, but not in a nice way...
Is there any documentation regarding displaying text on the display?

sfromis, 2024-05-14:
Not used that combo, but as a starting point I'd expect `DisplayText` to still work, with the usual docs for that.... https://tasmota.github.io/docs/Displays/#displaytext

helmar74, 2024-05-14:
unfourtnately not.
I tried DisplayText [x50y50r200:100] but nothing happens on the display.
Some further informations:

```
21:57:59.372 CMD: displaymode
21:57:59.378 MQT: stat/sonoffWemosMini_6/RESULT = {"DisplayMode":0}
21:58:03.378 CMD: displaymodel
21:58:03.383 MQT: stat/sonoffWemosMini_6/RESULT = {"DisplayModel":17}
```

helmar74, 2024-05-14:
And the docs for universal display driver shows a 404:
[image]

sfromis, 2024-05-14:
Docs (which I already linked to): https://tasmota.github.io/docs/Universal-Display-Driver/
However, that's not where you find `DisplayText`.

When it comes to that, I'd suggest you to post the full set of commands which you already had working, but which fails now.

helmar74, 2024-05-14:
OK, this command worked before and now shows rubbish on the display:
```
displaytext [z]
DisplayText [x0y0r128:64]
DisplayText [x10y2s2f1]Salzstand
displaytext [x30y19s2]53cm
displaytext [x30y37s2]56%
displaytext [x5y54s1]14.05.2024 22:23
```

The text is shown "overlapping" on the display

sfromis, 2024-05-14:
When using those commands on ESP8266 Tasmota 14.0 with SSD1306 + universal display, what I get does not look too bad. 
[image]
The yellow at the top is due to my SSD1306 having a band of yellow pixels, still monochrome.

helmar74, 2024-05-15:
ok, good to hear. did you compile with the same options as I did?

helmar74, 2024-05-15:
> displaytext [z]
> DisplayText [x0y0r128:64]
> DisplayText [x10y2s2f1]Salzstand
> displaytext [x30y19s2]53cm
> displaytext [x30y37s2]56%
> displaytext [x5y54s1]14.05.2024 22:23

did you also configure the display with script... or did you do it with rule 3?
Could you please send me the script or rule you used, please?

helmar74, 2024-05-15:
I build up a testdevice with the display and my compiled version with the user_conf settings above. The display is recognized by esp8266 but is dark and I can't send any text to it. I used the script above for configuration and made the same settings.. here some output of the device:
```
18:54:30.019 CMD: displaymode
18:54:30.025 MQT: stat/sonoffStandard_08C581/RESULT = {"DisplayMode":0}
18:54:34.006 CMD: displaymodel
18:54:34.011 MQT: stat/sonoffStandard_08C581/RESULT = {"DisplayModel":17}
18:54:39.247 MQT: tele/sonoffStandard_08C581/STATE = {"Time":"2024-05-15T18:54:39","Uptime":"0T00:50:10","UptimeSec":3010,"Vcc":3.062,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"SSID","BSSId":"9C:C7:A6:F7:72:A2","Channel":6,"Mode":"11n","RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:04"}}
18:56:41.454 CMD: i2cscan
18:56:41.474 MQT: stat/sonoffStandard_08C581/RESULT = {"I2CScan":"Device(s) found at 0x3c"}
18:56:51.530 CMD: backlog displaytext [z]; DisplayText [x0y0r128:64]; DisplayText [x10y2s2f1]Salzstand; displaytext [x30y19s2]53cm; displaytext [x30y37s2]56%; displaytext [x5y54s1]14.05.2024 22:23
18:56:51.610 MQT: stat/sonoffStandard_08C581/RESULT = {"DisplayText":"[z]"}
18:56:51.819 MQT: stat/sonoffStandard_08C581/RESULT = {"DisplayText":"[x0y0r128:64]"}
18:56:52.022 MQT: stat/sonoffStandard_08C581/RESULT = {"DisplayText":"[x10y2s2f1]Salzstand"}
18:56:52.271 MQT: stat/sonoffStandard_08C581/RESULT = {"DisplayText":"[x30y19s2]53cm"}
18:56:52.522 MQT: stat/sonoffStandard_08C581/RESULT = {"DisplayText":"[x30y37s2]56%"}
18:56:52.773 MQT: stat/sonoffStandard_08C581/RESULT = {"DisplayText":"[x5y54s1]14.05.2024 22:23"}
18:56:59.652 CMD: status 0
18:56:59.658 MQT: stat/sonoffStandard_08C581/STATUS = {"Status":{"Module":18,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"sonoffStandard_08C581","ButtonTopic":"0","Power":"1","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
18:56:59.662 MQT: stat/sonoffStandard_08C581/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"sonoffStandardGruppe","OtaUrl":"http://192.168.1.50:8080/static/firmware/sonoffStandard.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:52:30","StartupUTC":"2024-05-15T06:04:29","Sleep":50,"CfgHolder":4617,"BootCount":62,"BCResetTime":"2024-03-01T12:03:54","SaveCount":93,"SaveAddress":"F7000"}}
18:56:59.665 MQT: stat/sonoffStandard_08C581/STATUS2 = {"StatusFWR":{"Version":"14.0.0(tasmota)","BuildDateTime":"2024.05.14 19:53:47","Boot":31,"Core":"2_7_6","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"407/699"}}
18:56:59.669 MQT: stat/sonoffStandard_08C581/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SSID",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
18:56:59.680 MQT: stat/sonoffStandard_08C581/STATUS4 = {"StatusMEM":{"ProgramSize":674,"Free":328,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["0407","87AAC58F","042E82A1","020000CF","010013C0","C0007081","00004024","00001000","54000020","00000080","00000000"],"Drivers":"1,2,!3,!4,!5,!6,!8,9,10,12,13,!16,!18,!19,!22,!24,26,!35,!37,!45,62,!68","Sensors":"1,3,4,5,6,22","I2CDriver":"3,5,7,47"}}
18:56:59.684 MQT: stat/sonoffStandard_08C581/STATUS5 = {"StatusNET":{"Hostname":"sonoffStandard-08C581-1409","IPAddress":"192.168.1.177","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.2","DNSServer2":"0.0.0.0","Mac":"BC:DD:C2:08:C5:81","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:56:59.688 MQT: stat/sonoffStandard_08C581/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.50","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_08C581","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:56:59.692 MQT: stat/sonoffStandard_08C581/STATUS7 = {"StatusTIM":{"UTC":"2024-05-15T06:56:59Z","Local":"2024-05-15T18:56:59","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":99,"Sunrise":"05:27","Sunset":"20:42"}}
18:56:59.696 MQT: stat/sonoffStandard_08C581/STATUS10 = {"StatusSNS":{"Time":"2024-05-15T18:56:59"}}
18:56:59.701 MQT: stat/sonoffStandard_08C581/STATUS11 = {"StatusSTS":{"Time":"2024-05-15T18:56:59","Uptime":"0T00:52:30","UptimeSec":3150,"Vcc":3.061,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"SSID","BSSId":"9C:C7:A6:F7:72:A2","Channel":6,"Mode":"11n","RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```