arendst/Tasmota #21411: Confict warning in xsnss_53_sml.ino

HofeBY, 2024-05-15:
### PROBLEM DESCRIPTION

Compiling for tasmota32 & tasmota32c3 & tasmota32c6 & tasmota32s3:

|-- Wire @ 2.0.0
|-- XPT2046_Touchscreen
Building in release mode
Compiling .pio\build\tasmota32\src\tasmota.ino.cpp.o
D:/Tasmota/tasmota/tasmota_xsns_sensor/xsns_53_sml.ino:2741:40: warning: ignoring attribute 'section (".iram1.91")' because it conflicts with previous 'section (".iram1.7")' [-Wattributes]
 2741 | void IRAM_ATTR SML_CounterIsr(void *arg) {
      |                                        ^
D:/Tasmota/tasmota/tasmota.ino:5094:16: note: previous declaration here
Linking .pio\build\tasmota32\firmware.elf
lto-wrapper.exe: warning: using serial compilation of 30 LTRANS jobs
lto-wrapper.exe: note: see the '-flto' option documentation for more information

Compiling for tasmota32c2:

Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\AudioOutputSTDIO.cpp.o' from cache
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\AudioOutputSerialWAV.cpp.o' from cache
D:/Tasmota/tasmota/tasmota_xsns_sensor/xsns_53_sml.ino:2741:40: warning: ignoring attribute 'section (".iram1.27")' because it conflicts with previous 'section (".iram1.5")' [-Wattributes]
 2741 | void IRAM_ATTR SML_CounterIsr(void *arg) {
      |                                        ^
D:/Tasmota/tasmota/tasmota.ino:5094:16: note: previous declaration here
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\AudioOutputULP.cpp.o' from cache
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\driver\SinglePinI2SDriver.cpp.o' from cache
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libflac\bitmath.c.o' from cache
D:/Tasmota/tasmota/tasmota_xsns_sensor/xsns_53_sml.ino: In function 'void SML_Init()':
D:/Tasmota/tasmota/tasmota_xsns_sensor/xsns_53_sml.ino:3471:22: error: 'TWAI_TIMING_CONFIG_25KBITS' was not declared in this scope
 3471 |           t_config = TWAI_TIMING_CONFIG_25KBITS();
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libflac\bitreader.c.o' from cache
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libflac\cpu.c.o' from cache
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libflac\crc.c.o' from cache
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libflac\fixed.c.o' from cache
D:/Tasmota/tasmota/tasmota_xsns_sensor/xsns_53_sml.ino:3474:22: error: 'TWAI_TIMING_CONFIG_50KBITS' was not declared in this scope
 3474 |           t_config = TWAI_TIMING_CONFIG_50KBITS();
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libflac\float.c.o' from cache
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libflac\format.c.o' from cache
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libflac\lpc.c.o' from cache
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libflac\md5.c.o' from cache
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libflac\memory.c.o' from cache
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libflac\stream_decoder.c.o' from cache
D:/Tasmota/tasmota/tasmota_xsns_sensor/xsns_53_sml.ino:3477:22: error: 'TWAI_TIMING_CONFIG_100KBITS' was not declared in this scope
 3477 |           t_config = TWAI_TIMING_CONFIG_100KBITS();
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libflac\window.c.o' from cache
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libhelix-aac\aacdec.c.o' from cache
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libhelix-aac\aactabs.c.o' from cache
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libhelix-aac\bitstream.c.o' from cache
D:/Tasmota/tasmota/tasmota_xsns_sensor/xsns_53_sml.ino:3480:22: error: 'TWAI_TIMING_CONFIG_125KBITS' was not declared in this scope
 3480 |           t_config = TWAI_TIMING_CONFIG_125KBITS();
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libhelix-aac\buffers.c.o' from cache

Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libhelix-aac\dct4.c.o' from cache
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libhelix-aac\decelmnt.c.o' from cache
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libhelix-aac\dequant.c.o' from cache
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libhelix-aac\fft.c.o' from cache
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libhelix-aac\filefmt.c.o' from cache
D:/Tasmota/tasmota/tasmota_xsns_sensor/xsns_53_sml.ino:3483:22: error: 'TWAI_TIMING_CONFIG_250KBITS' was not declared in this scope
 3483 |           t_config = TWAI_TIMING_CONFIG_250KBITS();
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libhelix-aac\huffman.c.o' from cache
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libhelix-aac\hufftabs.c.o' from cache
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libhelix-aac\imdct.c.o' from cache
Retrieved `.pio\build\tasmota32c2\lib8b6\ESP8266Audio\libhelix-aac\noiseless.c.o' from cache
D:/Tasmota/tasmota/tasmota_xsns_sensor/xsns_53_sml.ino:3486:22: error: 'TWAI_TIMING_CONFIG_500KBITS' was not declared in this scope
 3486 |           t_config = TWAI_TIMING_CONFIG_500KBITS();
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~
[... 51 lines cut ...]
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Fresh installed Visual Studio Code V1.89.1 on Windows 10
Tasmota dev 14.0.0.1
Compiling for several types of ESP written above CompilerLog

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2024-05-15:
C2 is not (yet) supported with scripter

Jason2866, 2024-05-15:
@gemu2015 Please have a look to fix the warning
```
/tasmota/tasmota_xsns_sensor/xsns_53_sml.ino:2741:40: warning: ignoring attribute 'section (".iram1.27")' because it conflicts with previous 'section (".iram1.5")' [-Wattributes]
 2741 | void IRAM_ATTR SML_CounterIsr(void *arg) {
      |                                        ^
/Users/hans/Git/Tasmota/tasmota/tasmota.ino:5094:16: note: previous declaration here
```

HofeBY, 2024-05-15:
> C2 needs the setting `NO_USE_SML_CANBUS` since there is no CAN BUS support from the MCU. The Audio libs needs to be added to the lib ignore section for the C2. With this setup scripting does work to compile with the C2.

OK.
Updating files from #21407 to my active source

I also add

#if defined ESP32-C2
#define NO_USE_SML_CANBUS
#endif

to file user_config_override.h

Files tasmota32c2(.factory).bin are now being compiled but still with warnings.

These warnings in CompilerLog now are here:

Building in release mode
Compiling .pio\build\tasmota32c2\src\tasmota.ino.cpp.o
Compiling .pio\build\tasmota32c2\lib78f\WiFiHelper\WiFiHelper_ESP32.cpp.o
D:/Tasmota/tasmota/tasmota_xsns_sensor/xsns_53_sml.ino:2741:40: warning: ignoring attribute 'section (".iram1.27")' because it conflicts with previous 'section (".iram1.5")' [-Wattributes]
 2741 | void IRAM_ATTR SML_CounterIsr(void *arg) {
      |                                        ^
D:/Tasmota/tasmota/tasmota.ino:5094:16: note: previous declaration here
Compiling .pio\build\tasmota32c2\lib664\default\be_modtab.c.o
Archiving .pio\build\tasmota32c2\lib78f\libWiFiHelper.a

...

Creating esp32c2 image...
Merged 2 ELF sections
Successfully created esp32c2 image.
esp32_create_combined_bin([".pio\build\tasmota32c2\firmware.bin"], [".pio\build\tasmota32c2\firmware.elf"])
Read partitions from  partitions/esp32_partition_app2880k_fs320k.csv
//# Name,     Type,   SubType,   Offset,    Size,   Flags
nvs         data    nvs        0x9000     0x5000
otadata     data    ota        0xe000     0x2000
safeboot    app     factory    0x10000    0xD0000
app0        app     ota_0      0xE0000    0x2D0000
spiffs      data    spiffs     0x3B0000   0x50000
Creating filesystem with content:
No files added -> will NOT create littlefs.bin and NOT overwrite fs partition!
safeboot binary already in place.
    Offset | File
 -  0x0000 | D:\Tasmota\.pio\build\tasmota32c2\bootloader.bin
 -  0x8000 | D:\Tasmota\.pio\build\tasmota32c2\partitions.bin
 -  0xe000 | C:\Users\Station\.platformio\packages\framework-arduinoespressif32\tools\partitions\boot_app0.bin
 -  0x10000 | D:\Tasmota\variants/tasmota\tasmota32c2-safeboot.bin
 - 0xe0000 | D:\Tasmota\.pio\build\tasmota32c2/firmware.bin
esptool.py v4.7.2
SHA digest in image updated
Wrote 0x2d4ca0 bytes to file D:\Tasmota\.pio\build\tasmota32c2/firmware.factory.bin, ready to flash to offset 0x0
bin_map_copy([".pio\build\tasmota32c2\firmware.bin"], [".pio\build\tasmota32c2\firmware.elf"])
map_gzip([".pio\build\tasmota32c2\firmware.bin"], [".pio\build\tasmota32c2\firmware.elf"])
firm_metrics([".pio\build\tasmota32c2\firmware.bin"], [".pio\build\tasmota32c2\firmware.elf"])
C:\Users\Station\.platformio\penv\Scripts\python.exe -m tasmota_metrics "D:\Tasmota\firmware.map"
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\Station\.platformio\penv\Lib\site-packages\tasmota_metrics\__main__.py", line 4, in <module>
    main()
  File "C:\Users\Station\.platformio\penv\Lib\site-packages\tasmota_metrics\core.py", line 508, in main
    detected_target, segments, sections = load_map_data(args.map_file)
                                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Station\.platformio\penv\Lib\site-packages\tasmota_metrics\core.py", line 237, in load_map_data
    detected_chip = detect_target_chip(map_file)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Station\.platformio\penv\Lib\site-packages\tasmota_metrics\core.py", line 307, in detect_target_chip
    raise RuntimeError('Target not detected')
RuntimeError: Target not detected
*** Error 1
== [SUCCESS] Took 138.19 seconds ==
Environment    Status    Duration
-------------  --------  ------------
tasmota32c2    SUCCESS   00:02:18.190
== 1 succeeded in 00:02:18.190 ==
*  Terminal will be reused by tasks, press any key to close it.

arendst, 2024-05-15:
@gemu2015 warning just fixed.

HofeBY, 2024-05-15:
Thx at Support Team.

#21411 installed.
Compiler warnings are fixed.

closed 2024-05-15 by HofeBY