arendst/Tasmota #21444: Shutter + tilt matter device causes Tasmota to hang

maltic42, 2024-05-19:
When adding a shutter + tilt matter device Tasmota hangs after about 20 seconds. When adding a shutter device instead, everything works, the shutter is published as a matter device and works with Homekit and Google Home. After deleting the matter device (shutter) and adding the same as shutter + tilt Tasmota hangs, still singable, but the web interface and the system does not respond anymore.

Configuration output here (done from the same system with shutter configured, NOT shutter + tilt, because this system hangs)

STATUS 0 output here:

21:41:01.629 CMD: Status 0
21:41:01.633 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota_9A4B90","ButtonTopic":"0","Power":"00","PowerLock":"00","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
21:41:01.678 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:07:18","StartupUTC":"2024-05-19T20:33:43","Sleep":50,"CfgHolder":4617,"BootCount":32,"BCResetTime":"2024-02-06T22:16:00","SaveCount":61}}
21:41:01.705 RSL: STATUS2 = {"StatusFWR":{"Version":"14.0.0(release-tasmota32)","BuildDateTime":"2024-05-14T13:54:09","Core":"3_0_0","SDK":"5.1.3.240430","CpuFrequency":160,"Hardware":"ESP32-D0WDQ6 v1.1","CR":"408/699"}}
21:41:01.731 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TOAD",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C80001000600003C5A0A192800000000","40000080","00006000","00004000","00000020"]}}
21:41:01.770 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":2016,"Free":863,"Heap":72,"StackLowMark":2,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80B1","00000004"],"Drivers":"1,2,!3,!4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,82,!86,!87,!88,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
21:41:01.844 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-9A4B90-2960","IPAddress":"192.168.178.120","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"2003:f8:e706:5600:de15:c8ff:fe0b:1897","Mac":"08:3A:8D:9A:4B:90","IP6Global":"2003:f8:e706:5600:a3a:8dff:fe9a:4b90","IP6Local":"fe80::a3a:8dff:fe9a:4b90%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.178.1","DNSServer2":"2003:f8:e706:5600:de15:c8ff:fe0b:1897","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
21:41:01.906 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9A4B90","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:41:01.926 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-05-19T20:41:01Z","Local":"2024-05-19T21:41:01","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"05:02","Sunset":"20:30"}}
21:41:01.946 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-05-19T21:41:01","AM2301":{"Temperature":21.9,"Humidity":60.8,"DewPoint":14.0},"Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":90},"TempUnit":"C"}}
21:41:01.968 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-05-19T21:41:01","Uptime":"0T00:07:18","UptimeSec":438,"Heap":73,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":30,"MqttCount":0,"Berry":{"HeapUsed":49,"Objects":577},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"TOAD","BSSId":"F4:2E:7F:C7:3E:C2","Channel":1,"Mode":"HT20","RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:03"}}}
21:41:02.007 RSL: STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"1","TiltConfig":[0,90,40,90,0]}}}

### TO REPRODUCE
_Steps to reproduce the behavior:_

Configure a shutter according to the documentation

SetOption80 1
ShutterMode 0
Interlock Off
Interlock On
Interlock 1,2
ShutterMode 1

Add a shutter matter device with parameter 0 - add the device to Apple or Google - works.

Delete the device in Tasmota and Apple and Google.

Add a new shutter + tilt device with parameter 0 - the Tasmota device hangs after about 20 seconds.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

It is expected that the matter device is published and the Tasmota device does not hang.

maltic42, 2024-05-20:
Update: seems to be working in 13.4.0.

maltic42, 2024-05-20:
2nd Update: As stated above a matter shutter device in 14.0.0 does not cause Tasmota to hang. I updated to 14.0.0.1 in the meantime. After some time the shutter device (without tilt)  is shown as offline in Google Home, although Tasmota is still working. After restarting Tasmota and waiting for two minutes the devices is shown as online again in Google Home.

maltic42, 2024-05-22:
In 13.4.0 the shutter matter device stays online in Google Home, seems to be a problem in 14.0.0 and 14.0.0.1.

s-hadinger, 2024-05-22:
Thanks. Can you provide logs at level 3 `weblog 3` or `seriallog 3` when it's working and when it's not working ov v14.0.0.1?

maltic42, 2024-05-22:
Sorry for the questions: 1. There is not much history buffer in the weblog - can I extend the number of lines? I did not find anything related to that. 2. I have two problems - I can reproduce the first one (shutter + tilt) on 14.0.0.1 and e.g. 13.4.0, means keeping 13.4.0 running for at least the amount of time until 14.0.0.1 hangs. Is this what you need? Regarding problem two - it seems to take some time (maybe hours) until the device is shown as offline. Do you want me to keep 14.0.0.1 running and to send the weblog?

closed 2024-06-21 by github-actions[bot]