arendst/Tasmota #21445: Bloetooth BT BLE devives not recognized / found by Tasmota @ V 14.0.0

Rushmed, 2024-05-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
No bluetooth devices found on Tasmota V14.
V13.4 was working fine.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: 14.0.0_____
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA_____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
All bt deviches (mi senors, ibeacons) need to be recognized.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
Chech this discussion:
[https://github.com/arendst/Tasmota/discussions/21328](https://github.com/arendst/Tasmota/discussions/21328)

**(Please, remember to close the issue when the problem has been addressed)**

Mark-Muc, 2024-05-20:
Hello . I have nearly the same problem.
I compile a latest version (14.0.0.1 dev) and run it on a ESP32 device (Sonoff SwitchMan 2C) configurated as shutter.
After i updated to 14.0.0.1 dev. The device getting (respond) getting terribly slow and NO BLE devices show up. Switch back to 13.4.0 (stable) everything works fine and the BLE devices show up.
Compile with:
[image]

Regards
Mark_Muc

kroon040, 2024-05-20:
Same  issue here.  I have more devices with bluetooth and all lost connection with tasmota 14, reverting back to 13.4 resolves the issue.

Jason2866, 2024-05-20:
@btsimonh Can you have a look?

btsimonh, 2024-05-20:
looks like NimBLE may have some issues to be resolved to work well with core 3.0.0
https://github.com/h2zero/NimBLE-Arduino/issues/645

I think we should wait to see how @h2zero progresses....

Jason2866, 2024-05-20:
@btsimonh wrong repo. Tasmotpa does not use NimBLE-Arduino anymore.
Moved to https://github.com/h2zero/esp-nimble-cpp.
The issue you mentioned is not relevant for esp-nimble-cpp

Jason2866, 2024-05-20:
There are some small changes made to make the BLE driver compile with Arduino Core 3.0.0 and h2zero lib esp-nimble-cpp. Take a look at the commits.

btsimonh, 2024-05-20:
yep, spotted it.  NimBLE-arduino had a separate scan end callback, set in a different way.  Your commits meant that it was never called.  Just testing a change now, so let's see....
(the scan end callback caused the start of another scan.  so basically, currently scan once at start, and never again - and it's the scanning which hears adverts)

closed 2024-05-20 by arendst

Rushmed, 2024-05-20:
Thanks a lot!
When will the fix be available in precompiled files. Next version?

Jason2866, 2024-05-20:
@Rushmed It is already here in section `development` https://tasmota.github.io/install/

btsimonh, 2024-05-21:
@Rushmed @kroon040 @Mark-Muc - pls report success here.
For me, the mods showed my one pvv temp+hum sensor which was close, and 6 EQ3, 3 of which connection failed because of low signal, but the other three happily read valve position, etc. indicating connections success.  (EQ3 connection is always a little dodgy - an ideal device to stress tas's BLE stack! ).
I didn't test aliasing or any other features - but nothing has changed there, so all should be good.

kroon040, 2024-05-21:
@btsimonh ,

I got 4 EQ3 devices and 4 LYWSD03MMC devices. All the devices are working, also valve position and desired temp. Also aliasing also still works. I See one device he just needs to try 3 times and than it connects.

Mark-Muc, 2024-05-21:
Just to let you know. I also built today a version in the dev. brache and all BLE derives are visible.
one slight problem: 
if u use a "." (dot) in the aliasing . like:
`BLEAlias E72E0050A0CC=1.OG_Bad A4C1382D29C3=1.OG_Arbeit_2`

it shows in the UI like that:
[image]
[image]

as u can see it shows a "," (coma) instead of a "." (dot)
it is also in the version: 13.4.0 final.
Mark_Muc

arendst, 2024-05-21:
That's the auto decimal kicking in. It replaces a dot by a comma for Germany :-)

Let me see if it can be easily fixed....

arendst, 2024-05-21:
Try latest dev.

btsimonh, 2024-05-21:
I wish CSV import in Excel different countries was fixed so easily and quickly!

Mark-Muc, 2024-05-21:
but i do have an other problem.
not belog to the "bluetooth bug".
I foud out, that he newer versions of tasmota now connect my device (Sonoff SwitchMan 2C) now connect to my AP in 
`Mode HT40`
[image]

bevor it connected to my AP in 
`Mode 11n`
[image]

I think that will be the reason why it device is laging like hell.
The informiation of the device:
[image]

if it's not the right place .. pls be so kind to move it!
Mark_Muc

Jason2866, 2024-05-21:
The connection mode is most probably not the reason for your devices being laggy.
It is Bluetooth. Bluetooth and WiFi together enabled on an Esp32 is always problematic.

Mark-Muc, 2024-05-22:
> Well core3 does support both n (=HT20) and HT40 depending on your wifi AP. Reconfig your AP to HT20 and it should connect to HT20 (at least that's what my ubiquiti does.)

seems that this is not the issue .. fount out.
If u use a fresh intalles 32bit devive . flashed it with the webflasher included the final with bluetooth.
it works twith the network and no lagging.
BUT
if i restore my config .. the it lagging like hell.
Found out:
If u upload to the filesystem it happens.
i.e. I upload a file called: autoexec.bat the device hangs / lagging.
**autoexec.bat**
```
Mi32Option6 1
MI32Page 10
MI32Block A4C138465A57
MI32Block E7764500F442
BLEAlias 150311116C66=LYWSD02-1 A4C138AF4A0B=Temp&Hum(LYWSD03MMC) E72E0050A0CC=1.OG_Bad A4C138465A57=Pavillion_innen A4C1382D29C3=1.OG_Arbeit_2 A4C1389C4D2A=Thermostat
shutteropen
```

asap it handgs after upload.

It is fine in 13.4.0!!
Regards Mark

Rushmed, 2024-05-29:
> @Rushmed It is already here in section `development` https://tasmota.github.io/install/

Thanks, how can I download the .bin file? I have to flash OTA.

Jason2866, 2024-05-29:
Via official Tasmota OTA server https://ota.tasmota.com/tasmota32/

Rushmed, 2024-05-29:
> @Rushmed @kroon040 @Mark-Muc - pls report success here. For me, the mods showed my one pvv temp+hum sensor which was close, and 6 EQ3, 3 of which connection failed because of low signal, but the [...]

[Tasmota 14.0.0.3 (b61ea17-bluetooth)](https://bit.ly/tasmota) is working as expected. Thanks Sir!

AleXSR700, 2024-10-05:
I do not know if this is related, but after upgrading my EQ3 devices are not longer connecting reliably.
[image]

I have 8 Xiaomi devices and 8 EQ3s. The Xiaomis (running ATC firmware) seem to work but the EQ3s just show `20:44:07.951 EQ3: 001A2208C9FA: trv operation failed - retrying -12` (with the respective device ID).

Does anybody have similar issues?