arendst/Tasmota #21504: Latest code compilation error

xsp1989, 2024-05-27:
I am trying to build Tasmota 14.0.0 by myself, but I encountered an error stating that it cannot find the 'AsyncUDP.h' header file. However, this header file already exists on my local PC. How should I troubleshoot this issue? It seems like I haven't set up the dependencies correctly. Which file should I modify to add this dependency library?

And I also compiled the code on gitpod and got the same error. At the same time, I also compiled the latest development branch and got the same error.

```
Building in release mode
Compiling .pio\build\tasmota32-ZB-GW03-EN\src\tasmota.ino.cpp.o
In file included from E:/TMP/Tasmota/tasmota/tasmota_xdrv_driver/xdrv_01_9_webserver.ino:64:
C:/Users/renzhe/.platformio/packages/framework-arduinoespressif32/libraries/DNSServer/src/DNSServer.h:2:10: fatal error: AsyncUDP.h: No such file or directory

******************************************************************
* Looking for AsyncUDP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:AsyncUDP.h"
* Web  > https://registry.platformio.org/search?q=header:AsyncUDP.h
*
******************************************************************

    2 | #include <AsyncUDP.h>
      |          ^~~~~~~~~~~~
compilation terminated.
*** [.pio\build\tasmota32-ZB-GW03-EN\src\tasmota.ino.cpp.o] Error 1
Compiling .pio\build\tasmota32-ZB-GW03-EN\libc99\DNSServer\DNSServer.cpp.o
In file included from C:/Users/renzhe/.platformio/packages/framework-arduinoespressif32/libraries/DNSServer/src/DNSServer.cpp:1:
C:/Users/renzhe/.platformio/packages/framework-arduinoespressif32/libraries/DNSServer/src/DNSServer.h:2:10: fatal error: AsyncUDP.h: No such file or directory

```

corbo, 2024-05-27:
In ./platformio_override.ini comment out the line for ESP32 Async UDP in the section for [env:tasmota32_base] lib_ignore              
part:

```
[env:tasmota32_base]
; *** Uncomment next lines ";" to enable development Tasmota Arduino version ESP32
;platform                = https://github.com/tasmota/platform-espressif32.git
;platform_packages       = framework-arduinoespressif32 @ https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/1271/framework-arduinoespressif32-lwip_timeout-ed6742e7f0.zip
;                          framework-arduino-solo1 @ https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/1273/framework-arduinoespressif32-solo1-release_v4.4-804d12ce82.zip
;                          framework-arduino-ITEAD @ https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/1272/framework-arduinoespressif32-ITEAD-release_v4.4-804d12ce82.zip
;build_unflags           = ${esp32_defaults.build_unflags}
;build_flags             = ${esp32_defaults.build_flags}
;board                   = esp32
;board_build.f_cpu       = 240000000L
;board_build.f_flash     = 40000000L
;board_build.flash_mode  = qio
;board_build.flash_size  = 8MB
;board_upload.maximum_size = 8388608
;board_upload.arduino.flash_extra_images =
;board_build.partitions  = partitions/esp32_partition_app2944k_fs2M.csv
; *** Serial port used for erasing/flashing the ESP32
;upload_port             = COM4
;monitor_port            = COM4
;upload_speed            = 115200
monitor_speed           = 115200
upload_resetmethod      = ${common.upload_resetmethod}
lib_ignore              =
                          HTTPUpdateServer
                          ESP RainMaker
                          WiFiProv
                          USB
                          SPIFFS
                          ESP32 Azure IoT Arduino
                        ;   ESP32 Async UDP
                          ESP32 BLE Arduino
;                          SimpleBLE
                          NetBIOS
                          ESP32
                          Preferences
                          BluetoothSerial
; Disable next if you want to use ArduinoOTA in Tasmota32 (default disabled)
                          ArduinoOTA

lib_extra_dirs           = ${library.lib_extra_dirs}

```

xsp1989, 2024-05-27:
Thank you very much, I think it should be fine.

Jason2866, 2024-05-27:
Closing, since the commented entry does not exist in actual files.

closed 2024-05-27 by Jason2866