arendst/Tasmota #21506: SonOff MiniR4 looses MQTT subscription /out

Hans-Sh, 2024-05-27:
### PROBLEM DESCRIPTION
Sonoff MiniR4 (ESP32-D0WD-V3 v3.1) serial flashed with Tasmota 14.0.0 (GPIO26 FAN connected to Relais1 as idx1, PIR connected to GPIO 27 as Switch idx2) to connected to Domoticz. A dzventz starts the fan for an amount of time after an signal of the PIR. 
Works fine but after sevral hours upto days the relais isn't activated anymore despite there is a domoticz/out command to switch on or off.  In this state:
- Tasmota does sent the PIR switched message to domoticz,
- Domoticz does set the relais in Domoticz to ON
- Musquitto does sent a MQTT message for the Relais
- Tasmota does not act on this message and relais isn't switched to ON
- after restart Tasmota, Relais switches as it should for hours upto days.
- MQTT and weblogging set to debug shows:
- MQTT subscribed to 
-       cmnd/HLso37/#
-       cmnd/HLso37/tasmotas/#
-       cmnd/HLso37/DVES_48F960_fb/#
-  after Tasmota restart
-      the 3 above
- and to domoticz/out/#
- 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [V ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [V] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [V ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [V ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): SonOff MiniR4
- [V ] Tasmota binary firmware version number used: 14.0.0
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [V ] Flashing tools used: Tasmota Web installer
- [V ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
12:35:00.607 CMD: Backlog Template; Module; GPIO 255`: ```lua
12:35:00.608 SRC: WebConsole from 192.168.2.20
12:35:00.610 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 35, Pld -99, Data 'Template; Module; GPIO 255`: ```lua'
12:35:00.630 SRC: Backlog
12:35:00.631 CMD: Grp 0, Cmd 'TEMPLATE', Idx 1, Len 0, Pld -99, Data ''
12:35:00.639 MQT: stat/HLso37/RESULT = {"NAME":"Sonoff MINIR4","GPIO":[32,0,0,0,0,0,0,0,0,0,0,0,0,0,0,544,0,0,0,0,0,0,224,161,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
12:35:00.854 SRC: Backlog
12:35:00.856 CMD: Grp 0, Cmd 'MODULE', Idx 1, Len 0, Pld -99, Data ''
12:35:00.860 MQT: stat/HLso37/RESULT = {"Module":{"0":"Sonoff MINIR4"}}
12:35:01.104 SRC: Backlog
12:35:01.106 CMD: Grp 0, Cmd 'GPIO', Idx 1, Len 12, Pld 255, Data '255`: ```lua'
12:35:01.121 MQT: stat/HLso37/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"544":"LedLink"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"224":"Relay1"},"GPIO27":{"161":"Switch2"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [V ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
12:37:07.112 CMD: status 0
12:37:07.114 SRC: WebConsole from 192.168.2.20
12:37:07.115 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
12:37:07.120 MQT: stat/HLso37/STATUS = {"Status":{"Module":0,"DeviceName":"HLso37","FriendlyName":["Tasmota37"],"Topic":"HLso37","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"0001","SaveData":1,"SaveState":1,"SwitchTopic":"HLso37","SwitchMode":[15,1,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
12:37:07.129 MQT: stat/HLso37/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T17:08:39","StartupUTC":"2024-05-26T17:28:28","Sleep":50,"CfgHolder":4617,"BootCount":75,"BCResetTime":"2024-05-17T13:28:33","SaveCount":796}}
12:37:07.138 MQT: stat/HLso37/STATUS2 = {"StatusFWR":{"Version":"14.0.0(release-tasmota32)","BuildDateTime":"2024-05-14T13:54:09","Core":"3_0_0","SDK":"5.1.3.240430","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.1","CR":"423/699"}}
12:37:07.145 MQT: stat/HLso37/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"MqttLog":4,"SysLog":1,"LogHost":"","LogPort":514,"SSId":["HLinc","HLinc2"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80003000600003C5A0A192800000000","000000C0","00006000","00004000","00000000"]}}
12:37:07.165 MQT: stat/HLso37/STATUS4 = {"StatusMEM":{"ProgramSize":2016,"Free":863,"Heap":140,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80B1","00000004"],"Drivers":"1,2,!3,!4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,82,!86,!87,!88,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
12:37:07.179 MQT: stat/HLso37/STATUS5 = {"StatusNET":{"Hostname":"HLso37","IPAddress":"192.168.2.187","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.254","DNSServer2":"0.0.0.0","Mac":"A0:B7:65:48:F9:60","IP6Global":"","IP6Local":"fe80::a2b7:65ff:fe48:f960%st3","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.2.254","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
12:37:07.186 MQT: stat/HLso37/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.214","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_48F960","MqttUser":"DVES_USER","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":15}}
12:37:07.189 BRY: GC from 8525 to 4434 bytes, objects freed 9/45 (in 1 ms) - slots from 56/91 to 50/91
12:37:07.196 MQT: stat/HLso37/STATUS7 = {"StatusTIM":{"UTC":"2024-05-27T10:37:07Z","Local":"2024-05-27T12:37:07","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":99,"Sunrise":"05:54","Sunset":"21:39"}}
12:37:07.206 MQT: stat/HLso37/STATUS10 = {"StatusSNS":{"Time":"2024-05-27T12:37:07","Switch2":"OFF"}}
12:37:07.211 BRY: GC from 4703 to 4212 bytes, objects freed 2/45 (in 1 ms) - slots from 50/91 to 50/91
[... 40 lines cut ...]
	"stype" : "Switch",
	"svalue1" : "0",
	"switchType" : "On/Off",
	"unit" : 1
}
{
	"Battery" : 255,
	"LastUpdate" : "2024-05-27 12:05:02",
	"RSSI" : 12,
	"description" : "",
	"dtype" : "Light/Switch",
	"hwid" : "52",
	"id" : "000141A0",
	"idx" : 336,
	"name" : "HLso37 Relais",
	"nvalue" : 1,
	"org_hwid" : "52",
	"stype" : "Switch",
	"svalue1" : "0",
	"switchType" : "On/Off",
	"unit" : 1
}
But there is nothing visual in Tasmota Console and Relais isn't activated.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2024-05-27:
Already fixed in v14.0.0.1

Hans-Sh, 2024-05-27:
Txs for your quick response  I will update this case after upgrade and test period. Thanks for your superb work!

Hans-Sh, 2024-05-29:
After 2 days of testing the device keeps communicating with Domoticz via MQTT. But I do see several Restarts with reason Software reset CPU.  I prosume the MQTT subscription issue occured after an underlaying issue which is visual now. As there is no crash logging visual I have started to Log to my Synology Log server. 
Thanks for resolving the MQTT domoticz/out subscription issue. I will continue to investigate the frequent Software reset CPU issue