arendst/Tasmota #21555: Update EZSP firmware repository for (old) Sonoff Zigbee Bridge to support EZSP v13

lrswss, 2024-06-03:
### PROBLEM DESCRIPTION
The most recent EZSP firmware 7.4.x for the [Sonoff Zigbee Bridge (2020)](https://zigbee.blakadder.com/Sonoff_ZBBridge.html) is missing from [tools/fw_SonoffZigbeeBridge_ezsp](https://github.com/arendst/Tasmota/tree/development/tools). The new firmware supporting EZSP v13 is required to keep the old [Sonoff Zigbee Bridge working with Zigbee2Mqtt](https://www.zigbee2mqtt.io/advanced/remote-adapter/connect_to_a_remote_sonoff_zbbridge.html), which has recently [introduced the new ember adapter](https://github.com/Koenkk/zigbee2mqtt/discussions/21462) to communicate with Zigbee Dongles. The `ember` adapter is about to replace the old `ezsp` adapter.

### REQUESTED INFORMATION

- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ) - [only refers to old v6.7.x releases](https://tasmota.github.io/docs/Zigbee/) with outdated EZSP v8 support
- [x] Device used: Sonoff Zigbee Bridge (2020) configure as "Sonoff ZHABridge (0)" with Ziggbee2Mqtt 1.3.8
- [x] Tasmota precompiled binary firmware version number used: 13.4.0 with [ncp-uart-sw_7.4.3.0_115200.ota](https://github.com/xsp1989/zigbeeFirmware/blob/master/firmware/ZigbeeBridge_SM-011-signed/ncp-uart-sw_7.4.3.0_115200.ota)  firmware  

### ADDITIONAL CONTEXT
New EZSP firmware releases: https://github.com/xsp1989/zigbeeFirmware/tree/master/firmware/ZigbeeBridge_SM-011-signed

sfromis, 2024-06-03:
AFAIK, the newest versions are not working with Tasmota Z2T.

lrswss, 2024-06-03:
Well, my old Sonoff Zigbee Bridge with Tasmota 13.4.0 running in "ZHA Bridge (0)" mode and most recent EZSP firmware ncp-uart-sw_7.4.3.0_115200.ota is working perfectly for me with current Zigbee2MQTT v1.3.8. I had to upgrade from previous 6.7.x EZP firmware to 7.4.x to get it working again after upgrading from Z2M 1.3.4. Since the ezsp adapter is already marked as deprecated in Z2M other people might run into the same problem using the oudated 6.7.x EZSP firmware with Tasmota.

closed 2024-06-03 by lrswss

PierpaoloTRV, 2024-07-05:
I have followed your advice with either Tasmota 14.1.0 and also the same one you used 13.4.0 but when I tried to upload ncp-uart-sw_7.4.3.0_115200.ota I always had: "Upload Failed - Invalid file signature". 
According to the console file size of the ota file was correct:
File ncp-uart-sw_7.4.3.0_115200.ota (247752 bytes)
And I also double checked the module 75 command.
Any help or advice you can give me?

sfromis, 2024-07-05:
Just to be sure about what exactly you're doing ... since you normally use ZHA, you need to revert the full template, instead of just having `Module 75` enabled.
https://templates.blakadder.com/sonoff_ZBBridge.html

And of course also be using the appropriate Tasmota binary including Zigbee support for your device.

PierpaoloTRV, 2024-07-05:
Thanks for the prompt reply.
I was actually trying to update to the ember adapter to continue using my Sonoff Zigbee Bridge with Zigbee2MQTT in HA. Last time I followed the following guide up to step #6:
https://digiblur.com/2020/07/25/how-to-use-the-sonoff-zigbee-bridge-with-home-assistant-tasmota/

I have been finally able to upload the new "ncp-uart-sw_7.4.3.0_115200.ota" file, but I had to revert to the original Tasmota 10.1.0 version I had installed.
Now everything works and I have also updated to the latest 1.39.0-1 Zigbee2MQTT version.
Thanks anyway

joostlitjens, 2024-08-24:
> Well, my old Sonoff Zigbee Bridge with Tasmota 13.4.0 running in "ZHA Bridge (0)" mode and most recent EZSP firmware ncp-uart-sw_7.4.3.0_115200.ota is working perfectly for me with current [...]

Thanks for your useful comment! How stable is it working for you? Im on ZHA but thinking of upgrading