arendst/Tasmota #21562: 14.1.0 scripting: buttons bu(...) defined in >W appears corrupted on the web-page

nikvostro, 2024-06-04:
### PROBLEM DESCRIPTION
Buttons defined in the >W section for script now appears in corrupted way. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): Lilygo TTGO T8 and ESP32-C3
- [ ] Tasmota binary firmware version number used: 14.1.0
~~- [ ] Pre-compiled~~
  - [ ] Self-compiled
- [ ] Flashing tools used: VSC
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
12:15:55.954 CMD: Backlog Template; Module; GPIO 255
12:15:55.993 RSL: RESULT = {"NAME":"ESP32-S2-ttgo-display","GPIO":[33,0,0,0,0,0,5408,0,0,0,6720,704,736,672,0,1792,1824,225,224,0,0,0,3872,0,0,0,0,0,0,7808,7776,7840,0,0,0,0],"FLAG":0,"BASE":1}
12:15:56.205 RSL: RESULT = {"Module":{"0":"ESP32-S2-ttgo-display"}}
12:15:56.415 RSL: RESULT = {"GPIO0":{"33":"Button2"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"5408":"Button_tc1"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"6720":"SDCard CS"},"GPIO11":{"704":"SPI MOSI1"},"GPIO12":{"736":"SPI CLK1"},"GPIO13":{"672":"SPI MISO1"},"GPIO14":{"0":"None"},"GPIO15":{"1792":"SerBr Tx"},"GPIO16":{"1824":"SerBr Rx"},"GPIO17":{"225":"Relay2"},"GPIO18":{"224":"Relay1"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO33":{"3872":"Output Lo"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"},"GPIO40":{"7808":"I2S BCLK1"},"GPIO41":{"7776":"I2S DOUT1"},"GPIO42":{"7840":"I2S WS1"},"GPIO43":{"0":"None"},"GPIO44":{"0":"None"},"GPIO45":{"0":"None"},"GPIO46":{"0":"None"}}
```

- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`: - no rules but script
- [ ] Provide the output of this command: `Status 0`:
```lua
12:16:33.450 CMD: Status 0
12:16:33.454 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"VC1","FriendlyName":["VC1","",""],"Topic":"tasmota_013CEC","ButtonTopic":"0","Power":"010","PowerLock":"000","PowerOnState":0,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":0,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
12:16:33.502 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"","RestartReason":"Software reset digital core","Uptime":"0T00:14:14","StartupUTC":"2024-06-04T09:02:19","Sleep":50,"CfgHolder":4616,"BootCount":106,"BCResetTime":"2023-10-27T21:26:27","SaveCount":336}}
12:16:33.541 RSL: STATUS2 = {"StatusFWR":{"Version":"14.1.0(vc_lilygo_ttgo_t8_s2_nav)","BuildDateTime":"2024-06-04T11:06:08","Core":"3_0_0","SDK":"5.1.4.240529","CpuFrequency":240,"Hardware":"ESP32-S2 v0.0","CR":"311/699"}}
12:16:33.580 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["domuk",""],"TelePeriod":0,"Resolution":"558180C0","SetOption":["00008002","1405C80005000600003C5A00192800000000","00800080","00006000","00044000","00000000"]}}
12:16:33.627 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1411,"Free":1468,"Heap":119,"StackLowMark":3,"PsrMax":8192,"PsrFree":8112,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164068","FlashFrequency":80,"FlashMode":"DIO","Features":["0809","07A0058A","00080021","00000003","00000000","40000001","008080A0","00200000","54000008","00000080","00000004"],"Drivers":"1,!2,!4,8,9,10,!12,13,!24,38,40,!42,50,52,62,!68,!121","Sensors":"1,2,127"}}
12:16:33.673 RSL: STATUS5 = {"StatusNET":{"Hostname":"VC1","IPAddress":"192.168.xx.yy","Gateway":"192.168.11.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.11.1","DNSServer2":"0.0.0.0","Mac":"7C:DF:A1:xx:yy:zz","IP6Global":"","IP6Local":"fe80::7edf:a1ff:xxxx:yyyy%zzzz","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":15.0}}
12:16:33.721 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-06-04T09:16:33Z","Local":"2024-06-04T12:16:33","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+03:00","Sunrise":"03:49","Sunset":"21:05"}}
12:16:33.762 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-06-04T12:16:33"}}
12:16:33.807 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-06-04T12:16:33","Uptime":"0T00:14:14","UptimeSec":854,"Heap":119,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":0,"Berry":{"HeapUsed":5,"Objects":75},"POWER1":"OFF","POWER2":"ON","POWER3":"OFF","Wifi":{"AP":1,"SSId":"domuk","BSSId":"7C:10:C9:xx:yy:zz","Channel":11,"Mode":"HT20","RSSI":100,"Signal":-50,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log: - n/a

### TO REPRODUCE
Define button bu(....) in >W section

My >W section for ESP32-S2 device:
```lua
>W
srecLast=%srecLast%
rcgNoLast=%0rcgNoLast%
Response=%verbResp%
nm(0 79 1 inpRcgNo "Record # (0-79)" 200 0)
bu(btnRecord "Train record")
bu(btnInitVR "Init VR")
bu(btnClrRcg "Clear VR")
<a href="http://192.168.xx.zzz/ufse?file=/autoexec.be" target="_blank">autoexec.be</a>
<a href="http://192.168.xx.zzz/s10?" target="_blank">Script</a>

<button onclick="document.location='http://192.168.xx.yyy/cm?cmnd=brrestart'" formtarget="_blank">Restart Berry VM</button>
```

and for ESP32-C3 device:
```lua
>W
@<font size="5">Temp. set: %0tempinp%</font> (temp=%temp%, 0-off)
@sl(18 30 tempinp "18" "24" "30")
@<font size="5">Fan speed: %0faninp%</font>
@sl(0 4 faninp "0-Auto" "2-Mid" "4-Max")
;nm(18 30 1 tempinp "Temperature (18..30)")
;nm(0 4 1 faninp "Fan (0-auto, 4-max)")
bu(acsetcoolbtn "AC cool")bu(acsetheatbtn "AC heat")bu(acsetfanbtn "AC fan")
bu(acoffbtn "AC off")
Uptime: %0uptimeD%d %2.0uptimeH%:%2.0uptimeM%
```

### EXPECTED BEHAVIOUR
Buttons to appear normally

### SCREENSHOTS
Bad appearance:
[image]
[image]
[image]

The buttons view that was on 13.4.0 and all lower versions:
[image]

### ADDITIONAL CONTEXT
I've did upgrade from v13 to 14 on my 2 devices S2 and C3 and faced with the sinilar bugs on both.

gemu2015, 2024-06-04:
very strange that that ever has worked with that syntax.
you must provide 2 string arguments with button elements. (on state text, off state text, see docs).

nikvostro, 2024-06-04:
Hmmmm... you're definetely right! and now it works...
Thank you!

closed 2024-06-04 by Jason2866