arendst/Tasmota #21613: Matter Custom/Unknown Cluster Cluster exposed on Endpoint 2 ClusterId 5 (0x005)

lboue, 2024-06-11:
### PROBLEM DESCRIPTION
Matter Custom/Unknown Cluster Cluster exposed on Endpoint 2 ClusterId 5 (0x005)

[image]

[image]

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): M5 [ATOM Lite](https://docs.m5stack.com/en/core/atom_lite)
- [ ] Tasmota binary firmware version number used: Tasmota 14.1.0.2 (87148ae-tasmota32) 
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: [_____](https://tasmota.github.io/install/)
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

lboue, 2024-06-11:
@s-hadinger told me 0x0005 is the old Zigbee Scene cluster.

This seems to have been replaced by cluster 0x0062. This can be found in the ‘Matter-1.3-Core-Specification.pdf’: ‘ 1.4. Scenes Management Cluster'.

s-hadinger, 2024-06-12:
This was indeed changed between Matter 1.2 and Matter 1.3:
"Based on the ZCL Scenes Cluster, Updated the Cluster ID to 0x0062, the name to Scenes Man­agement[...]"

It's surprising that cluster 0x0005 is marked as unknown, it is still valid up to Matter 1.2. Using cluster 0x0062 requires the device revision number to be updated, which is not the case in Tasmota.

Jason2866, 2024-06-12:
Looks like a bug in Home Assistant to me.

lboue, 2024-06-12:
I will open an issue for this bug. Thanks.

closed 2024-06-12 by Jason2866