arendst/Tasmota #21621: Range extender compilation errors with Platformio

fdaubercy, 2024-06-13:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): ESP32
- [x ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ x] Flashing tools used: Platformio
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
Hi all.
I ask for your help.

I can't compile properly with platformio on an ESP32 with a personal firmware integrating Range extender.
Here are the parameters placed in the platformio-tasmota-cenv.ini file:

```
[env:tasmota32-serveur-debitmetres-pac]
upload_port             = COM5
monitor_port            = COM5
extends                 = env:tasmota32_base
build_flags             = ${env:tasmota32_base.build_flags}
                            -D FIRMWARE_TASMOTA32
                            -D MY_LANGUAGE=fr_FR 
                            -D FIRMWARE_SERVEUR_DEBITMETRES_PAC
                            -D OTA_URL='"http://ota.tasmota.com/tasmota32/release/tasmota32-FR.bin"'

                            -D USE_WIFI_RANGE_EXTENDER
                            -D USE_WIFI_RANGE_EXTENDER_NAPT
```

This configuration only works up to version 13.4.0.

I'm trying to compile with versions v14.0.0 and v14.1.0 but I'm getting the following compilation errors (see in the section 'ADDITIONAL CONTEXT' below) :

The following configuration provided in the tasmota doc also does not work :
```
[env:tasmota32-rangeextender]
extends                 = env:tasmota32idf4
[... 218 lines cut ...]
  407 |   err = tcpip_adapter_set_dns_info(TCPIP_ADAPTER_IF_AP, ESP_NETIF_DNS_MAIN, &ip_dns);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/fdaub/Documents/Github/Tasmota/tasmota/tasmota_xdrv_driver/xdrv_58_range_extender.ino:409:3: error: 'tcpip_adapter_dhcps_option' was not declared in this 
scope
  409 |   tcpip_adapter_dhcps_option(ESP_NETIF_OP_SET, ESP_NETIF_DOMAIN_NAME_SERVER, &opt_val, 1);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/fdaub/Documents/Github/Tasmota/tasmota/tasmota_xdrv_driver/xdrv_58_range_extender.ino:410:9: error: 'tcpip_adapter_dhcps_start' was not declared in this scope; did you mean 'tcpip_adapter_if_t'?
  410 |   err = tcpip_adapter_dhcps_start(TCPIP_ADAPTER_IF_AP);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~
      |         tcpip_adapter_if_t
Compiling .pio\build\tasmota32-serveur-debitmetres-pac\lib64d\Adafruit_SGP30-1.2.0\Adafruit_SGP30.cpp.o
Archiving .pio\build\tasmota32-serveur-debitmetres-pac\lib241\libAdafruit_PM25AQI-1.0.6.a
*** [.pio\build\tasmota32-serveur-debitmetres-pac\src\tasmota.ino.cpp.o] Error 1
Compiling .pio\build\tasmota32-serveur-debitmetres-pac\libc97\Adafruit_SGP40-1.1.0\Adafruit_SGP40.cpp.o
================================================================== [FAILED] Took 312.19 seconds ==================================================================

Environment                        Status    Duration
---------------------------------  --------  ------------
tasmota32-serveur-debitmetres-pac  FAILED    00:05:12.186
============================================================== 1 failed, 0 succeeded in 00:05:12.186 ==============================================================
 *  Arrêt du processus de terminal "C:\Users\fdaub\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'tasmota32-serveur-debitmetres-pac'". Code de sortie : 1. 
 *  Le terminal sera réutilisé par les tâches, appuyez sur une touche pour le fermer. 
```

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2024-06-13:
closing since duplicate of #21200

closed 2024-06-13 by Jason2866

sfromis, 2024-06-13:
The release notes also warns about the range extender functionality not being restored to work with Arduino Core3.
https://github.com/arendst/Tasmota/blob/development/RELEASENOTES.md#known-issues-with-v1410