arendst/Tasmota #21642: Script example for simple switch missing in doc

muchcodesuchwow, 2024-06-15:
Hello,
i am trying to add a switch to a esp8266 which has no rules but scripts enabled. (i also want to use SML on this esp, which requiers scripts).

The documentation is missing a simple example of a switch and i am struggeling hard to get this working. There is no real intro into the scripts. Please add a example script to read a switch, which also shows up in Home Assistant and give an easy introduction, how to get scripts working. I.e. i just discovered, that i need to write "Weblog 2" into the console, to see print output of the script after long struggle to understand the scripts.

My not working switch script is:

`>D
>B
>S
switchmode 1
switchtopic 0

>E
print event happened

sw=sw[1]
if chg[sw]>0
then
  if sw==1
  then
    publish stat/%topic%/PIR1 ON
    print motion
  else
    publish stat/%topic%/PIR1 OFF
    print no motion
  endif
endif`

Thank you very much

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

pkkrusty, 2024-06-17:
Try this:

```
>D
sw=0
>B
switchmode 1
switchtopic 0
>S

>E

sw=sw[0]
if chg[sw]>0 {
    =>publish stat/%topic%/PIR1 %sw%
}
```

need to declare sw in the >D section. Switch1 as configured in GUI is sw[0], not sw[1]. Need to preface console commands with =>.

muchcodesuchwow, 2024-07-15:
Please excuse my late response.
Thank you very much, it works now.
Can we add this to the tasmota documentation, so others can find this?

Based on https://blakadder.com/pir-in-tasmota-integration/ i added the discovery into home assistant.
If anyone is curious, this is my final script:
`>D
sw=0
>B
=>sensor53 r
=>sensor53 l2
=>switchmode 1
=>switchtopic 0
=>SetOption19 0
=>SetOption114 1

=>publish2 homeassistant/binary_sensor/%macaddr%_motion/config {"name":"Bewegungsmelder Kabel Parkplatz","state_topic":"stat/%topic%/PIR1","device_class":"motion","force_update":true,"availability_topic":"tele/%topic%/LWT","payload_available":"Online","payload_not_available":"Offline","unique_id":"%macaddr%_motion","device":{"connections":[["mac","%macaddr%"]]}}

>M 1
+1,3,s,0,9600,
1,77070100010800ff@1000,Verbrauch gesamt,KWh,Gesamt,20
1,77070100100700ff@1,Verbrauch aktuell,W,Leistung,18
#

>E
sw=sw[0]
if chg[sw]>0 {
    =>publish stat/%topic%/PIR1 %sw%
}`

pkkrusty, 2024-07-15:
Maybe we can move this to discussions so it's more easily searchable, since it's more of a support request vs issue.

closed 2024-07-15 by muchcodesuchwow