arendst/Tasmota #21651: Is 512K H801 modules still supported in some way ?

FulvioSpelta, 2024-06-17:
Starting from issue #2982 I kindly ask if there is a way to tasmotize them.
I've searched 14.x sources but i can't find any 512k linker directive.

No problem to compile myself a specific ultra minimal version.

Thank a lot.

barbudor, 2024-06-17:
If you mean ESP8266 with 512k flash, then answer is not supported
Minimum is 1MB

You may be able to transplant a better module or flash chip depending on the board layout

sfromis, 2024-06-17:
Assuming board layout similar to current [H801](https://tasmota.github.io/docs/devices/H801/#serial-connection), the easiest upgrade would be to replace the flash chip, just above the ESP8266 in the board image.

closed 2024-06-17 by Jason2866