arendst/Tasmota #21654: Crash with tcp server in `>B` or `>BS` section

Thalhammer, 2024-06-18:
### PROBLEM DESCRIPTION
ESP crashes when `wso(x)` is executed in a script `>B` or `>BS` section. Delaying the initialization using a timer fixes the issue, but is obviously less then ideal. Enabling the script after boot also works, but is not ideal as well.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): bitShake SmartMeterReader-Air (ESP32-C3)
- [X] Tasmota binary firmware version number used: 2115e09839b218779a152577e3a7aa28113bdb7a
  - [ ] Pre-compiled
  - [X] Self-compiled
- [X] Flashing tools used: Web Update followed by esp-tool for troubleshooting
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
09:22:34.321 CMD: Backlog Template; Module; GPIO 255
09:22:34.341 RSL: RESULT = {"NAME":"ESP32C3","GPIO":[1,1,1,1,1,1,1,1,1,1,1,0,0,0,0,0,0,0,1,1,1,1],"FLAG":0,"BASE":1}
09:22:34.569 RSL: RESULT = {"Module":{"1":"ESP32C3"}}
09:22:34.776 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
Not using Rules
```
- [X] Provide the output of this command: `Status 0`:
```lua
09:23:53.737 CMD: Status 0
09:23:53.740 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
09:23:53.746 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32c3.bin","RestartReason":"Software reset digital core","Uptime":"0T00:07:53","StartupUTC":"2024-06-18T07:16:00","Sleep":50,"CfgHolder":4619,"BootCount":16,"BCResetTime":"2024-06-18T08:48:36","SaveCount":33}}
09:23:53.750 RSL: STATUS2 = {"StatusFWR":{"Version":"14.1.0.2(tasmota32)","BuildDateTime":"2024-06-18T09:13:07","Core":"3_0_1","SDK":"5.1.4.240602","CpuFrequency":160,"Hardware":"ESP32-C3 v0.4","CR":"343/699"}}
09:23:53.754 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["<redacted>",""],"TelePeriod":15,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
09:23:53.765 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":2079,"Free":800,"Heap":176,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":80,"FlashMode":"DIO","Features":["0809","979AD7DF","001DA001","B7F7BFCF","05DA9BC4","E0360DCF","480840D2","20200000","D4BC482D","810A80B1","00000015"],"Drivers":"1,2,!3,!4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,!71,!73,82,!86,!87,!88,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,53,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
09:23:53.772 RSL: STATUS5 = {"StatusNET":{"Hostname":"x2496","IPAddress":"<redacted>","Gateway":"<redacted>","Subnetmask":"255.255.255.0","DNSServer1":"<redacted>","DNSServer2":"0.0.0.0","Mac":"<redacted>","IP6Global":"","IP6Local":"fe80::424c:caff:fec8:c9c0%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"<redacted>","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":19.0}}
09:23:53.778 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"<redacted>","MqttPort":1883,"MqttClientMask":"","MqttClient":"","MqttUser":"","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
09:23:53.783 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-06-18T07:23:53Z","Local":"2024-06-18T09:23:53","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":99,"Sunrise":"<redacted>","Sunset":"<redacted>"}}
09:23:53.788 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-06-18T09:23:53"}}
09:23:53.793 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-06-18T09:23:53","Uptime":"0T00:07:53","UptimeSec":473,"Heap":174,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":4,"Objects":46},"Wifi":{"AP":1,"SSId":"<redacted>","BSSId":"<redacted>","Channel":6,"Mode":"HT20","RSSI":62,"Signal":-69,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [X] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048b82
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x48
load:0x403cc710,len:0x644
load:0x403ce710,len:0x2294
entry 0x403cc710

00:00:00.001 CMD: Using USB CDC
00:00:00.001 HDW: ESP32-C3 v0.4 
00:00:00.004 UFS: FlashFS mounted with 304 kB free
00:00:00.008 CFG: Loaded from File, Count 33
00:00:00.010 SER: Set to 8N1 115200 bit/s
00:00:00.010 SER: HWCDC supports 115200 bit/s only
00:00:00.014 QPC: Count 1
00:00:00.016 SCR: ufilesystem found
00:00:00.021 SCR: nv=1, tv=0, vns=4, vmem=33, smem=8192, gmem=424, tmem=8653 
00:00:00.047 BRY: Berry initialized, RAM used 4400 bytes
00:00:00.055 Project tasmota - Tasmota Version 14.1.0.2(tasmota32)-3_0_1(2024-06-18T09:13:07)
00:00:00.056 tcp server started
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x420e6cd0  RA      : 0x420d6d6e  SP      : 0x3fca2020  GP      : 0x3fc8c000  
[... 124 lines cut ...]
4. Observe crash in serial console
5. ESP reboots with script disabled

```lua
>D
res=0
>B
res=wso(8080)
print res=%res%
```

### EXPECTED BEHAVIOUR
ESP boots normally and opens a tcp port.

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
I tried various versions between 14.0 and develop as well as prebuilds I could find (sadly TCP_SERVER is not enabled by default), all of them crash the same way.

I also added extra log lines to the script interpreter during debugging and it seems to crash inside `glob_script_mem.tcp_server->begin();` [here](https://github.com/arendst/Tasmota/blob/2115e09839b218779a152577e3a7aa28113bdb7a/tasmota/tasmota_xdrv_driver/xdrv_10_scripter.ino#L6229).

While debugging I also noticed a secondary bug (however thats not my issue), that being if `new WiFiServer(port)` where to ever return null it would crash as well because it unconditionally references the pointer after setting the result code.

In addition theres a memory leak if a script calls wso more than once because it does not check/delete the WiFiServer instance before reassigning it.

gemu2015, 2024-06-18:
you may not open a tcp server or access any wifi related resources when wifi is not connected. 
on a  reboot this is never the case during >B
you can open the port when wifi is connected which you can check e.g during >S with 'wifis'

Thalhammer, 2024-06-18:
> you may not open a tcp server or access any wifi related resources when wifi is not connected.

Yeah thats what I assumed (though opening it with a timer of 1s still does before the wifi is connected and works fine), however I couldn't find anything related to that in the docs and all the example scripts on the internet I found (apearently) have it working fine in `>B`.

However I think the other two issues I noted in additional context are still valid.

PR #21660 (2024-06-21): fix tcp server
## Description:

**Related issue (if applicable):** fixes #21654
also fixes google chart timeline crash

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x] The code change is tested and works with Tasmota core ESP8266 V.2.7.6
  - [x] The code change is tested and works with Tasmota core ESP32 V.3.0.1
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-l [... cut]
closed 2024-06-21 by arendst