arendst/Tasmota #21658: Berry Scripting Language on Tasmota32 Driver: json.load(tasmota.read_sensors()) returns nil

MechatronikMonkey, 2024-06-19:
### PROBLEM DESCRIPTION
`json.load(tasmota.read_sensors())` returns `nil` after `tasmota.response_append(msg)` in a driver

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): generic
- [ ] Tasmota binary firmware version number used: [Tasmota 14.1.0 (release-tasmota32)]
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: VS-Code
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

Hello World.
nil
Hello World.
nil
Hello World.
nil
Hello World.
nil

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Copy this to Berry Console:
`
class MyTestClass : Driver

    def DoIt()
        print("Hello World.")

        # Read Sensor data
        import json
        var MySensors = json.load(tasmota.read_sensors())
        print(MySensors)

    end

    def every_second()
        if !self.DoIt return nil end
        self.DoIt()

    end

    def web_sensor()
        import string
        var msg = string.format(
                  "Show this in Web UI")
        tasmota.web_send(msg)
    end

    def json_append() 
        import string
        var msg = string.format(",\"TestValue\":123}")
        tasmota.response_append(msg)
    end

end

test_instance = MyTestClass()
tasmota.add_driver(test_instance)
`

### EXPECTED BEHAVIOUR

if you remove the response_append statement, the json.load can load the sensors as expected:
`

    def json_append() 
        import string
        var msg = string.format(",\"TestValue\":123}")
        #tasmota.response_append(msg)
    end

`

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
It looks like that the response_append(msg) method kind of crashes json. If you try to parse a json string after response_append() you see the nil pointer. 

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2024-06-19:
You have an unexpected closing bracket that break the JSON syntax
Correct would be
```
var msg = string.format(",\"TestValue\":123")
```

If you were just printing the result from `tasmota.read_sensors()` without trying to load it, you would have seen that your JSON was broken

There is no problem wiht `tasmota.response_append()`

You should better open a discussion instead of an issue when you don't have a qualified bug

MechatronikMonkey, 2024-06-19:
Ok that is the problem... dohh ... 

I printed the json string over and over again, but didn´t see the incorrect syntax... 

Lol, what do you mean I don't have a qualified bug... I just thought it was a bug... I didn't know I needed to qualify somewhere first... Sorry, I am unworthy. Forgive me, honorable Tasmota gods, for I have sinned.

barbudor, 2024-06-19:
The best way to qualify if it's a bug worth an issue is to first start a discussion
If someone say "right, looks like you found a bug" then you're good to go for the Issue 🤣 

Enjoy Berry

sfromis, 2024-06-19:
Beware that while the JSON syntax indeed gets better by keeping those `{}` balanced, what you produce is not how a Berry driver is meant to produce sensor values. The appropriate format would be including a name for the sensor object in the JSON, to house individual values, like:
`var msg = ',"MyTest":{"TestValue":123}'`
As you had no string formatting, no use for `string.format`.

But if you had, like a var named `myvalue`, you could use a f-string like:
`var msg = f',"MyTest":{{"TestValue":{myvalue}}}'`
A downside of this easy syntax is that you need to double "your own" `{}` in the string.

(I also took the opportunity to improve readability by having `'` as string delimiter around the whole string, meaning that then it is no longer needed to escape as `\"`).

BTW... I'm not saying **can't** work without, more like recommending to follow the usual JSON formatting, to be compatible with other tools handling the JSON payloads produced.

MechatronikMonkey, 2024-06-20:
> The best way to qualify if it's a bug worth an issue is to first start a discussion If someone say "right, looks like you found a bug" then you're good to go for the Issue 🤣
> 
> Enjoy Berry

Got it... I open a discussion first in the future...