arendst/Tasmota #21680: Tasmota32 not bringing up Wifi on ESP32-S3-zero board

dolocyl2, 2024-06-22:
Hi all

I've been using Tasmota mostly on ESP8266 and on some ESP32-WROOM without issues.
Now I have two new boards of Type: https://www.waveshare.com/wiki/ESP32-S3-Zero 
I've flashed both ESP-S3 boards using Webinstaller as well as ESP-Flasher with the same result.
There is no wifi coming up. 
On different hardware (ESP32-WROOM) I have no issue (using a non -S3 binary).

### PROBLEM DESCRIPTION
Tasmota is booting on the ESP32-S3-zero and I can interact in the console, but there is no wifi. 
-->No initial AP. 
-->Also no connection to my WLAN if I set the SSID, PWD using the console (see logs below):
Backlog SSID1 OpenWrt; Password1 <myPwd>;

### REQUESTED INFORMATION
Device: ESP32-S3-zero
https://www.waveshare.com/wiki/ESP32-S3-Zero
Tasmota
- [x ] Tasmota binary firmware version number used: 14.1.0.2 on ESP32; tasmota32s3.factory.bin
  - [x] Pre-compiled
  - [] Self-compiled
- [x] Flashing tools used: Webinstaller, ESP-Flasher.exe
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
00:09:57.792 RSL: RESULT = {"NAME":"ESP32S3","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,0,0,0,0,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":1}
00:09:58.041 RSL: RESULT = {"Module":{"1":"ESP32S3"}}
00:09:58.294 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"},"GPIO40":{"0":"None"},"GPIO41":{"0":"None"},"GPIO42":{"0":"None"},"GPIO43":{"0":"None"},"GPIO44":{"0":"None"},"GPIO45":{"0":"None"},"GPIO46":{"0":"None"},"GPIO47":{"0":"None"},"GPIO48":{"0":"None"}}
00:09:58.741 APP: Serial logging disabled

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
00:10:53.647 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_532C80","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
00:10:53.650 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32s3.bin","RestartReason":"Usb uart reset digital core","Uptime":"0T00:10:54","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"1970-01-01T00:00:00","SaveCount":10}}
00:10:53.653 RSL: STATUS2 = {"StatusFWR":{"Version":"14.1.0(release-tasmota32)","BuildDateTime":"2024-06-03T11:56:29","Core":"3_0_0","SDK":"5.1.4.240529","CpuFrequency":240,"Hardware":"ESP32-S3 v0.2","CR":"389/699"}}
00:10:53.654 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
00:10:53.661 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1974,"Free":905,"Heap":250,"StackLowMark":4,"PsrMax":2048,"PsrFree":2022,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":80,"FlashMode":"QIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80B1","00000014"],"Drivers":"1,2,!3,!4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,!73,82,!86,!87,!88,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
00:10:53.666 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-532C80-3200","IPAddress":"0.0.0.0","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"34:B7:DA:53:2C:80","IP6Global":"","IP6Local":"","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"0.0.0.0","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":19.0}}
00:10:53.669 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_532C80","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
00:10:53.671 RSL: STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T00:10:53Z","Local":"1970-01-01T00:10:53","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":"+00:00","Sunrise":"20:13","Sunset":"05:47"}}
00:10:53.674 RSL: STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:10:53"}}
00:10:53.676 RSL: STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:10:53","Uptime":"0T00:10:54","UptimeSec":654,"Heap":250,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":4,"Objects":42}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
1) Flashing tasmota32s3.factory.bin using Webinstaller
2) Power-Cycle after webinstaller claims success
3a) Waiting for AP to come up
3b) Configuring SSID and PW in console and waiting for ESP32 to connect

### EXPECTED BEHAVIOUR
for 3a) Initial AP does come up
for 3b) ESP32 connects to WLAN

### SCREENSHOTS
Console log after flashing (3a):
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037c95a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3000,len:0x970
load:0x403c9700,len:0xa8c
load:0x403cc700,len:0x2808
entry 0x403c9868

00:00:00.000 CMD: Using USB CDC
00:00:00.000 HDW: ESP32-S3 v0.2 (PSRAM)
00:00:00.010 UFS: FlashFS mounted with 308 kB free
00:00:00.021 CFG: Loaded from File, Count 5
00:00:00.022 SER: Set to 8N1 115200 bit/s
00:00:00.023 SER: HWCDC supports 115200 bit/s only
00:00:00.029 QPC: Count 1
00:00:00.048 BRY: Berry initialized, RAM used 3904 bytes
00:00:00.055 Project tasmota - Tasmota Version 14.1.0(release-tasmota32)-3_0_0(2024-06-03T11:56:29)
00:00:00.358 WIF: WifiManager active for 3 minutes
00:00:00.003 HTP: Web server active on tasmota-532C80-3200 with IP address 192.168.4.1
00:00:06.758 QPC: Reset

### ADDITIONAL CONTEXT
console Log after manually configuring SSID, PW in console:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40374f2f
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x158
load:0x403c9700,len:0x4
load:0x403c9704,len:0xa88
load:0x403cc700,len:0x2834
entry 0x403c9868

00:00:00.000 CMD: Using USB CDC
00:00:00.001 HDW: ESP32-S3 v0.2 (PSRAM)
00:00:00.019 UFS: FlashFS mounted with 308 kB free
00:00:00.033 CFG: Loaded from File, Count 6
00:00:00.034 SER: Set to 8N1 115200 bit/s
00:00:00.034 SER: HWCDC supports 115200 bit/s only
00:00:00.040 QPC: Count 1
00:00:00.064 BRY: Berry initialized, RAM used 3904 bytes
00:00:00.072 Project tasmota - Tasmota Version 14.1.0(release-tasmota32)-3_0_0(2024-06-03T11:56:29)
00:00:01.001 WIF: Connecting to AP1 OpenWrt in mode HT40 as tasmota-5312F4-4852...
00:00:06.715 QPC: Reset
00:00:13.630 WIF: Connect failed with AP timeout
00:00:14.740 WIF: Connecting to AP1 OpenWrt in mode HT40 as tasmota-5312F4-4852...
00:00:27.066 WIF: Connect failed with AP timeout
00:00:29.000 WIF: Connecting to AP1 OpenWrt in mode HT40 as tasmota-5312F4-4852...
00:00:41.509 WIF: Connect failed with AP timeout
00:00:42.587 WIF: Connecting to AP1 OpenWrt in mode HT40 as tasmota-5312F4-4852...
00:00:54.896 WIF: Connect failed with AP timeout
00:00:56.974 WIF: Connecting to AP1 OpenWrt in mode HT40 as tasmota-5312F4-4852...

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2024-06-22:
Any luck after executing command ``wifi 4``?

dolocyl2, 2024-06-22:
unfortunately not.  I configured ssid1 and executed wifi 4 and still not getting any wifi connection:
`00:00:41.916 CMD: wifi 4
00:00:41.917 RSL: RESULT = {"Wifi":"ON","Mode":"HT40"}
00:00:47.067 WIF: Connect failed with AP timeout
00:00:49.000 WIF: Connecting to AP1 OpenWrt in mode HT40 as tasmota-532C80-3200...
00:00:59.513 WIF: Connect failed with AP timeout`

dolocyl2, 2024-06-22:
since this is new hardware, on which I have never seen wifi working, I decided to try micropython.
Same result there: If I configure a simple access point, there is no wifi signal.
Therefore, I conclude that this is faulty hardware... 2 out of 2 :-( 
Closing this issue.
@arendst: thank you for your support!

closed 2024-06-22 by dolocyl2

msx80, 2024-08-26:
Hi there, just to let people know that the Wifi runs for me. I've installed "Tasmota32 LVGL (english)" (development) from the web installer and everything went smooth.