arendst/Tasmota #21695: MaxPower does not have effect on Channel 1 on SONOFF Dual R3

garageeks, 2024-06-26:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I have a Sonoff POW Dual R3 running Tasmota 13.4.0(tasmota32)]
I set up "MaxPower 300" command to switch off power when the more than 300 Watts are pulled.
This works on channel 0, but not on channel 1.
I tried "MaxPower0 300" like "Power0" style command that is applied on all channels, but it doesn't appear to have an effect.
The documentation doesn't specify how to set power limit to a specific channel.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Sonoff POW Dual R3
- [X] Tasmota binary firmware version number used: 13.4.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

11:55:17.365 CMD: MaxPower
11:55:17.369 RSL: RESULT = {"MaxPower":300}
11:55:21.013 CMD: SetAllPower
11:55:21.022 RSL: RESULT = {"Command":"Unknown"}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Set power limit command with MaxPower x where x is a threshold for number of watts

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Both channel 0 and channel 1 would switch off

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

A similar issue on a Shelly device was reported here and marked as fixed, but it is not clear which commands are used.
https://github.com/arendst/Tasmota/issues/6340

A similar issue for the same device is in the discussions, but got not reply
https://github.com/arendst/Tasmota/discussions/18789

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2024-06-26:
From the code, this feature do not support channels
The command is 'MaxPower' without index and it applies only to the 1st channel

Behavior may be modified by 'MaxPowerHold' and 'MaxPowerWindow'

There is no command 'SetAllPower' this your error

garageeks, 2024-06-26:
@barbudor thank you for the quick reply.
Indeed the fix I mentioned https://github.com/arendst/Tasmota/commit/d44104135e7194120715c3267236dc93e316bb8a is switch off both channel when Channel 0 threshold is met.
I don't necessarily need to set up different power limits for each channel, it would be great that MaxPower would be effective on both channels.

arendst, 2024-06-26:
I'll have a look if implementing multi channel is an option.

garageeks, 2024-06-26:
@arendst thank you, really appreciated. I'm always available for testing.

arendst, 2024-06-27:
Give dev v14.1.0.3 a try.

Individual phases can now be configured and controlled with ``MaxPower2 300`` for phase/channel 2 and ``MaxPower1 200`` for phase/channel 1.

For backward compatibility ``MaxPower1 500`` with any other ``MaxPower<x> 0`` will result in all phase/channel power control.

garageeks, 2024-06-28:
@arendst Thank you for the quick update! I will have access to my test setup next week, will try it and provide feedback.

garageeks, 2024-07-02:
From my initial testing of 14.1.0.3 latest dev version (tasmota.bin	648k	20240702 12:17) the new commands appear to be broken.
I have set up `MaxPower2 20` to Channel 2 and applied a 25W load to Channel 2, it stays on all the time.
Thinking it might be too close to the threshold, I tried `MaxPower2 60` and appliead a 110W load, it stays on all the time.
However if I apply such a load to Channel 1, it works as intended and switches off both loads.
Interestingly, if I type `MaxPower2 300`, and then query `MaxPower1`, it is set up to 300 too
```
17:18:23.841 CMD: MaxPower2 300
17:18:23.844 RSL: RESULT = {"MaxPower":300}
17:18:27.728 CMD: MaxPower1
17:18:27.731 RSL: RESULT = {"MaxPower":300}
```

Let me know if it is clear enough and if you need more tests.
Cheers,
Nick

arendst, 2024-07-02:
``MaxPower<x>`` is related to ``MaxPowerHold<x>`` (default 10 seconds), ``MaxPowerWindow<x>`` (default 30 seconds) and ``SetOption33`` (default 5 retries).

So first chk these values with command ``Status 9`` and ``SO33``

```
17:39:48.411 CMD: status 9
17:39:48.412 SRC: WebConsole from 192.168.2.1
17:39:48.414 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 9, Data '9'
17:39:48.425 MQT: stat/dualr3c/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0],"PowerLow":[0,0],"PowerHigh":[0,0],"VoltageLow":[0,0],"VoltageHigh":[0,0],"CurrentLow":[0,0],"CurrentHigh":[0,0],"MaxPower":[0,20],"MaxPowerHold":[10,10],"MaxPowerWindow":[30,30],"MaxEnergy":[0,0],"MaxEnergyStart":[0,0]}}
17:40:08.421 CMD: so33
17:40:08.422 SRC: WebConsole from 192.168.2.1
17:40:08.424 CMD: Grp 0, Cmd 'SO', Idx 33, Len 0, Pld -99, Data ''
17:40:08.428 MQT: stat/dualr3c/RESULT = {"SetOption33":5}
```

I do see edge cases when MaxPowerHold = 0 and MaxPowerWindow = 0. I will fix those but for your test pls set these  parameters to 10 and 30 respectivily and restest.

garageeks, 2024-07-02:
Sure, here's the output:
```
18:02:21.226 CMD: Status 9
18:02:21.230 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0,0,0,0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:03:40.830 CMD: SO33
18:03:40.832 RSL: RESULT = {"SetOption33":5}
```

I have left MaxPowerHold and MaxPowerWindow values to their defaults.

For completeness, I'm using this template as configuration (took from https://templates.blakadder.com/sonoff_DUALR3_v2.html)
`{"NAME":"Sonoff Dual R3 v2","GPIO":[32,0,0,0,0,0,0,0,0,576,225,0,0,0,0,0,0,0,0,0,0,3200,8128,224,0,0,0,0,160,161,0,0,0,0,0,0],"FLAG":0,"BASE":1}`

arendst, 2024-07-02:
The output of command ``status 9`` hints that you're using an "old" version of the dev branch. I've fixed and changed the output a few days ago as can be seen from my output. 

Pls provide the output of command ``status 0``

garageeks, 2024-07-02:
Indeed in Information I see 
Program Version	14.1.0 (release-tasmota32)
Build Date & Time	2024-06-03T11:56:19
Core/SDK Version	3_0_0/5.1.4.240529

I have flashed the binary from this URL https://ota.tasmota.com/tasmota/tasmota.bin
Which appears to be the ESP8266 version, sorry for the mistake...
Interestingly it works on an ESP32 device...

I'll flash the correct FW from https://ota.tasmota.com/tasmota32/tasmota32.bin and will let you know.

Output of status 0 is
```
18:13:15.680 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"sonoff1","FriendlyName":["Tasmota",""],"Topic":"tasmota_89D408","ButtonTopic":"0","Power":"00","PowerLock":"00","PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
18:13:15.684 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Vbat power on reset","Uptime":"0T01:09:15","StartupUTC":"2024-07-02T15:04:00","Sleep":50,"CfgHolder":4617,"BootCount":8,"BCResetTime":"2024-03-15T12:45:54","SaveCount":77}}
18:13:15.686 RSL: STATUS2 = {"StatusFWR":{"Version":"14.1.0(release-tasmota32)","BuildDateTime":"2024-06-03T11:56:19","Core":"3_0_0","SDK":"5.1.4.240529","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.0","CR":"414/699"}}
18:13:15.690 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["stopcharge",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00208009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
18:13:15.703 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":2033,"Free":846,"Heap":141,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80B1","00000014"],"Drivers":"1,2,3,!4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,!73,82,!86,!87,!88,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82"}}
18:13:15.709 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-89D408-5128","IPAddress":"192.168.1.6","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"3C:E9:0E:89:D4:08","IP6Global":"","IP6Local":"fe80::3ee9:eff:fe89:d408%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
18:13:15.712 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_89D408","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:13:15.716 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-07-02T16:13:15Z","Local":"2024-07-02T18:13:15","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T02:00:00","Timezone":99,"Sunrise":"05:52","Sunset":"21:56"}}
18:13:15.719 RSL: STATUS9 = {"StatusPTH":{"PowerDelta":[0,0,0,0,0,0,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:13:15.727 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-07-02T18:13:15","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2024-03-15T00:00:00","Total":0.015,"Yesterday":0.001,"Today":0.013,"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Voltage":236,"Current":[0.000,0.000],"BL09XX":{"Temperature":35.4}},"TempUnit":"C"}}
18:13:15.733 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-07-02T18:13:15","Uptime":"0T01:09:15","UptimeSec":4155,"Heap":144,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":4,"Objects":42},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"mynetwork","BSSId":"CC:2D:E0:F9:E5:BD","Channel":1,"Mode":"HT20","RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```

garageeks, 2024-07-03:
Hi, apologies for the version mishap.
I correctly flashed `Tasmota 14.1.0.3 (0e6e4e8-tasmota32)` and tested, and work as intended.
When `MaxPower1` and `MaxPower2` are set to a value greater than zero, the power monitoring of the two channels works indipendently.
When `MaxPower1` > 0 and `MaxPower2` = 0, both channels are switched off when power monitoring of Channel 1 is triggered, as you described for backwards compatibility.

Here some minor nitpicking.
I noticed a `MaxPowerReachedRetry1` was triggered after only two `MaxPowerReached1` events, despite `SetOption33` being set to the default 5 retries.
Maybe there is a case where `Energy->mplr_counter[phase]` original value is not restored after previous attempts?

See the logs: channels where off, power monitoring has been turned off, then set up indipendent `MaxPower1` and `MaxPower2` values, turn both channels on, channel 1 is permanently off after two retries.

```
10:37:59.398 CMD: MaxPower1 0
10:37:59.402 RSL: RESULT = {"MaxPower1":0}
10:38:01.735 CMD: MaxPower2 0
10:38:01.738 RSL: RESULT = {"MaxPower2":0}
10:38:19.812 RSL: STATE = {"Time":"2024-07-03T10:38:19","Uptime":"0T16:20:10","UptimeSec":58810,"Heap":144,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":4,"Objects":46},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"stopcharge","BSSId":"CC:2D:E0:F9:E5:BD","Channel":1,"Mode":"HT20","RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:03"}}
10:38:19.822 RSL: SENSOR = {"Time":"2024-07-03T10:38:19","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2024-03-15T00:00:00","Total":0.024,"Yesterday":0.013,"Today":0.009,"Period":[1,1],"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Voltage":237,"Current":[0.000,0.000],"BL09XX":{"Temperature":40.4}},"TempUnit":"C"}
10:43:19.791 RSL: STATE = {"Time":"2024-07-03T10:43:19","Uptime":"0T16:25:10","UptimeSec":59110,"Heap":141,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":4,"Objects":46},"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"stopcharge","BSSId":"CC:2D:E0:F9:E5:BD","Channel":1,"Mode":"HT20","RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:03"}}
10:43:19.801 RSL: SENSOR = {"Time":"2024-07-03T10:43:19","Switch1":"ON","Switch2":"ON","ENERGY":{"TotalStartTime":"2024-03-15T00:00:00","Total":0.024,"Yesterday":0.013,"Today":0.009,"Period":[0,0],"Power":[0,0],"ApparentPower":[0,0],"ReactivePower":[0,0],"Factor":[0.00,0.00],"Voltage":238,"Current":[0.000,0.000],"BL09XX":{"Temperature":39.4}},"TempUnit":"C"}
10:43:26.139 CMD: MaxPower1 60
10:43:26.142 RSL: RESULT = {"MaxPower1":60}
10:43:29.521 CMD: MaxPower2 40
10:43:29.524 RSL: RESULT = {"MaxPower2":40}
10:43:34.146 RSL: RESULT = {"POWER1":"ON"}
10:43:34.148 RSL: POWER1 = ON
10:43:35.685 RSL: RESULT = {"POWER2":"ON"}
10:43:35.687 RSL: POWER2 = ON
10:43:39.820 RSL: WARNING = {"Time":"2024-07-03T10:43:39","MaxPowerReached1":116}
10:43:39.825 RSL: SENSOR = {"Time":"2024-07-03T10:43:39","ENERGY":{"TotalStartTime":"2024-03-15T00:00:00","Total":0.025,"Yesterday":0.013,"Today":0.009,"Power":[116,26],"ApparentPower":[116,26],"ReactivePower":[0,0],"Factor":[1.00,1.00],"Voltage":238,"Current":[0.489,0.108],"BL09XX":{"Temperature":39.6}}}
10:43:39.831 RSL: RESULT = {"POWER1":"OFF"}
10:43:39.833 RSL: POWER1 = OFF
10:44:42.825 RSL: RESULT = {"Time":"2024-07-03T10:44:42","PowerMonitor1":"ON"}
10:44:42.830 RSL: RESULT = {"POWER1":"ON"}
10:44:42.831 RSL: POWER1 = ON
10:44:47.829 RSL: WARNING = {"Time":"2024-07-03T10:44:47","MaxPowerReached1":113}
10:44:47.835 RSL: SENSOR = {"Time":"2024-07-03T10:44:47","ENERGY":{"TotalStartTime":"2024-03-15T00:00:00","Total":0.025,"Yesterday":0.013,"Today":0.010,"Power":[113,25],"ApparentPower":[113,25],"ReactivePower":[0,0],"Factor":[1.00,1.00],"Voltage":234,"Current":[0.484,0.107],"BL09XX":{"Temperature":39.4}}}
10:44:47.840 RSL: RESULT = {"POWER1":"OFF"}
10:44:47.841 RSL: POWER1 = OFF
10:45:50.793 RSL: WARNING = {"Time":"2024-07-03T10:45:50","MaxPowerReachedRetry1":"OFF"}
10:45:50.799 RSL: SENSOR = {"Time":"2024-07-03T10:45:50","ENERGY":{"TotalStartTime":"2024-03-15T00:00:00","Total":0.026,"Yesterday":0.013,"Today":0.011,"Power":[0,25],"ApparentPower":[0,25],"ReactivePower":[0,0],"Factor":[0.00,1.00],"Voltage":234,"Current":[0.000,0.107],"BL09XX":{"Temperature":40.0}}}
10:45:50.805 RSL: RESULT = {"POWER1":"OFF"}
10:45:50.807 RSL: POWER1 = OFF
```

When I tried again, `MaxPowerReachedRetry1` was triggered after 6 `MaxPowerReached1` events which corresponds as the counter is set up this way
`Energy->mplr_counter[phase] = Settings->param[P_MAX_POWER_RETRY] +1;  // SetOption33 - Max Power Retry count`

Is it correct to have the +1? Because `SetOption33` is 5, it should turn off completely after 5 times, instead of 6 times.

```
10:57:05.644 RSL: RESULT = {"POWER1":"ON"}
10:57:05.646 RSL: POWER1 = ON
10:57:09.785 RSL: WARNING = {"Time":"2024-07-03T10:57:09","MaxPowerReached1":113}
10:57:09.791 RSL: SENSOR = {"Time":"2024-07-03T10:57:09","ENERGY":{"TotalStartTime":"2024-03-15T00:00:00","Total":0.031,"Yesterday":0.013,"Today":0.015,"Power":[114,25],"ApparentPower":[114,25],"ReactivePower":[0,0],"Factor":[1.00,1.00],"Voltage":234,"Current":[0.485,0.108],"BL09XX":{"Temperature":43.2}}}
10:57:09.798 RSL: RESULT = {"POWER1":"OFF"}
10:57:09.800 RSL: POWER1 = OFF
10:58:12.814 RSL: RESULT = {"Time":"2024-07-03T10:58:12","PowerMonitor1":"ON"}
10:58:12.819 RSL: RESULT = {"POWER1":"ON"}
10:58:12.820 RSL: POWER1 = ON
[... cut ...] ature":44.0}},"TempUnit":"C"}
11:03:52.798 RSL: WARNING = {"Time":"2024-07-03T11:03:52","MaxPowerReachedRetry1":"OFF"}
11:03:52.804 RSL: SENSOR = {"Time":"2024-07-03T11:03:52","ENERGY":{"TotalStartTime":"2024-03-15T00:00:00","Total":0.034,"Yesterday":0.013,"Today":0.019,"Power":[0,25],"ApparentPower":[0,25],"ReactivePower":[0,0],"Factor":[0.00,1.00],"Voltage":234,"Current":[0.000,0.107],"BL09XX":{"Temperature":44.4}}}
11:03:52.809 RSL: RESULT = {"POWER1":"OFF"}
11:03:52.811 RSL: POWER1 = OFF
```

arendst, 2024-07-03:
Thx for testing.

Your observations are useful. 

I refactored the MaxPower retry loop and think the current solution works much better:

```
13:32:22.638 CMD: so33 2
13:32:22.639 SRC: WebConsole from 192.168.2.1
13:32:22.641 CMD: Grp 0, Cmd 'SO', Idx 33, Len 1, Pld 2, Data '2'
13:32:22.648 MQT: stat/dualr3c/RESULT = {"SetOption33":2}
13:32:23.287 CFG: Saved, Count 4641, Bytes 4096
13:32:27.061 CMD: power1 1
13:32:27.062 SRC: WebConsole from 192.168.2.1
13:32:27.064 CMD: Grp 0, Cmd 'POWER', Idx 1, Len 1, Pld 1, Data '1'
13:32:27.070 MQT: domoticz/in = {"idx":398,"nvalue":1,"svalue":"","Battery":100,"RSSI":10}
13:32:27.077 MQT: stat/dualr3c/RESULT = {"POWER1":"ON"}
13:32:27.083 MQT: stat/dualr3c/POWER1 = ON
13:32:28.189 CFG: Saved, Count 4642, Bytes 4096
13:32:39.828 MQT: stat/dualr3c/WARNING = {"Time":"2024-07-03T13:32:39","MaxPowerReached1":23}
13:32:39.838 MQT: tele/dualr3c/SENSOR = {"Time":"2024-07-03T13:32:39","ENERGY":{"TotalStartTime":"2023-04-14T15:18:02","Total":[4.898,8.120],"Yesterday":[0.059,0.079],"Today":[0.008,0.046],"Power":[23.280,0.000],"ApparentPower":[24.041,0.000],"ReactivePower":[6.000,0.000],"Factor":[0.97,0.00],"Frequency":50.02,"Voltage":229.2,"Current":[0.105,0.000]}}
13:32:39.841 SRC: MaxPower
13:32:39.848 MQT: domoticz/in = {"idx":398,"nvalue":0,"svalue":"","Battery":100,"RSSI":9}
13:32:39.854 MQT: stat/dualr3c/RESULT = {"POWER1":"OFF"}
13:32:39.857 MQT: stat/dualr3c/POWER1 = OFF
13:32:40.315 CFG: Saved, Count 4643, Bytes 4096
13:33:12.857 MQT: stat/dualr3c/RESULT = {"Time":"2024-07-03T13:33:12","PowerMonitor1":"ON"}
13:33:12.859 SRC: MaxPower
13:33:12.864 MQT: domoticz/in = {"idx":398,"nvalue":1,"svalue":"","Battery":100,"RSSI":10}
13:33:12.866 MQT: stat/dualr3c/RESULT = {"POWER1":"ON"}
13:33:12.868 MQT: stat/dualr3c/POWER1 = ON
13:33:13.182 CFG: Saved, Count 4644, Bytes 4096
13:33:25.838 MQT: stat/dualr3c/WARNING = {"Time":"2024-07-03T13:33:25","MaxPowerReached1":23}
13:33:25.848 MQT: tele/dualr3c/SENSOR = {"Time":"2024-07-03T13:33:25","ENERGY":{"TotalStartTime":"2023-04-14T15:18:02","Total":[4.898,8.120],"Yesterday":[0.059,0.079],"Today":[0.008,0.046],"Power":[23.265,0.000],"ApparentPower":[23.997,0.000],"ReactivePower":[5.900,0.000],"Factor":[0.97,0.00],"Frequency":50.02,"Voltage":229.0,"Current":[0.105,0.000]}}
13:33:25.850 SRC: MaxPower
13:33:25.858 MQT: domoticz/in = {"idx":398,"nvalue":0,"svalue":"","Battery":100,"RSSI":10}
13:33:25.866 MQT: stat/dualr3c/RESULT = {"POWER1":"OFF"}
13:33:25.872 MQT: stat/dualr3c/POWER1 = OFF
13:33:26.179 CFG: Saved, Count 4645, Bytes 4096
13:33:58.858 MQT: stat/dualr3c/RESULT = {"Time":"2024-07-03T13:33:58","PowerMonitor1":"ON"}
13:33:58.860 SRC: MaxPower
13:33:58.864 MQT: domoticz/in = {"idx":398,"nvalue":1,"svalue":"","Battery":100,"RSSI":10}
13:33:58.866 MQT: stat/dualr3c/RESULT = {"POWER1":"ON"}
13:33:58.868 MQT: stat/dualr3c/POWER1 = ON
13:33:59.326 CFG: Saved, Count 4646, Bytes 4096
13:34:11.828 MQT: stat/dualr3c/WARNING = {"Time":"2024-07-03T13:34:11","MaxPowerReachedRetry1":"OFF"}
13:34:11.838 MQT: tele/dualr3c/SENSOR = {"Time":"2024-07-03T13:34:11","ENERGY":{"TotalStartTime":"2023-04-14T15:18:02","Total":[4.898,8.120],"Yesterday":[0.059,0.079],"Today":[0.008,0.046],"Power":[23.305,0.000],"ApparentPower":[24.053,0.000],"ReactivePower":[6.000,0.000],"Factor":[0.97,0.00],"Frequency":50.02,"Voltage":229.3,"Current":[0.105,0.000]}}
13:34:11.841 SRC: MaxPower
13:34:11.850 MQT: domoticz/in = {"idx":398,"nvalue":0,"svalue":"","Battery":100,"RSSI":10}
13:34:11.854 MQT: stat/dualr3c/RESULT = {"POWER1":"OFF"}
13:34:11.857 MQT: stat/dualr3c/POWER1 = OFF
13:34:12.201 CFG: Saved, Count 4647, Bytes 4096
```

With ``SO33 2`` the ``MaxPowerReachedRetry1`` message is now shown after the second power on from the ``PowerMonitor1`` as expected.

closed 2024-07-06 by arendst