arendst/Tasmota #21726: After migration to Universal Display Driver WS2812 and OLED SSD1306 somehow interconnected

dkadioglu, 2024-07-04:
I still have the issue, that power controls always both the WS2812 LED stripe and the OLED display and transfer the discussion to an issue, I hope that is ok.

### Discussed in https://github.com/arendst/Tasmota/discussions/21679

<div type='discussions-op-text'>

<sup>Originally posted by **dkadioglu** June 22, 2024</sup>
I have migrated to Universal Display Driver according to https://tasmota.github.io/docs/Universal-Display-Driver/#migrating-to-udisplay. However, now I have an issue with my also connected WS2812 LED stripe. When I run DisplayDimmer x, also all 20 LEDs of my WS2812 get enabled. When I disable them again with Power off, the display is also switched off. I also reset the device to the default settings, still the same issue. Is this something known? Anything I can do about that?
I have a ESP8266 NodeMCU board and a SSD1306 OLED display with a self built build, as I need MQTT_TLS.

Edit 240623:
What I realized just now is, that before the update I had two Toggle buttons in the web ui, one for the LED stripe and one for the OLED display. After the update I now have only one button, which influences both devices.</div>

If you need further info, please ask.

arendst, 2024-07-05:
I notice the same if I setup a display with a light. 

I'll chk if anything can be done to overrule the display backlight to become a light.

arendst, 2024-07-05:
Pls provide your GPIO config (command ``gpio``) and output of command ``status 0``.

dkadioglu, 2024-07-05:
It works again! Thank you very much for the immediate fix! I'll close this as it is fixed for me.

closed 2024-07-05 by dkadioglu