arendst/Tasmota #21733: TASMOTA ADC0 pin relay

yaminahacs, 2024-07-05:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
After replacing a switch module with TASMOTA I found out that the relay is on ADC0 pin and I was unable to use it. I would like to request to enable ADC0 pin as ADC relay and ADC switch, just like you can select ADC button.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
10:45:38.943 CMD: status 0
10:45:38.951 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Lounge","FriendlyName":["Light_1","Light_2","Light_3"],"Topic":"tasmota_0A13FB","ButtonTopic":"0","Power":"000","PowerLock":"000","PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[5,5,5,5,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
10:45:38.993 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T14:02:23","StartupUTC":"2024-07-04T18:43:15","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2024-07-04T17:53:29","SaveCount":110,"SaveAddress":"F6000"}}
10:45:39.028 RSL: STATUS2 = {"StatusFWR":{"Version":"14.1.0(release-tasmota)","BuildDateTime":"2024-06-03T11:53:14","Boot":31,"Core":"2_7_6","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"355/699"}}
10:45:39.062 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MikroTik",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004001","00000000"]}}
10:45:39.106 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":647,"Free":356,"Heap":24,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,!3,!4,!5,!6,7,!8,9,10,12,!16,!18,!19,!20,!21,!22,!24,26,!27,29,!30,!35,!37,!45,62,!68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
10:45:39.146 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-0A13FB-5115","IPAddress":"10.0.0.51","Gateway":"10.0.0.1","Subnetmask":"255.255.255.0","DNSServer1":"209.203.10.208","DNSServer2":"196.22.218.248","Mac":"4C:11:AE:0A:13:FB","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
10:45:39.180 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0A13FB","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
10:45:39.218 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-07-05T08:45:39Z","Local":"2024-07-05T10:45:39","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+02:00","Sunrise":"05:54","Sunset":"21:55"}}
10:45:39.252 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-07-05T10:45:39","Switch1":"ON","Switch2":"ON","Switch3":"ON"}}
10:45:39.286 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-07-05T10:45:39","Uptime":"0T14:02:24","UptimeSec":50544,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","Wifi":{"AP":1,"SSId":"MikroTik","BSSId":"48:A9:8A:13:A7:AC","Channel":11,"Mode":"11n","RSSI":100,"Signal":-49,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2024-07-05:
Alas, ADC0 on ESP8266 is hardware input only. No way to control a relay.

sfromis, 2024-07-05:
To handle such a replacement, you'll have to wire the relay to another gpio. And yes, I know that your screenshot shows none available, but that does not change what I wrote.

What I'd do would be to use an ESP32-based module in the same form factor, allowing all the pins to be used as digital I/O pins. Look in https://templates.blakadder.com/esp32.html to find a module matching the form factor you used when replacing.

closed 2024-07-05 by Jason2866

yaminahacs, 2024-07-09:
noted