arendst/Tasmota #21746: Unintended restart with "Exception":5,"Reason":"Load access fault"

fasterit, 2024-07-08:
### PROBLEM DESCRIPTION
Athom Plug V3 restarted unintentionally (cutting the power for a short moment, too) with the following log entry:

12:15:09.506 RSL: INFO1 = {"Info1":{"Module":"Athom Plug V3","Version":"14.1.0(release-tasmota32)","FallbackTopic":"cmnd/DVES_F96B08_fb/","GroupTopic":"cmnd/tasmotas/"}}
12:15:09.519 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-F96B08-2824","IPAddress":"<redacted>","IP6Global":"","IP6Local":"<redacted>"}}
12:15:09.533 RSL: INFO3 = {"Info3":{"RestartReason":{"Exception":5,"Reason":"Load access fault","EPC":"420da2ec","EXCVADDR":"000000e3","CallChain":["420b0696","42049524","420aa32a","40387eca","403876e0","4038765c","4038814c","4207a116","42014472","421462d0","42075216","42075290","420143d8","42075226","420b8884","42134f02"]},"BootCount":17}}

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Athom Plug V3
- [X] Tasmota binary firmware version number used: 14.1.0(release-tasmota32)
  - [X] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: web-updater
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`: see above
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
see above for relevant excerpt

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
n/a

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
not crash :-)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
Ran fully stable on 13.3 before

Just creating this issue in case you can use the call stack above and/or other people have the same issue.

Kind regards,
DLange/Faster IT GmbH

Jason2866, 2024-07-11:
Just a random crash without more context is useless.

closed 2024-07-11 by Jason2866

fasterit, 2024-07-17:
Happened again after 10d4h of uptime. Will re-flash to 13.4 and see whether that runs stable again (13.3 did, see initial issue report).

fasterit, 2025-03-16:
Fixed by https://github.com/arendst/Tasmota/pull/23115 which lead to commit https://github.com/arendst/Tasmota/commit/e3eb47078fb70aa31e63aec442927a3dbc279b16#diff-b543f71d90ed1df4067eaf318db87ff9a7ae57c1e63c7f326058366f8e6cefe7L511 .